turtlebot/turtlebot4 #274: Nav2 map frame does not exist

mbergman257, 2023-09-21:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed: Installed
 - Package version: ros-galactic-turtlebot4-navigation 0.1.2-1focal.20221208.082815
 - TurtleBot4 Lite : G.5.2
 - Fast DDS
 - Discovery Server

 **Expected behaviour**
 Use nav2 stack to load saved map.yaml and launch navigation on Turtlebot4 lite

 **Actual behaviour**
 Map frame does not exist
[image]

**To Reproduce**
Provide the steps to reproduce:
1. Successfully generate and save map with slam_sync (ref: https://turtlebot.github.io/turtlebot4-user-manual/tutorials/generate_map.html) Turned off TB4...
2. Turn on Turtlebot4
3. ssh into rpi, run: `ros2 launch turtlebot4_navigation nav_bringup.launch.py slam:=off localization:=true map:=/home/ubuntu/lab.yaml` (ref: https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html)
4. Open desktop terminal and run: `ros2 launch turtlebot4_viz view_robot.launch.py`

**Errors**
* Nav terminal: gets stuck at `[controller_server-5] [INFO] [1677764723.699949411] [local_costmap.local_costmap]: Creating Costmap`
* RVIZ Terminal: `[rviz2-1] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist [rviz2-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.5/src/buffer_core.cpp`
* RVIZ Window: see image above
Also tried in new terminal, which allows rviz to shows the map but does not update or show anything else (ref: #40): 
`ros2 lifecycle set map_server configure` 
`ros2 lifecycle set map_server activate`

[image]

**Other notes**
I've tried a lot of things, but I'm not sure what to do now. Looks like the map needs to be published and a tf between map and base_link is needed? Thanks-
* No other ROS devices on network
* Applied this change: ntp config create3 #216
* Using fastrtps fastdds discovery server (ref: https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html)
* All topics seem to be available on desktop.
[image]
* tf tree (cropped) [image]

hilary-luo, 2023-09-21:
After launching localization did you set the 2d Pose Estimate? If you are not sure what I am referring to see [this section](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html#2d-pose-estimate) of the navigation tutorial.

mbergman257, 2023-09-21:
*facepalm* yes I forgot to do this step. I should go to bed. Thanks!

closed 2023-09-21 by mbergman257