turtlebot/turtlebot4 #275: Navigation doesn't work if more than 1 turtlebots are turned ON

ipa-rar-sy, 2023-09-21:
- OS: ubuntu 22.04
 - ROS Distro: humble
 - Built from source or installed: Installed

 **Expected behaviour**
localization launch file for one robot should start and work.

 **Actual behaviour**
 localization does not work when more than 1 namespaced turtlebots are turned on.

I am launching a customized localization file which works if only one robot is powered on. But fails if two robots are powered on.
Also (with single robot navigation) there is an error that says 'Couldn't determine robot's pose associated with laser scan', but robot moves to the nav goal. 

**Other notes**
[image]

my launch file:
[file.txt](https://github.com/turtlebot/turtlebot4/files/12682371/file.txt)

ipa-rar-sy, 2023-09-21:
@hilary-luo, @MikeDegany. please help.

hilary-luo, 2023-09-22:
@ipa-rar-sy Do you mind clarifying what you are trying to achieve with these changes? 

You hardcoded the namespace that you pass through to the nav2 localization launch file to `/agent_0` so it makes sense that you are not able to use this launch file on multiple robots. It would be attempting to launch duplicate nodes etc.

```
localization_agent_0 = GroupAction([
        PushRosNamespace(namespace_agent_0),

        IncludeLaunchDescription(
            PythonLaunchDescriptionSource(
                PathJoinSubstitution(
                    [pkg_nav2_bringup, 'launch', 'localization_launch.py'])),
            launch_arguments={'namespace':'/agent_0',
                              'map': LaunchConfiguration('map'),
                              
                              'use_sim_time': use_sim_time,
                              'params_file': LaunchConfiguration('params')}.items()),
    ])
```

From my brief pass, it looks like you have broken the namespacing and changed the default namespace and changed the default parameter file. If you are just needing to launch with particular namespaces and paramater files then you can do so with the default turtlebot4 localization launch file. Either pass that information in when you call it on the terminal or create a new launch file that calls the original file but that passes in your modified namespaces / parameters.

ipa-rar-sy, 2023-09-22:
@hilary-luo Thank you for the reply. 

I am trying to navigate more than one robot in a single room with one map, for that reason I tried to launch two nodes for agent_0 and agent_1 independently, using 2 different launch nodes and stating the namespaces at the end. It didn't work and I got the error which stated 'deactivating map_server'....  I tried in my PC as well as rpis separately, but both tries failed.

So I tried to make one file in which I can launch localization for different agents. The code that you have highlighted is for one robot. 
This works if I power on only 1 robot. Robot can move to the goal pose. But still gives error "Couldn't determine robot's pose associated with laser scan".  with one map
and 
"Failed to transform initial pose in time (Lookup would require extrapolation into the future.  Requested time 1692802681.726622 but the latest data is at time 1692802681.664459, when looking up transform from frame [base_link] to frame [odom])".

But if I power on 2 robots simultaneously and launch the file for 1 robot it doesn't work.For both robots by uncommenting the code at below also doesn't work. The different namespace and independent yaml files are provided in the code. I am trying this because I have one simulation code which works and I was trying to make similar thing for real robots.

[localization.txt](https://github.com/turtlebot/turtlebot4/files/12698521/localization.txt)

If you have some other way by which I can do navigation of more than one robot in one map simultaneously, then please tell me that as well.

hilary-luo, 2023-09-24:
If you are wanting to run multiple robots all using the same map-server node then you will not be able to just edit the turtlebot4 localization launch file. The way this file works is that it will launch its own namespaced map server. It does so by pushing a namespace and then calling the default nav2 localization_launch.py, which launches a map_server, amcl node and lifecycle manager - each time. 

The multirobot navigation that we currently support is running the two robots completely separately with their own map-servers and everything although they can be located in the same room (aka launching localization twice, each time with a different robot's namespace). Running multiple robots off of the same map server is definitely something that you should be able to do with the turtlebots but is not something that I can directly support you in doing. This is an exercise in knowing how to configure and launch nav2, not the turtlebot4 itself. I would encourage you to take advantage of the tutorials online such as https://www.youtube.com/watch?v=cGUueuIAFgw and reach out on [robotics stack exchange](https://robotics.stackexchange.com/?newreg=b1356041b96546379a29dc15be5fa786) for support. The latest file you shared seems like you are heading in the right direction.

If you launch the default turtlebot4 navigation as intended and are getting errors in the terminal output (such as the deactivated map or failed to transform error) then that is something that I can help you troubleshoot. If this is the case then please specify exactly how you are launching it and provide the full terminal output with the errors.

ipa-rar-sy, 2023-10-04:
Hello @hilary-luo. Thank you for the reply.
I understood the multiple robot tutorial part. But I have one more doubt. 
If I launch nav for only one robot with ros2 launch turtlebot4_navigation localization.launch.py namespace:=/agent_0 and 
ros2 launch turtlebot4_navigation nav2.launch.py namespace:=/agent_0, and agent_0, agent_1 are powered ON at the same time. The navigation should work with agent_0. but it doesn't run. Both agent_0 and agent_1 are on same ros domain ID.

ibluesna, 2023-10-06:
Hello. I'm trying to run nav2 with only one robot.However, although it's just a single agent, navigation package doesn't work ( I guess it's related to the network setup(?) - issue: connecting map and odom frame. Can you please let me know your network setup? I just followed the manual (simple discovery server: connected PI with 5G and creator 2.4G on the same router).

ipa-rar-sy, 2023-10-09:
@ibluesna Can you post the issues? So that I can know what is the problem. It takes a a while for my system to generate Tf transformations, but 3-4 min later I can see the tfs. You can try waiting for a while and see if you see the transformations. And also send a picture of bashrc if you can.

hilary-luo, 2023-10-12:
@ibluesna Please open your own ticket, with full details about your setup. Your situation is very different from the one we are discussing here which is specifics about mutli-robot functionality. 

@ipa-rar-sy I appreciate your patience, I have been very busy. In the following:
 
> If I launch nav for only one robot with ros2 launch turtlebot4_navigation localization.launch.py namespace:=/agent_0 and ros2 launch turtlebot4_navigation nav2.launch.py namespace:=/agent_0, and [...]

are you referring to the unmodified localization and navigation launch files? If not then try it with the unmodified launch files. If you are using the unmodified launch files, can you confirm that each of the robots is properly namespaced including the create3 topics. Then can you recreate the situation and give it some time (~5min) and then share the create3 logs from each of the robots?

closed 2023-11-16 by hilary-luo