turtlebot/turtlebot4 #276: battery issue

ynrng, 2023-09-23:
**Please provide the following information:**
 - OS: (. Ubuntu 22.04)
 - ROS Distro: (. humble)
 - installed: 

 **Expected behaviour**
turtlebot power on 

 **Actual behaviour**
raspberry pi powers on shortly for several seconds then dies off.  
i opened the shell. and very certain the lights on Raspberry Pi went off after ~3 seconds after connecting to the charging station. in this 3 seconds, the lidar module was on and spinning. then after that the lights on Raspberry Pi went off and lidar module stopped. 
the button light on the robot is on during the whole process. and it showed green light at one point(after about 5min). circling white light at most time.
the led screen never turned on.
there's a yellow light on in lower bottom case (below Raspberry Pi )
as soon as I took it off the charging station everything stopped.

I tried to take out the battery and re flush the sd card. nothing changed.
(obviously can't ssh into that bc Raspberry Pi  is off.)

when first connected to charging station:

https://github.com/turtlebot/turtlebot4/assets/10446823/a392f6d4-f40f-428b-a510-df7a3e5d02ce

Raspberry Pi:

[image]

**To Reproduce**
Provide the steps to reproduce:
1. connect to charging station

**Other notes**
it was good yesterday and turned off automatically. didn't do anything later. now it can't be powered on.

hilary-luo, 2023-09-24:
It sounds like your raspberry pi is losing power. This is independent to the light indication on the create3. 

If you ever have questions about the light on the create3 (the light around the button) you can see the reference info here: https://iroboteducation.github.io/create3_docs/hw/face/. The light ring spinning white is what happens while the robot is turning on and a green flash indicates that it connected to the wifi. Please ensure that the battery was reinstalled properly and securely and I would encourage you **not** to take the robot off of the dock while it is actively booting up.

To troubleshoot power to the raspberry pi, can you please verify that the following connections are secure / undamaged:
- USB C connection on the pi: https://github.com/turtlebot/turtlebot4/issues/198#issuecomment-1598076029
- User interface board: https://github.com/turtlebot/turtlebot4/issues/198#issuecomment-1599457193

And then try again to power up the robot and confirm if the pi, lidar and screen retain power? To ensure that the create3 battery has been reinstalled properly, can you verify that after it finishes booting up and you remove it from the dock, that the create3 retains power?

ynrng, 2023-09-24:
Yes, they were correctly attached. 
I left it overnight uncharged and tried it again today. The problem (miraculously) goes away by itself.
I appreciate your support nonetheless.

closed 2023-09-24 by ynrng