turtlebot/turtlebot4 #278: The standard version of Raspberry Pi is replaced with nx and other SBCs.

ncnynl, 2023-09-24:
**Describe the the feature you would like**
The standard version of Raspberry Pi is replaced with nx and other SBCs.

**Other notes**
I have basically finished switching the standard version from Raspberry Pi 4B to Xavier nx. The same applies to other nano, orin nx, and Xavier.

I have a question here. I found that the pin port for i2c does not use the regular i2c of the Raspberry Pi. The current display on the pcba uses pin7 (GPIO4) and pin29 (GPIO5) of the Raspberry Pi as i2c instead of Using pin3 (SDA) and pin4 (SCL) of Raspberry Pi identity? If this position is used, there is no need to change the wiring, and the original wiring can be used directly.

At the same time, the program also needs to make corresponding pin port correspondences.

   gpio:
       user_button_1: 130 # 19 for rpi4, 130 for xaiver nx
       user_button_2: 133 # 16for rpi4, 133 for xaiver nx
       user_button_3: 151 # 26 for rpi4, 151 for xaiver nx
       user_button_4: 129 # 20 for rpi4, 129 for xaiver nx

       led_green_power: 112 # 17 for rpi4, 112 for xaiver nx
       led_green_motors: 127 # 18 for rpi4, 127 for xaiver nx
       led_green_comms: 149 # 27 for rpi4, 149 for xaiver nx
       led_green_wifi: 152 # 24 for rpi4, 152 for xaiver nx
       led_green_battery: 16 # 22 for rpi4, 16 for xaiver nx
       led_red_battery: 153 # 23 for rpi4, 153 for xaiver nx
       led_green_user_1: 150 # 25 for rpi4, 150 for xaiver nx
       led_green_user_2: 106 # 6for rpi4, 106 for xaiver nx
       led_red_user_2: 128 # 21 for rpi4, 128 for xaiver nx

       display_reset: 84 # 13 for rpi4, 84 for xaiver nx

base:
       gpiochip0: 1 # 0 for rpi4, 1 for xaiver nx
       gpiochip1: 2 # 2 for xavier nx
       i2c_bus: 8 #3 for rpi4, 4 for nx
       i2c_device_id: 0x3c #0x3C for rpi4 nx

for led_green_battery: 16 # 22 for rpi4, 16 for xaiver nx, is under gpiochip2, the other is under gpiochip1

hilary-luo, 2023-09-24:
> I have a question here. I found that the pin port for i2c does not use the regular i2c of the Raspberry Pi. The current display on the pcba uses pin7 (GPIO4) and pin29 (GPIO5) of the Raspberry Pi as [...]

Can you please clarify what the question is that you are asking here? are you looking for confirmation of this statement?

>**Describe the the feature you would like**
The standard version of Raspberry Pi is replaced with nx and other SBCs.

Can you also please clarify what the feature is that you are asking for? are you asking for the software to support this as an option or instructions on how to do this?

ncnynl, 2023-09-24:
I don't know if you have tried using xavier nx instead of raspberry pi. If you have ever made a replacement, you may know what needs to be changed. I've finished using NX as a replacement for a raspberry pi and can also use the pcba for LEDs, buttons and displays.

Our problem is that we can make the hardware and software more versatile and support other SBCs to use PCBA

From what I've put together so far, both hardware and software changes are required.

Hardware change issues:

1. Use NX to connect to pcba. The only problem at present is that the pins of the display screen do not use conventional i2c.

i2c under NX uses bus 8, pin3 and pin5
Raspberry Pi's i2c uses bus 1, pin3 and pin5

But in fact, the Raspberry Pi uses bus 3, pin7 and pin9. This results in the corresponding pins not being supported by default. If the Raspberry Pi also uses pin3 and pin5 of bus 1. But currently, jumpers are needed to connect nx to pcba. Pin3 and pin5 need to be connected to pin7 and pin9 on pcba.

If the pcba hardware can change the use of pin3 and pin5, there is no need for jumpers on NX. The wiring of PCBA can be directly connected to the Raspberry Pi.

Software change issues:

1. Use NX to connect to PCBA. In terms of software, first configure the pins. The only problem currently is that it crosses the chipset.

The Raspberry Pi uses gpipchip0, while the NX requires the use of two chipsets, gpiochip0 and gpiochip1. Currently, the program only supports one chipset setting, which is still the default setting.

2. The pins corresponding to NX can be configured in turtlebot4.yaml

3. Pin 16 is the LED light corresponding to the battery. This is on the gpiochip1 chipset and needs to be programmed separately.

4. To improve the program, you can add configurations to turtlebot4.yaml, such as setting the chipset, setting the i2c bus number, and setting the i2c device number (this may not be necessary, because your i2c device is ssd1306, and the device number is is 0x3c)

The above hardware and software changes will make it more versatile and make other SBCs more convenient to access.

I don’t know if my description can be clearer.

ncnynl, 2023-09-25:
Or I will submit the modified version of my software to you for reference later. My modification only involves turtlebot4_base and turtlebot4_bringup.

ncnynl, 2023-09-25:
This is my code 
https://github.com/ncnynl/turtlebot4_robot/tree/galactic

ncnynl, 2023-09-26:
This is the Turtlebot4+NX version that I completed the upgrade.

[image]

euivmar, 2023-10-19:
> This is the Turtlebot4+NX version that I completed the upgrade.
> 
> [image]

Thank you for sharing your work. I was wondering about making an upgrade to rasp5 since I need more computational power.  I believe it would be easier since the Pinout is the same (correct me if I am wrong). 

May I ask why are you still using Galactic and not Humble?

ncnynl, 2023-10-21:
@euivmar 
Theoretically, Raspberry Pi 5 should be able to directly replace Raspberry Pi 4, but in terms of computing power, I’m not sure how much difference there is between nx and Raspberry Pi 5. If you need GPU ecological support, Nvidia’s board may be more suitable.

I upgraded the Raspberry Pi version to humble in the early days, but it had more problems than galactic. A particularly serious problem is that when used for a period of time, the chassis will automatically shut down. Since the galactic version is more stable, it has not been upgraded to humble for the time being. At the same time, nx only supports the version of ubuntu20.04, so combined, currently only the galactic version is used.

It will be upgraded to humble in the future and is also compatible with different host computers. After all, the Raspberry Pi as a host computer is only a minimum configuration option.

Relatively speaking, I prefer the lite version, which provides an extensible degree of freedom. It would be better if there was a pure chassis version. PC, radar, and camera can be freely selected. We are in China and currently cannot purchase this chassis directly.

martincerven, 2024-06-09:
Is there any update on this? One can use Create3 packages with Jetson devices, but it might be beneficial for students to use turtlebot4 packages with Jetson/Rpi5 without any special hacking @hilary-luo

hilary-luo, 2024-06-13:
@martincerven You should be able to use the turtlebot4 packages (configured as a turtlebot4-lite) on a raspberry pi 5 or other PC without the pcba. I cannot confirm but I think the pi 5 should work with the pcba without changes. There have been several examples of people using other computers to build a turtlebot4 lite. There is a setup script with instructions in the setup repository, described here: https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_setup.html#install-scripts that works on a base ubuntu server image. For different build systems (like arm) you may have to build many of the packages from source but other than that the packages should be fully functional. 

I don't currently have any information on an update to add easy pcba support for other computers nor do I currently have information on arm releases.