turtlebot/turtlebot4 #279: setup without wifi

ynrng, 2023-09-25:
Is it possible to setup the robot ( raspberry pi and create 3 ) without wifi connection?
i.e. Is it possible to connect my PC and the robot with a wired Ethernet cable, and run commands from my PC, to move/slam/navigate the robot through the environment. 
We are trying to set up the robot in university. we couldn't connect it to eduroam and other network/wifi access is also constrained.

hilary-luo, 2023-10-06:
Hi @ynrng,

My apologies for taking so long to respond. Yes you can set up the robot without a wifi connection. You can connect with a wired ethernet cable if you wish. Are you not able to setup a router for an isolated local network that does not have internet access? From my understanding many other schools / labs are doing this and it is allowed for them because it is not connected to any external network. This allows them to still operate the robots wirelessly.

ynrng, 2023-10-29:
Hi,
We tried to set it up with wired connection only. We configured the network to be web server for create 3. Then wired our laptop to the robot. But we can’t see any ros topics from our laptop this way. I might have missed some setup steps, if you can point out which steps are needed for this to be working that would be great. I couldn’t find anything dedicated to setting the robot up with wired connections on the website.
We did successfully try using a local network, but it’s not stable as in every time we boot up the  turtle bot, there are large chances that they can’t find their ip address and get an unknown. Then we have to do all the setups for WiFi which takes up looots of time.

hilary-luo, 2023-10-30:
@ynrng when you say:
>We configured the network to be web server for create 3

Do you mean that you connected the raspberry pi to internet in order to access the create3 webserver? or did you connect both the create3 and the raspberry pi?

In order to see the ros topics on your laptop you do need to set up the laptop for the specific networking configuration that you are using. (Irrelevant of wired or wireless, it is still the two same network configuration options) If you haven't changed it from default then you will need to follow the simple discovery instructions found here: https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html#user-pc. There are equivalent [instructions for discovery server](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html#user-pc).

If you are having issues with the DHCP on the local network and want to pursue wireless operation then you could reserve IP addresses for each of the robots (using a setting in the router). That way they always get assigned the same IP address. If a different device already has that IP address leased then you may need to cancel the lease. There is a lot of information available online for doing this sort of thing.

hilary-luo, 2023-12-17:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-12-17 by hilary-luo

hysamah, 2023-12-22:
We are having identical issues with our set-up (also a university campus). when we our laptop to the robot we can’t see any ros topics from our laptop this way. Is there a fix to this? 
Thanks