turtlebot/turtlebot4 #280: No communication between RPi and Create3 base

HyydenJakcz, 2023-09-27:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: installed
 - Discovery server
 - Fast DDS

 **Expected behaviour**
running the turtlebot tests and passing every test

 **Actual behaviour**
None of the ring lights changed color, the robot did not move during the driving test, pressed the create3 buttons but it did not receive an input

**To Reproduce**
Provide the steps to reproduce:
1. setup the discovery server on the turtlebot
2. ssh into the turtlebot and run the tests
3. see the error

additional notes:
only 3 out of 5 of the indicator lights are on (left, middle and right). The error appeared sometime after setting up the network and the motor wheels disengaged, so I restarted the robot.

HyydenJakcz, 2023-09-28:
I reimaged the pi and setup the network again and it seems to have done the trick. It would be nice to know why that happened though

ibluesna, 2023-10-06:
Maybe the creator3 is connected with Wifi providing 5G ? I also experienced it when I choose the wrong network. In manual, it says creator3 is only able to work with 2.4G. After that, I hope you can run navigation package, as i'm struggling from launching it. :(

HyydenJakcz, 2023-10-06:
As far as I remember, the create3 was factory resetted, so I don't think that was the problem. So far, I haven't run into this same issue again. I just pay extra attention to disconnect and power off everything properly.

hilary-luo, 2023-10-12:
@HyydenJakcz I can't tell you for certain why this happened but there is a high likelyhood that it is related to the create3 cpu at 100% issue that is being looked into. If this does happen again, please see if you can capture a [create3 log](https://iroboteducation.github.io/create3_docs/webserver/logs/) from the event and share it. 

Are you looking for any other support at this time?

HyydenJakcz, 2023-10-12:
Thank you for the response. No, I don't have any more problems at the moment.

closed 2023-10-12 by HyydenJakcz