turtlebot/turtlebot4 #281: Odom<-> Map frame issue (AMCL not running)

ibluesna, 2023-10-06:
I'm having troubles when launching navigation packages.
I guess the reason is that the odom frame and map frame are not connected. (AMCL is launched)
Also, the teleop keyboard randomly works. I also found out that the odom frame comes up very slowly and suddenly disappear and appear again.
I launch every navigation related nodes in my remote PC and odom, tf, lidar data is published through the PI.

**Remote PC & Robot Setup**
1. Remot PC: Ubuntu 22.04 + Humble
2. Turtlebot PI Version: Humble v1.0.0
3. Simple Discovery server: Pi+IPTIME 5g, Creator3+IPTIME 2.4G
4. Creator3 version H.2.3 
5. TF tree below
[image]

**Problems**
1. When I tried to run navigation on remote PC (NOT Pi), sometimes(or more closer to often) suddenly the pi is dead and started to reboot
2. If the navigation package runs, AMCL cannot connect with map and odom frame.
==> I tried to set the global frame with odom in Rviz, then the TF says wheel frames aren't connected, although the tf tree shows wheel frames.
3. Odom frame pops up very slowly.

[image]

**Other notes**
I'm struggling from those issues since I got the robot... 
Also is there any way to communicate through the LAN port?
I tried to navigate through LAN Port not using WIFI using discovery server, I couldn't get any topics or nodes through LAN (Maybe I have little knowledge about DDS, so I have no idea how to use it with wired network)

Please help!!!

hilary-luo, 2023-10-12:
Thanks for your patience. You should be able to communicate through the wired LAN port but you would need to be targeting the static ethernet IP address (`192.168.185.3`) instead of the wifi IP address displayed on the screen. That being said, it should not be necessary to wire directly to the robot.

What you are describing does not sound like normal performance. I am suspicious that there could be excessive CPU usage for some reason or issues with the power cabling to the pi.

Can you start by checking the CPU usage on the raspberry pi and sharing the [create3 logs](https://iroboteducation.github.io/create3_docs/webserver/logs/) from during one of these attempts? If you have run the turtlebot4 ROS 2 Tests then also please make sure to remove the test records from the pi in order to ensure that there is enough memory available. 

After doing so, try disabling the diagnostics on the turtlebot4 (through the turtlebot4-setup tool) and let me know if that impacts the operation. 

As a side note, you mention that you are using the original Humble image for the PI, have you updated your packages since then? if not I would suggest that you do so so that you have the latest updates and bug fixes (`sudo apt update`, `sudo apt upgrade`)

ibluesna, 2023-10-25:
I checked the CPU it usa up to 60~70% when I run the navigation nodes. But it doesn't work and odom frame still disapper and appear continuously ;< maybe i'll try with discovery server. Also apt get update & upgrade had been

The log file says 
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 syslog.info syslogd started: BusyBox v1.30.1
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.notice kernel: klogd started: BusyBox v1.30.1 (2023-09-14 18:29:12 UTC)
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.notice kernel: [    0.000000] Linux version 4.14.115-yocto-standard-custom (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 Thu Sep 14 18:38:47 UTC 2023
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Board has DDR2
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Analog PMU set to hw control
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Digital PMU set to hw control
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] MIPS: machine is iRobot Sundial
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Determined physical RAM map:
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000]  memory: 04000000 @ 00000000 (usable)
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Initial ramdisk at: 0x82da0000 (2057376 bytes)
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Zone ranges:
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Movable zone start for each node
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Early memory node ranges
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 80562820, node_mem_map 81000060
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.debug kernel: [    0.000000]   Normal zone: 128 pages used for memmap
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x47c with crng_init=0
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.notice kernel: [    0.000000] Kernel command line: bl_version=4.0.3 console=ttyS1,115200 earlyprintk=serial,ttyS1,115200,keep loglevel=3 ubi.mtd=5 ubi.block=0,rootfs root=/dev/ram rw
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Writing ErrCtl register=0003f53a
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Readback ErrCtl register=0003f53a
Mar 22 21:59:52 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 kern.info kernel: [    0.000000] Memory: 56900K/65536K available (4211K kernel code, 192K rwdata, 1112K rodata, 172K init, 218K bss, 8636K reserved, 0K cma-reserved)
[... cut ...]
Mar 22 22:34:50 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 user.notice create-platform: [INFO] [1679524490.577231667] [system_health]: CPU usage: max 100 [%] mean 100 [%] RAM usage: 38/59 [MB]
Mar 22 22:35:46 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 user.notice ntpd: ntpd: timed out waiting for 192.168.1.137, reach 0x00, next query in 129s
Mar 22 22:35:50 iRobot-D5CFD846BE8C4DCBA425ADBC38D883D6 user.notice create-platform: [INFO] [1679524550.578777605] [system_health]: CPU usage: max 100 [%] mean 96 [%] RAM usage: 39/59 [MB]

ibluesna, 2023-10-25:
After changing it to discovery server, I can't see any odom frame in tf tree but odom topic is still exist :<

hilary-luo, 2023-10-26:
@ibluesna Can you also make sure that you have disabled the diagnostics and then try to replicate running navigation and provide a new log from the create3? I want to see the CPU usage on the create3 while nav was active after those other changes.

closed 2023-12-17 by hilary-luo