turtlebot/turtlebot4 #282: TB4 not under control

gmz123513, 2023-10-07:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed:
 - Package version: sudo dpkg -s ros-ROS2-turtlebot4-PACKAGE_WITH_ISSUE`

 **Expected behaviour**
 The robot should be controlled by the directional keys

 **Actual behaviour**
 While driving the TurtleBot4 Lite version, the chassis does not respond to directional keys. Pressing the forward key causes it to continuously spin in place, and there have been instances where pressing the left or right keys also only makes it move forward. Rebooting the machine and upgrading the Create3 chassis have not resolved the issue. However, occasionally, it can be controlled using keyboard functions. What could be causing this, and are there any solutions?

https://github.com/turtlebot/turtlebot4/assets/75788043/aa3a0221-6fe7-4335-b3ff-409d6d76a551

**To Reproduce**
Provide the steps to reproduce:
1. run something
2. launch something else
3. see the error

**Other notes**
Add anything else you thing is important.

gmz123513, 2023-10-13:
I may not have clearly stated this issue. My directional keys are controlled by i, j, and l, but I know they are used to indicate forward. The main problem is that when I press i to advance, TB4 rotates in place, as shown in the video, and I keep pressing i to advance

hilary-luo, 2023-10-15:
Thank you for the clarification. Have you tried echoing the `/cmd_vel` topic to make sure that only the commands that you expect are being seen there and that they correspond with the commands that you are sending? 

If you can replicate this issue, can you please provide the [create3 log](https://iroboteducation.github.io/create3_docs/webserver/logs/) of one such occurence?

closed 2023-11-16 by hilary-luo