turtlebot/turtlebot4 #284: 4 turtlebots do not respond to commands, if connected to the same network.

ipa-rar-sy, 2023-10-09:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: Installed
 

 **Expected behaviour**
 After connecting 4 turtlebots to same wifi, they should respond to a command

 **Actual behaviour**
None of the turtlebots respond, I cannot even ssh into Rpi if 4 turtlebots are connected to same network.

I am trying to teleop 4 turtlebots at the same time in a formation after running a script. 
All turtlebots do not respond to the cmd vel. Just one or sometimes 2 robots respond. 
I am able to see the topics and even velocity for eah robot getting published but somehow robots fail to move.

They are in the same network, have same Ros Domain ID. I am using simple discovery to perform my task. 
@hilary-luo Can you please tell me an efficient way of network connection that will resolve this issue?

hilary-luo, 2023-10-12:
Are you namespacing your robots? next steps will vary greatly based on the answer to this.

ipa-rar-sy, 2023-10-13:
@hilary-luo Thank you for the reply. 
Yes. I am namespacing all the robots. I have 4 robots namspaced as /agent_0,  /agent_1, /agent_2, /agent_3. All four are connected to same wifi, they have same rod_domain_ID.

hilary-luo, 2023-10-15:
Can you start by trying to connect two robots at the same time and try to drive them with your script and then download and share the [create3 logs](https://iroboteducation.github.io/create3_docs/webserver/logs/) with me?

hilary-luo, 2023-10-31:
@ipa-rar-sy Some of these robots seem to be having the CPU at 100% issue. Can you go through and make sure that you have disabled diagnostics on all of these and see if that impacts operation? Please send new create3 logs after this change. (diagnostics can be disabled through the turtlebot4-setup tool)

There is one robot that doesn't seem to be acting the same as the others. Does agent_2 work on its own? I see no reason why it shouldn't be listening to commands, it is essentially sitting idle. This makes me wonder if there is also an issue with network. Can you confirm that the messages are being sent to it properly and monitor your network bandwidth when you are doing the multi-robot tests?

closed 2023-12-17 by hilary-luo