turtlebot/turtlebot4 #286: Lidar will not stop spinning

sardanian, 2023-10-12:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 22.04)
 - ROS Distro: (e.g. Humble)
 - Built from source or installed: installed from image

My lidar is constantly spinning the moment the raspberry pi boots up. It will not stop. I have tried to stop it via 
`ros2 service call /stop_motor std_srvs/srv/Empty {}`

But that doesn't stop it.

I have to unplug it or the raspberry pi to get it to stop when I am not using it.

hilary-luo, 2023-10-12:
The lidar will keep spinning as long as there is a subscriber on the lidar topic. If you don't want the lidar to spin then make sure that you don't have any active subscribers on that topic. If I recall correctly, that means that you need to turn diagnostics off from the turtlebot4-setup tool. 

The stop motor service is not available when the lidar is set to automatically start and stop. If you want to control that manually then you will need to disable the automatic functionality.

sardanian, 2023-10-12:
I cannot see anything related to diagnostics in the setup. This is on a
fresh boot of the raspberry pi as well.

hilary-luo, 2023-10-12:
You can verify the number of subscribers on the lidar topic by running `ros2 topic info /scan` (assuming you have not added any namespacing). 

You can turn diagnostics off by running `turtlebot4-setup` -> go under ROS Setup -> Bash Setup -> Select TURTLEBOT4_DIAGNOSTICS -> Disabled -> Save -> Back out to the main menu -> Apply Settings -> Yes

PhilerXu, 2023-11-12:
Thanks Hilary!  But it did not seem to work in my case.  Any other suggestions?

> You can verify the number of subscribers on the lidar topic by running `ros2 topic info /scan` (assuming you have not added any namespacing).
>
> You can turn diagnostics off by running [...]

hilary-luo, 2023-11-16:
@PhilerXu If you have checked to make sure that there are no subscribers on the /scan topic already, make sure that you have the latest version of the rplidar package (>=2.1.3). If that still doesn't work try subscribing to scan and then ending the subscription and see if that works.

CurtinRobotAI, 2023-11-24:
I have the same issue, after turning diagnostics off and no subscriber to the topic /scan. The rplidar package is 2.1.4.

hilary-luo, 2023-12-17:
@CurtinRobotAI after turning the robot on with no subscribers on that topic, can you try subscribing in the terminal and then stop your subscription and see if it turns off when it sees the number of subscribers go from 1 to 0? Let me know what you find, if there is an issue we can enter a ticket on the rplidar repo

837liu, 2023-12-24:
Just to add my experience here as a data point.  I set the AutoStandby to False in rplidar.launch.py per the comments in #233 but launched and stopped the process but no joy.  I then turned off diagnostics per earlier comment here and verified that the /scan topic disappeared with 'tos2 topic list', but the lidar motor still didn't stop.  I then ran rplidar.launch.py with AutoStandby set to false (from before), and this time with no subscriber from the diagnostics, and then stopped the process.  That stopped the motor.

CurtinRobotAI, 2023-12-24:
> @CurtinRobotAI after turning the robot on with no subscribers on that topic, can you try subscribing in the terminal and then stop your subscription and see if it turns off when it sees the number [...]

@hilary-luo Thanks for the comments. Once I turned on the Turtlebot4, I used  'ros2 topic info /scan -v' to check whether there is a subscriber for the topic '/scan'. It showed zero subscriber. However, I ran the ros2 topic echo /scan, and then Ctrl+C to kill the echo process, the motor stopped.

arkcom55, 2024-03-25:
- Make sure turtlebot4 diagnostics is turned off
- Run something like the following script automatically on pi boot and LIDAR should stop spinning:

LOG=/home/ubuntu/lidar2.log

function MyKillAll () {
    PROG=$1
    CMD="ps ax|grep '$PROG'|grep -v grep|awk '{\$1=\$1};1'|cut -f 1 -d ' '"
    PS=$(eval "$CMD")
    IFS=$'\n'
    for item in $PS
    do
        echo "Killing $PROG: $item" >> $LOG
        sudo kill -9 $item
    done
}

{
echo "=============================="
CDATE=`date`
echo "Date: $CDATE"

echo "------------------------------"
echo "Sourcing turtlebot4 setup"
export ROBOT_SETUP=/etc/turtlebot4/setup.bash
source $ROBOT_SETUP

echo "------------------------------"
echo "Waiting for scan topic"
while [ 1 ]
do
    ros2 topic list | grep scan
    if [ $? -eq 0 ]
    then
    	echo "See scan topic"
        break
    fi
    sleep 5
done

echo "------------------------------"
echo "Kill any scan"
MyKillAll "scan"
echo "Run: timeout 60 ros2 topic echo /scan --once"
timeout 60 ros2 topic echo /scan --once

echo "DONE--------------------------"
} 2>&1 >> $LOG

moriarty, 2024-09-07:
> You can verify the number of subscribers on the lidar topic by running `ros2 topic info /scan` (assuming you have not added any namespacing).
>
> You can turn diagnostics off by running [...]

On a turtlebox that I just unboxed two weeks ago I can confirm that these instructions worked, the constant lidar spinning while I the robot was charging was annoying