turtlebot/turtlebot4 #287: something wrong with the lidar

ouyangtiezhu74, 2023-10-12:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Galactic
 - Built from source or installed:source

 **Expected behaviour**
Open the terminal, enter this: 'ros2 launch turtlebot4_navigation slam_sync.launch.py,' then enter this: 'ros2 launch turtlebot4_viz view_model.launch.py.' It should open RViz software, and you should be able to see the point clouds returned by the camera and lidar sensors.

 **Actual behaviour**
 nothing was dispalyed ,no camera video,no cloud points.

**To Reproduce**
Provide the steps to reproduce:
1. 'ros2 launch turtlebot4_navigation slam_sync.launch.py
2. ros2 launch turtlebot4_viz view_model.launch.py
3. see the error

hilary-luo, 2023-10-12:
First we need to make sure that you have communication set up between your robot and your PC.

Are you using [Simple Discovery](https://turtlebot.github.io/turtlebot4-user-manual/setup/simple_discovery.html#user-pc) or [Discovery Server](https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html#user-pc) and have you followed the instructions for setting up your PC for the respective networking configuration? 

Are you able to see the topics from your robot on your PC? (run `ros2 topic list` in a terminal and share the output).

ouyangtiezhu74, 2023-10-13:
The communication between robot and PC had been set up. l can opics from PC, I can even control my robot to move by keyboard.The camera video is unstable,somethings l can see the video, sometimes l cant,donno why,but the lidar never works.

ouyangtiezhu74, 2023-10-13:
Thanks for your reply.The communication  between robot and PC has been set up ,I can see those topics and control the robot to move from keyboard.The camera video is unstable,sometimes I got the video,sometimes I dont,I dont know why.But the lidar never works,I never saw the pointcloud,and l was told is because of hardware failure?Do you think so?

hilary-luo, 2023-10-25:
@ouyangtiezhu74 Lets start with the lidar. Can you try sshing into the robot and stopping the turtlebot4 service (`sudo systemctl stop turtlebot4.service`) and then launch the lidar service directly from the terminal (`ros2 launch turtlebot4_bringup rplidar.launch.py`). Share the output in this terminal. Also, check to make sure the lidar is spinning, that the scan topic is present and try to echo the topic. If these things are not successful try switching which usb port that the lidar is plugged into and see if that changes anything.

ouyangtiezhu74, 2023-11-12:
I can sshing into it.I can see the output of lidar,its all zeros and inf .It is spining.The topic is still there.USB port changed,still nothing.Could it just be a hardware problem?

ouyangtiezhu74, 2023-11-12:
I can ssh into it.I can see the output of lidar,its all zeros and inf .It is spining.The topic is still there.USB port changed,still nothing.Could it just be a hardware problem?

hilary-luo, 2023-11-16:
@ouyangtiezhu74 It is possible but to help understand what is happening, I need you to run those commands and share the terminal output with me.

hilary-luo, 2023-12-17:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

closed 2023-12-17 by hilary-luo

kscottz, 2025-07-17:
I am running into this issue too with a recent update to Jazzy.

```
ubuntu@turtlebot4:~$ ls -l /dev/RPLIDAR
lrwxrwxrwx 1 root root 7 Jul  2 10:04 /dev/RPLIDAR -> ttyUSB0
ubuntu@turtlebot4:~$ sudo systemctl stop turtlebot4.service
[sudo] password for ubuntu: 

^C
ubuntu@turtlebot4:~$ ^C
ubuntu@turtlebot4:~$ sudo systemctl stop turtlebot4.service
^C
ubuntu@turtlebot4:~$ ros2 launch rplidar_ros rplidar.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2025-07-17-19-38-31-424537-turtlebot4-10208
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rplidar_composition-1]: process started with pid [10277]
[rplidar_composition-1] [INFO] [1752795512.637801064] [rplidar_composition]: RPLIDAR running on ROS 2 package rplidar_ros. SDK Version: '1.12.0'
[rplidar_composition-1] [INFO] [1752795515.148969983] [rplidar_composition]: RPLIDAR S/N: 438BED93C0EA98C7C2E29EF5B1514060
[rplidar_composition-1] [INFO] [1752795515.149120425] [rplidar_composition]: Firmware Ver: 1.29
[rplidar_composition-1] [INFO] [1752795515.149187777] [rplidar_composition]: Hardware Rev: 7
[rplidar_composition-1] [INFO] [1752795515.150829684] [rplidar_composition]: RPLidar health status : '0'
[rplidar_composition-1] [INFO] [1752795515.151023071] [rplidar_composition]: Start
[rplidar_composition-1] [INFO] [1752795515.695390935] [rplidar_composition]: current scan mode: Sensitivity, max_distance: 12.0 m, Point number: 7.9K , angle_compensate: 2, flip_x_axis 0
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rplidar_composition-1] [INFO] [1752795556.825989801] [rclcpp]: signal_handler(signum=2)
```

The LIDAR seems to not to be providing usable data, just "inf" values.