turtlebot/turtlebot4 #288: The turtlebot4 is not showing up in the map

Tatef3, 2023-10-16:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: installed

 **Expected behaviour**
 the robot is showing up in the map Rviz while launching SLAM

 **Actual behaviour**
 using sync or asyn the map is being updated in Rviz but the the robot is not showing up in the map 

**To Reproduce**
Provide the steps to reproduce:
1. run ros2 launch turtlebot4_navigation slam.launch.py
2. run ros2 launch turtlebot4_viz view_robot.launch.py
3. see errors

A further description of the problem : 
i am using discovery server (using Uni Wifi) so the create3 is not connected to wifi . only the PC and Rpi is connected to wifi . i am having these errors as shown in the pictures. On the remote PC i cannot see any feedback when i echo topic /odom . and there is no transformation map -> odom -> base_link to see the robot although the map is being created and updated. For now i can only see the robot when i use run ros2 launch turtlebot4_viz view_model.launch.py to see laser scans .

[image]
[image]
[image]

Tatef3, 2023-10-19:
Update on the issue : I was getting the odom and battery_state for example data on RPi4 but cant echo that on the PC . i tried to sync the time between the three and now i got another problem that the map is not showing up anymore and on the RPi4 i am having a message dropping filter for lidar data and i didnt have that before time sync .

Tatef3, 2023-10-27:
@hilary-luo Please check my issue , i think you missed it while answering other issues

hilary-luo, 2023-10-30:
@Tatef3 I don't always have time to answer all of the questions so sometimes I can only go through and answer the the ones that I have already started or sometimes I can only go through and answer the quickest ones. I am working my way through catching up on the past few weeks.

I am having difficulties understanding the details of your issue. You mentioned that the map is showing up but not the robot. However, in the images, I don't see either show up when you reference the map frames. More recently you have said that you can't echo data on the PC and that made it so that the map doesn't show up anymore. Can you clarify what you mean when you say the map is or isn't showing up and explain to me what you did when you "tried to sync the time between the three"?

As a baseline, as things stand now, can you see the full topic list on your pc and can you echo the topics?

closed 2023-12-17 by hilary-luo