turtlebot/turtlebot4 #289: Battery issues TurtleBot4

ValentinLeglise, 2023-10-18:
Hello,

A customer acquired his TB4 in march 2023 and he has a problem with it.

"The battery does not seem to charge or the system reads the charge incorrectly. In fact, the robot "loaded"
( 100% battery on the display and green battery led ) try to do the Undocking operation and then the battery drops to 0% and the LED switches from green to red and start blinking"

What is the cause of the problem ?

Thank you, Valentin.

hilary-luo, 2023-10-25:
@ValentinLeglise In order to fully reset the system, please have them turn the robot off and then remove the battery. Wait a few minutes and then put it back in and put it back on the dock to charge. Once the battery indicates 100% again, try to undock. Let me know if the issue persists.

ValentinLeglise, 2023-11-02:
@hilary-luo Hi. We got a response from our customer and his problem persists. 

He says : "The robot starts to undock but suddenly it blocks with 0% of battery and red led. Maybe it is an hardware problem, but it is very strange as we use the robot just for few weeks."

What should the procedure for this ? Thank you.

hilary-luo, 2023-11-02:
@ValentinLeglise I'm sorry to hear that they are still having this issue. 

Have them disconnect the [adapter board](https://turtlebot.github.io/turtlebot4-user-manual/electrical/create3.html) from within the cargo bay. Then have them remove the battery for at least 15 min. After that time is up reconnect the battery but leave the adapter board disconnected and charge the robot for 4 hours. Then power the robot off and reconnect the adapter board. Let me know if the issue still persists.

ValentinLeglise, 2023-11-16:
Hi @hilary-luo . 

Unfortunately, our customer still have the problem after trying your previous instrucions. "The robot undock but suddenly it gives a battery error (with the red led that blink) with the battery that goes to 0%."

Maybe its a hardware problem. What do you think ?

hilary-luo, 2023-11-20:
@ValentinLeglise it is possible that there is a hardware issue but it is not a presentation that I have seen before. Are you in a position to be able to troubleshoot more directly? Specifically I am needing the [create3 log](https://iroboteducation.github.io/create3_docs/webserver/logs/), the firmware and software versions that it is running and if they can echo the `/battery_state` topic and confirm that they are seeing the same result there. 

Also do they own more than one turtlebot? such that they could swap the batteries between two of them to see if the issue follows the battery or the turtlebot itself?

ValentinLeglise, 2023-11-22:
We can't directly troubleshoot his robot. I'll ask him his logs and versions you need and i come back to you. And what do you mean with "echo the /battery_state topic ? And unfortunately, they don't have another tb. Thank you.

hilary-luo, 2023-11-23:
echo the `/battery_state` topic is terminology relating to ROS 2. The system is broadcasting data on different topics and echoing is one way to view the data that is being broadcast. If they ssh into the robot they should be able to run `ros2 topic echo /battery_state` and share the output (they should confirm that it matches the screen too). 

If they are using namespacing they will need to modify that command to add their namespace and they may need to source ROS2 in the terminal first before sending that command. If they are using the ROS2 functionality then they should be somewhat familiar with this terminology.

ValentinLeglise, 2023-11-24:
[logs_turtlebot4.txt](https://github.com/turtlebot/turtlebot4/files/13456610/logs_turtlebot4.txt)

There you can see their logs, and their firmware version is G.3.1. Tell me if it is ok or if i have to tell them to send back the robot.

Regards

ValentinLeglise, 2023-11-30:
Hello,

Can i have an answer for this please ? its pretty urgent.

Regards

> These are logs : [logs_turtlebot4.txt](https://github.com/turtlebot/turtlebot4/files/13456610/logs_turtlebot4.txt)
> 
> This is the echo : ![battery_echo](https://private-user-images.githubusercontent.com/148329757/285445180-bfd07e29-ecc1-4581-8f76-07b67cbd230f.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEzNDA3NjAsIm5iZiI6MTcwMTM0MDQ2MCwicGF0aCI6Ii8xNDgzMjk3NTcvMjg1NDQ1MTgwLWJmZDA3ZTI5LWVjYzEtNDU4MS04Zjc2LTA3YjY3Y2JkMjMwZi5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTEzMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzExMzBUMTAzNDIwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDU3ZTIxMDYzZmVlMDdjZTllYzIzZjg1NjQ4NjcxZjg2OTI1NzY0YTRjZDkzYWIxODEyYzU2NTBjNjUzZmVkNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.4ZAn13Iy8OFOaN6RCuV49pXK6IrjyzagEmgBwnO0Z-k)
> 
> There you can see their logs, and their firmware version is G.3.1. Tell me if it is ok or if i have to tell them to send back the robot.
> 
> Regards

ValentinLeglise, 2023-12-06:
@hilary-luo  Hello. Thanks for your answer.

How to process ? Can you send us a new battery and the instructions to replace it.

Find the shipping adress: 
Name: Julien Amiel
Email adress: julien.amiel@generationrobots.com
Phone number: +33 (0)5 56 39 37 05
Company: Génération Robots
Street: 10 Rue Joule
City: Mérignac
Post Code: 33700
Country: France

If you need broker adress, please let me know.
The customer needs his robot for the begining of the next year, i think its urgent.

Best regards, Valentin.

RustyCPR, 2023-12-07:
Hi Valentin - I replied to you directly from our Support email - lets correspond there. 
Thank you,
Rusty

hilary-luo, 2023-12-17:
Closing since this issue as it is now being handled through Clearpath Support to address the hardware.

closed 2023-12-17 by hilary-luo