turtlebot/turtlebot4 #290: Faulty charging of Turtlebot4 Lite

younesraoui, 2023-10-25:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: installed
 - Package version: ros-2-turtlebot4-package_with_issue

 **Expected behaviour**
I am expecting the Turtlebot4 Lite to shine and the light turns white when I start it. Th
[image]
[image]
is take no more than some minutes. **Actual behaviour**
In some times when I plug the TB4 in the docker, it doen't start and the light remains solid red for a long time. In this case, I am pushed to remove and reinstall the battery because I am unable to turn it off. This happens the first time after the robot turns off suddenly. 

**To Reproduce**
Provide the steps to reproduce:
Start the TB4 Lite by pressing the button in the center of the irobot

**Other notes**

hilary-luo, 2023-10-26:
Can you please try to replicate the situation and then see if you can connect to the webserver and download and share the [log from the create3](https://iroboteducation.github.io/create3_docs/webserver/logs/)?

younesraoui, 2023-10-26:
Hello,
I have attached the log to this message.
Best regards

Oct 26 10:00:15 turtlebot4 syslog.info syslogd started: BusyBox v1.30.1
Oct 26 10:00:15 turtlebot4 kern.notice kernel: klogd started: BusyBox v1.30.1 (2023-04-13 15:02:55 UTC)
Oct 26 10:00:15 turtlebot4 kern.notice kernel: [    0.000000] Linux version 4.14.115-yocto-standard-custom ***@***.***) (gcc version 8.3.0 (GCC)) #1 Thu Apr 13 15:12:11 UTC 2023
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Board has DDR2
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Analog PMU set to hw control
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Digital PMU set to hw control
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] MIPS: machine is iRobot Sundial
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Determined physical RAM map:
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000]  memory: 04000000 @ 00000000 (usable)
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Initial ramdisk at: 0x82d9f000 (2058793 bytes)
Oct 26 10:00:15 turtlebot4 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Oct 26 10:00:15 turtlebot4 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Zone ranges:
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Movable zone start for each node
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Early memory node ranges
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Oct 26 10:00:15 turtlebot4 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Oct 26 10:00:15 turtlebot4 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 80562820, node_mem_map 81000060
Oct 26 10:00:15 turtlebot4 kern.debug kernel: [    0.000000]   Normal zone: 128 pages used for memmap
Oct 26 10:00:15 turtlebot4 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Oct 26 10:00:15 turtlebot4 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Oct 26 10:00:15 turtlebot4 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x47c with crng_init=0
Oct 26 10:00:15 turtlebot4 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Oct 26 10:00:15 turtlebot4 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
Oct 26 10:00:15 turtlebot4 kern.notice kernel: [    0.000000] Kernel command line: bl_version=4.0.3 console=ttyS1,115200 earlyprintk=serial,ttyS1,115200,keep loglevel=3 ubi.mtd=5 ubi.block=0,rootfs root=/dev/ram rw
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Writing ErrCtl register=000234ae
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Readback ErrCtl register=000234ae
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] Memory: 56900K/65536K available (4211K kernel code, 192K rwdata, 1112K rodata, 172K init, 218K bss, 8636K reserved, 0K cma-reserved)
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] NR_IRQS: 256
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] intc: using register map from devicetree
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] CPU Clock: 575MHz
Oct 26 10:00:15 turtlebot4 kern.crit kernel: [    0.000000] timer_probe: no matching timers found
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6647862422 ns
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000012] sched_clock: 32 bits at 287MHz, resolution 3ns, wraps every 7469508094ns
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.000057] Calibrating delay loop... 380.92 BogoMIPS (lpj=1904640)
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.059947] pid_max: default: 32768 minimum: 301
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.060232] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.060248] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Oct 26 10:00:15 turtlebot4 kern.info kernel: [    0.062625] devtmpfs: initialized
[... cut ...]
Oct 26 15:54:50 turtlebot4 user.notice create-platform: [DBG] Refl: enable [Proximity Slowdown]
Oct 26 15:54:50 turtlebot4 user.notice create-platform: [DBG] Refl: Arbiter on
Oct 26 15:54:50 turtlebot4 user.notice create-platform: [WAR] ThermalMonitor - Could not read thermal configuration from /etc/thermal-engine.conf... setting values to default
Oct 26 15:54:50 turtlebot4 user.notice create-platform: VSTAT: No config for vstat_nav_os_temp_c
Oct 26 15:54:50 turtlebot4 user.notice create-platform: VSTAT:

shamlian, 2023-10-30:
I don't see anything obviously wrong in this log. I'm curious to know if you're just seeing the robot in a state for which the battery is low and it is showing you a red light in order to emphasize that it would not be a good idea to run the robot at that state of charge. What is the output of `ros2 topic echo /battery_state` ? It would be good to see if the robot's battery is taking charge or not. Also, what version of Create 3 code is this robot running? Finally, is this robot a single unit, or one of many (and if it's one of many, do all of the robots have different namespaces or domain IDs)?

younesraoui, 2023-11-06:
Thanks for your answer.
This week, the issue has stopped occurring. The first time it occurs is
when the battery is empty. When I attempt to start it, the red light
remains solid. I removed the battery from the robot's back and then
replaced it again. The problem has repeated three times. I attempted to
remove the battery and wait 15 minutes before replacing it. I no longer see
this problem.
How can I see the version of the Create 3 code.
[image: image.png]

hilary-luo, 2023-11-16:
@younesraoui we cannot see the image you sent.

Removing the battery for 15min will reset it and can resolve some types of charging issues so that was a good thing to try. If your issue is resolved you can close this ticket. 

To answer your last question, you can see which version of firmware that the create3 is running from the [webserver](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#accessing-the-create-3-webserver). The home page will display a version number starting with G for Galactic or H for Humble.

closed 2023-11-24 by younesraoui