turtlebot/turtlebot4 #294: Cannot see topics in RPI terminal.

ipa-rar-sy, 2023-10-31:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: (e.g. humble)
 - Built from source or installed: Installed
 

 **Expected behaviour**
Should see topics in the remote pc and raspberrypi after ssh .

 **Actual behaviour**
topics are only visible in remote pc not in raspberrypi.

All leds are turned on. I cannaot see reboot robot option on irobot webpage.

[image]
[image]
[image]

ipa-rar-sy, 2023-11-02:
@hilary-luo  
Thank you for the reply. I have tried ' ros2 daemon stop; ros2 daemon start' but it did not work. I have rebooted the robot manually so many times now but I cannot see the topic as well as the reboot robot option in create3. 
Are there any other solutions to this that you can suggest? 
Thank you.

hilary-luo, 2023-11-02:
Can you share the outputs of `env` and `ip a` from on your pi?

closed 2023-11-03 by ipa-rar-sy