turtlebot/turtlebot4 #295: Oak-D not working

Akalpita30, 2023-10-31:
**Please provide the following information:**
OS: Ubuntu version 22.04
ROS Distro: Humble
Built from source
Package version: H.2.2

 **Expected behaviour**
 I want to visualise the video using Oak-D pro on Turtlebot4 standard.

 **Actual behaviour**
On using 
ros2 launch turtlebot4_bringup oakd.launch.py namespace:=/robot1
And it runs but is seemed to be stuck at the point in image below.
[image]

I also modified the namespace in setup. The picture below shows the parameters. I am not using multiple robots.
[image]

**To Reproduce**
1.ssh to RPI
2.run- ros2 launch turtlebot4_bringup oakd.launch.py namespace:=/robot1

**Other notes**
i also would appreciate help on how can i record the video bit using OAK-D.

hilary-luo, 2023-11-01:
@Akalpita30 The Oakd camera is launched immediately when the robot boots so you should not need to launch it directly. That being said, it has a power saving mode active that turns off the camera while docked. So, you should just need to boot up your robot and undock and then view the camera feed from rviz. If you try to launch the node again while it is already running, it will ultimately cause errors. In order to record the data you can record a rosbag for that particular topic. Specify the topic though so that you don't overload the system / create too large of a bag.

Akalpita30, 2023-11-02:
@hilary-luo i tried running the camera after rebooting and undocking. it still gives the same output.

hilary-luo, 2023-11-02:
@Akalpita30 You should not be launching the camera directly at all, launching the node again when it is already up will cause issues. After you have rebooted and undocked, check for the `oakd/rgb/preview/image_raw` topic and display it in rviz.

Simba185, 2023-11-03:
Please provide direct instructions with command required. 

you state "check for the oakd/rgb/preview/image_raw topic and display it in rviz."

How?

Akalpita30, 2023-11-03:
thank you @Simba185. 
I would appreciate help on converting this recorded rosbag to video format.

hilary-luo, 2023-11-10:
@Akalpita30 unfortunately since these issues are regarding the turtlebot packages we can't really help directly for that. I would suggest reaching out on [robotics stack exchange](https://robotics.stackexchange.com/) or looking up rosbag to video projects through an internet search.

closed 2023-12-17 by hilary-luo