turtlebot/turtlebot4 #297: Unable to get the depth data

vatsuak, 2023-11-02:
**Please provide the following information:**
 - OS: 20.04 (on host) 22 on robot
 - ROS Distro: Galactic ( on host), Humble (on robot)
 - Built from source or installed: installed
 - Package version:1.0.1-1jammy.20231017.132111

 **Expected behaviour**
Expected to receive some data on this topic `ros2 topic echo /stereo/depth`

 **Actual behaviour**
No data is shown
 [image]

I can see the data from `/oakd/rgb/preview/image_raw` topic, so the camera seems to be working.

flymaxty, 2023-11-05:
I got same issue on Humble, I can see color image from /oakd/rgb/preview/image_raw, but nothing from /stereo/depth.
Is there anything missing from my step? Or I misunderstand something?

flymaxty, 2023-11-08:
Hi @roni-kreinin @hilary-luo ,
Is there any solution for this issue? I do also see the same issue on #263 
I'm using `turtlebot4_standard_humble_1.0.0.zip`

vatsuak, 2023-11-11:
Hello, is anyone following up on this? Would appreciate a response.

hilary-luo, 2023-11-16:
@flymaxty @vatsuak /stereo/depth is not the correct topic for stereo data any more (an upstream package was changed). That being said, the stereo node is not launched by default because of the significant load that it puts on the system and not everyone uses it. You are seeing that topic show up mistakenly because the diagnostics node was not updated at one point. Since then the diagnostics package has been updated and as of the next release you won't see this anymore.

For instructions on enabling the stereo node see: https://github.com/turtlebot/turtlebot4/issues/277#issuecomment-1815024838

In regards to https://github.com/turtlebot/turtlebot4/issues/263, as I said on that ticket, I am not aware of anyone having issues with the camera not working, just people misunderstanding how the packages work. If you have the chance then feel free to put in a PR for the documentation to better document how to enable stereo or for the robot packages to make it easier to turn stereo on.

flymaxty, 2023-11-21:
Hi @hilary-luo ,thanks for you help! depth image work fine now.

closed 2023-12-17 by hilary-luo

Ubrillo, 2025-02-17:
How can i get depth from turtlebot4. When i echo this topic:  **/oakd/rgb/preview/image_raw/compressedDepth**  nothing appears.

hilary-luo, 2025-02-18:
@Ubrillo instructions are shared 2 messages up from yours. Follow the link for instructions on enabling the stereo node. If you have issues enabling the node then please open your own ticket instead of responding on someone else's closed ticket. We do not always see or respond to additional comments on closed tickets.