turtlebot/turtlebot4 #300: Cannot create Map or start RVIZ ,Global status: Error Frame [map] dose not exist.

Simba185, 2023-11-03:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: installed

 **Expected behavior**
 run ros2 launch turtlebot4_navigation slam.launch.py
and ros2 launch turtlebot4_viz view_robot.launch.py
and when rviz opens it shows data and position

 **Actual behavior**
 run ros2 launch turtlebot4_navigation slam.launch.py
and ros2 launch turtlebot4_viz view_robot.launch.py
and when rviz opens it shows global error and no data.

**To Reproduce**
Provide the steps to reproduce:
 run ros2 launch turtlebot4_navigation slam.launch.py
and ros2 launch turtlebot4_viz view_robot.launch.py

**Other notes**
I get no data or indication of data when I try and run the NAV to generate a map.  When RViz is opened the global status is Error Frame [map] does not exist.

If I run "
ros2 topic list" on my pc and the TB they match and show /map 

I have rested this and now /map isnt there, why?

[image]
[image]

Simba185, 2023-11-03:
I was able to run ros2 launch turtlebot4_viz view_robot.launch.py 
and add image display, and set topic to /oakd/rgb/preview/image_raw

and i can see data, so I know the TB4 is communicating with ROS on my pc. 

what is causing the global error?

roni-kreinin, 2023-11-03:
From the terminal outputs it looks like the `slam_toolbox` node is not receiving scan data. Are you able to echo the `/scan` topic from your PC?

You can also check for lidar data in rviz. In the top left under 'Global Options' change the fixed frame from `map` to `odom`. This should allow you to visualise the lidar data.

As a side note, your RPi is showing two ethernet IP addresses and no Wifi IP address. Is that intentional?
[image]

Simba185, 2023-11-06:
Hi there Roni-Kreinin,

Thank you for your assistance , yes the ip is intentional (it was connected via wired ethernet at the time, now it is working on 5g wifi ( cant work on 2.4))

I have set the fixed frame to odom and this shows the robot model and the TF but no map or lidar data.

/scan shows data is working from the lidar.

Rviz shows - Map Status: Warn, No map received.

Thanks.

roni-kreinin, 2023-11-06:
It is strange that the lidar data is not showing. Can you expand the RobotModel status error and share that?

Can you also try launching rviz with `ros2 launch turtlebot4_viz view_model.launch.py` and check if the laserscan data shows in that configuration?

Simba185, 2023-11-06:
Robotmodel Error is No TF from Left and right wheel, and drop left and drop right.

running ros2 launch turtlebot4_viz view_model.launch.py is working perfect with no errors.

[image]
[image]

roni-kreinin, 2023-11-07:
Can you access the Create 3 webserver and share the firmware version that you are using, as well as the logs when this issue occurs? I have a suspicion that this is a time synchronisation issue between the Create3 and RPi4.

I would also try configuring the ntp server as mentioned [here](https://github.com/turtlebot/turtlebot4/issues/216) to see if that fixes your issue.

Simba185, 2023-11-07:
I am using H2.3, Attached is the logs.

I don't have access to ntp, but the data and time are set the same.
Results are the same - No Map Received.

roni-kreinin, 2023-11-07:
You can see that the time stamp on the transform is significantly behind the timestamp for the slam toolbox node. That does indicate that the Create 3 time is not synced. 

In the Create 3 logs we can see the message `user.notice ntpd: ntpd: bad address '3.irobot.pool.ntp.org'` which indicates that the Create 3 is still trying to talk to iRobot's ntp server rather than the RPi4. Make sure to restart the ntpd service on the Create 3 through the webserver after saving the ntp config. Or just restart the robot.

Simba185, 2023-11-07:
So I re Tested this after a full reboot of all devices and it seems it is working. I can see the ntp line showing sync from the PI4.

I have map data available now. I will test making a few maps and setting a nav goal.

Simba185, 2023-11-08:
So I was able to get the nav 2 working. it took 1 extra step that is not described in the manual. 
I had to start the Keyboard teleop node to allow the robot to move at all, without this it would send data to the controller but nothing moved.

It also seems very unstable. (the robot stops randomly and rviz seems to stop all data then returns, its like the TB4 freezes) 
In the Create3 logs I can see things like (NTP delay too high, ignoring) and the CPU usage is often 100%

What could be causing this starting and stopping.

roni-kreinin, 2023-11-08:
The first issue might be related to the backup limit that the Create 3 has by default. You can go to the Create 3 webserver and change the [safety override](https://iroboteducation.github.io/create3_docs/webserver/application/#:~:text=minutes%20to%20complete.-,Application%20Configuration,-There%20are%20two) parameter to allow the robot to backup.

I would also recommend you take a look at the [TurtleBot4 Navigator examples](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/turtlebot4_navigator.html). They typically call the undock action before trying to navigate.

The 100% CPU usage is a known issue that is being tracked [here](https://github.com/turtlebot/turtlebot4/issues/266). It occurs when there are a lot of nodes being discovered by the Create 3. I would recommend trying to switch to the discovery server configuration to reduce CPU usage.

Simba185, 2023-11-08:
So I will disable the Backup setting and test.

How would I call the undock command directly? ( I tried "ros2 action send_goal /dock irobot_create_msgs/action/Dock {}" but I get "action type invalid") It does work if I use the "undock" command from the menu.

If nav2 is having these issues what will we see with further development, 3D lidar or stereo vision cameras and more sensors. The system doesn't seem reliable. ( like a robot should be). If Discovery Server is the best method to use with TB4 then the manual should state as such. ( will have to re-setup all networking for Discovery server).  ( also issue 266 seems to point to a modified RMW profile, not discovery server vs simple, and as Hilary states it is still not resolved.)

roni-kreinin, 2023-11-08:
`ros2 action send_goal /dock irobot_create_msgs/action/Dock {}` would be the docking command, and `ros2 action send_goal /undock irobot_create_msgs/action/Undock {}` would be for undocking. If you are sending this from your PC, make sure you have the `irobot_create_msgs` package installed.

Regarding reliability, we are actively working on fixing these issues. Discovery server and simple discovery both have their use cases depending on how the user wants to use their robot. We will work to improve the documentation as there are many possible network configurations and it can get complicated. 

Regarding #266, those RMW profiles are intended to specifically improve simple discovery performance.

closed 2023-12-17 by hilary-luo