turtlebot/turtlebot4 #302: Controller Bluetooth Range extremely low

Simba185, 2023-11-07:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 20.04) Ubuntu 22.02
 - ROS Distro: (e.g. Galactic) Humble
 - Built from source or installed: Installed
 
 **Expected behavior**
Bluetooth controller is able to operate at ~33ft

 **Actual behavior**
 Bluetooth controller is erratic (input locks on or is not responsive) after 10ft.

**To Reproduce**
Provide the steps to reproduce:
Boot the TB4
Drive 10 Ft behind a desk or other object. 
Control is erratic.

**Other notes**
How to Upgrade to Logitech F710 ( used with other Clearpath products, without issues)

roni-kreinin, 2023-11-08:
If you remove the PCB, check that the ribbon cable connecting the RPi to the PCB is not directly over the RPi itself. This can block the signal.

Regarding the F710 controller, TTB4 does not officially support it but it should be fairly easy to use. You will need to replace the turtlebot4 controller parameter file with something like [this](https://github.com/clearpathrobotics/clearpath_common/blob/humble/clearpath_control/config/j100/teleop_logitech.yaml). You will also need to add a udev rule like [this](https://github.com/clearpathrobotics/clearpath_robot/blob/6bd596b635fdfde4a22b51fe1127cea300318041/clearpath_robot/debian/udev#L13) to symlink the device.

adnan-saood, 2023-11-09:
Hello, I am having the same issue with Bluetooth and also WiFi signal, the cable connecting RPi to PCB (the ribbon?) is not looping over the RPi.

Simba185, 2023-11-14:
My cable seems correctly installed but clearly sits above the pi so there is no option to not block some signal.

Bluetooth wont pair after about 6ft and 1 desk in-between. ( Fyi the wifi works best on 5GHZ)

@roni-kreinin are you able to provide steps to change PS4 controller for F710? This would allow for External Receivers that could be mounted on the main PCB or wherever is best for signal.

Thanks,

roni-kreinin, 2023-11-16:
@Simba185 I don't have instructions for this but the general steps would be:

- On the RPi, create a workspace and clone [turtlebot4_robot](https://github.com/turtlebot/turtlebot4_robot).
- Change the contents of [turtlebot4_controller config](https://github.com/turtlebot/turtlebot4_robot/blob/humble/turtlebot4_bringup/config/turtlebot4_controller.config.yaml) to [this](https://github.com/clearpathrobotics/clearpath_common/blob/humble/clearpath_control/config/j100/teleop_logitech.yaml)
- Build the workspace
- Create a udev rule file under `/etc/udev/rules.d/`
- Paste these rules in: 
```# Udev rule for the Logitech controllers
KERNEL=="js*", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c21f", SYMLINK+="input/f710"
KERNEL=="js*", ATTRS{idVendor}=="046d", ATTRS{idProduct}=="c219", SYMLINK+="input/f710"
```
- Trigger the new rule: `sudo udevadm control --reload && sudo udevadm trigger`
- Run the `turtlebot4_setup` tool and in the bash menu change the workspace to the `turtlebot4_robot` workspace you built.
- Save those changes, then use the `turtlebot4_setup` tool to reinstall the `robot_upstart` job.

I haven't tested this so there may be some additional steps required.

closed 2023-12-17 by hilary-luo