turtlebot/turtlebot4 #303: nav2 goal does not work.

gmchoi96, 2023-11-08:
**Please provide the following information:**
 - OS: Ubuntu 22.04)
 - ROS Distro: Humble
 - Built from source or installed: installed
 - Package version: ros2-humble-turtlebot4

hello,

1. I tested 3 Turtlebots, and one Turtlebot does not work as shown in the following video
2. Is there a way to operate faster in a simple discovery environment other than a discovery server environment?

**Other notes**
Explanation for question 1.
* Tried different ways
- Even if you run the nav2 goal, there is no response, and all other python tutorials do not run.
- Wait for 20 minutes
- I tried operating it in simple discovery and discovery server, but it doesn't work in both environments.
- time sync matched
- After initializing the Turtlebot SD card, I downloaded a new image and proceeded with it.
- I proceeded with the create path in the tutorial package, but it did not proceed after Publishing Initial Pose.

Explanation for question 2.
- When all three robots are connected simple discovery, it takes about 2 minute for the launch file to run. If it operates slower, it takes about 5 ~ 7 minutes.
- The issue of slow operation time when connecting to Simple Discovery is the most frequently asked question by everyone testing it.

[image]
[Screencast from 11-02-2023 05_48_04 AM.webm](https://github.com/turtlebot/turtlebot4/assets/149559257/bc11a524-8b35-47c9-a072-daecf63faf2a)

I want to check if the robot I'm testing is defective. The other robots were slow but worked normally.

Thank you

ipa-rar-sy, 2023-11-09:
Are all turtlebots running at the same time? if yes, do they have namespace? Do they have same ROS_DOMAIN_ID?

RustyCPR, 2023-11-14:
Hi @gmchoi96 

1. Can you please replicate and share the [Create3 logs](https://iroboteducation.github.io/create3_docs/webserver/logs/)? 
2. When running do you see all [topics](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_common.html#turtlebot-4-node)? 
3. What are you launching? 
4. What do you mean by time sync is matched? 
5. While recording the video which network configuration were you using - Simple Discovery or Discovery Server?

myamo1, 2023-11-28:
I am having similar issues when trying nav2. I created a map but when trying the navigation, this is what I have
[image]

hilary-luo, 2023-12-17:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

@myamo1 I know you have your issue being tracked in https://github.com/turtlebot/turtlebot4/issues/322

closed 2023-12-17 by hilary-luo