turtlebot/turtlebot4 #304: Basic setup uncomplete

myamo1, 2023-11-10:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Built from source or installed: Installed

 **Expected behaviour**
 Connected and functional remote

 **Actual behaviour**
- The remote is not connected (not paired). After clicking the home button, the white light blinks a few times and disappears.
- Only two LEDs of TB4 are ON (MTR & Power)

**Other notes**
- I am using an access point (2.4 GHz)
- My TB4 is brand new. I followed the basic setup for the first time, and everything worked until I got to the remote, which didn't pair. I suspected that the RPi or Create 3 was not connected. I tried to reaccess the RPi via ethernet/USB and found that the Wi-Fi settings were correct, but now I can't even reaccess the RPi even with ethernet/USB (my PC can't connect to it), and only the LEDs (MTR and power) are turning ON. Is there any way to reset TB4 to the default settings?

smatar22, 2023-11-14:
Hello,
Regarding your two questions:

1. Are you able to pair the controller using the pairing method [outlined in the TB4 user manual]([image])? 
2. Are you able to access the Create3's cargo bay? It should be at the rear of the TB4. Opening the cargo bay should allow you to access the electrical adapter which is used to power and communicate with the Raspberry Pi. Try resetting the connection by first powering down the TB4, unplugging the USB-C cable found connected to the power adapter board, waiting 10-15s then plugging the USB-C cable again. Dock the TB4 again to turn it back on and see if PI comes back on.

Let me know how these recommendations work for you.

myamo1, 2023-11-15:
Hello,

Thanks for your reply.
1- I can't pair the controller since I am unable to ssh to the RPi.
2- I followed your instructions and nothing has changed.

I forgot to mention that when turning TB4 ON, a few seconds after the 2
LEDs (Power and MTR) turn ON, the light of the center button turns red for
about 20-30 then TB4 makes a chiming sound and the light becomes white. But
again the other LEds are off and the screen shows unknown and 0
charging even though it's fully charged.

ipa-rar-sy, 2023-11-15:
Hi, all the things that you have mentioned in your comment happened to my robot as well. But after waiting for a certain time, all LEDs turn On. Regarding the IP address on display stating unknown, have you modified it from default IP adress? If yes then you have to turn DHCP in turtlebot4-setup wifi settings to false first then modify it again. I had same problem and I reflashed my rpi and then first turned dhcp false before changing the IP address.

myamo1, 2023-11-16:
Hello,

Thanks for the recommendations. My problem is almost solved as everything
came back to normal after resetting the RPi's Wi-Fi to the default settings
through HDMI. All the LEDs are working fine, I can pair the remote and the
IP is showing on the screen. My only issue now is that I can't connect the
create 3 to the Wi-Fi network. When I go to the web server and try to
connect to my Wi-Fi, it says connecting but it does not. I waited for the
chiming sound but nothing. Any suggestions?

hilary-luo, 2023-11-16:
@myamo1 try holding down the 1 and 2 buttons on the create3 until you see a spinning blue light. This will put the Create3 into access point mode. Then try again. You should still be able to access the webserver through the raspberry pi or you can use the generated access point.

myamo1, 2023-11-17:
@hilary-luo I tried this before. I think it is connecetd even if it does not the chiming sound since in the webserver it shows an ip adrress which is also shown in the logs of my Wi-Fi. My new issue is when I flashed my RPi, I had to repair the remote controller and it connects for a second and then disconnect. I don't know why. I tried several times but it writes:  "connected: yes" then "connected: no".

hilary-luo, 2023-11-20:
Can you share the [create3 log](https://iroboteducation.github.io/create3_docs/webserver/logs/)? 

Have you tried to [re-pair the controller](https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#turtlebot-4-controller-setup)?

myamo1, 2023-11-21:
I am attaching the Creat3 log.

Yes, I tried to re-pair it several times and the same issue happens.

Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 syslog.info syslogd started: BusyBox v1.30.1
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.notice kernel: klogd started: BusyBox v1.30.1 (2023-09-14 18:29:12 UTC)
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.notice kernel: [    0.000000] Linux version 4.14.115-yocto-standard-custom ***@***.***) (gcc version 8.3.0 (GCC)) #1 Thu Sep 14 18:38:47 UTC 2023
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Board has DDR2
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Analog PMU set to hw control
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Digital PMU set to hw control
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] MIPS: machine is iRobot Sundial
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Determined physical RAM map:
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000]  memory: 04000000 @ 00000000 (usable)
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Initial ramdisk at: 0x82da0000 (2057376 bytes)
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Zone ranges:
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Movable zone start for each node
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Early memory node ranges
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 80562820, node_mem_map 81000060
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.debug kernel: [    0.000000]   Normal zone: 128 pages used for memmap
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x47c with crng_init=0
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.notice kernel: [    0.000000] Kernel command line: bl_version=4.0.3 console=ttyS1,115200 earlyprintk=serial,ttyS1,115200,keep loglevel=3 ubi.mtd=5 ubi.block=0,rootfs root=/dev/ram rw
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Writing ErrCtl register=00045a3b
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Readback ErrCtl register=00045a3b
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] Memory: 56900K/65536K available (4211K kernel code, 192K rwdata, 1112K rodata, 172K init, 218K bss, 8636K reserved, 0K cma-reserved)
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Nov 21 16:00:08 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 kern.info kernel: [    0.000000] NR_IRQS: 256
[... cut ...]
Nov 21 16:00:47 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 daemon.err network[1465]: [ERROR] (src/main.c:203 errno:Resource temporarily unavailable) Error publishing sta:0 FIRR request.
Nov 21 16:00:47 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 user.notice create-platform: VSTAT: No config for vstat_nav_rosKinemEngineTimer_dur_us
Nov 21 16:00:47 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 user.notice create-platform: VSTAT: No config for vstat_nav_rosKinemEngineTimer_int_ms
Nov 21 16:00:47 iRobot-6BCF045C0C114A11A2190EB8AE258AE1 user.notice create-platform: VSTAT: No conf

myamo1, 2023-11-26:
My controller finally got connected by following the steps in this link https://salamwaddah.com/blog/connecting-ps4-controller-to-raspberry-pi-via-bluetooth. I think that the only difference from the user-guide is that we first connect then trust.

closed 2023-11-26 by myamo1