turtlebot/turtlebot4 #307: "Some messages are lost" warning showing in the camera topic in rviz - Camera USB cable

Vasanth28897, 2023-11-16:
I received the turtlebot4 with the malfunctioned cable for the oakd-camera. After knowing that,  I contacted support, they suggested the cable with version 2.0, but the minimum version shall be 3.0 mentioned in [this](https://github.com/turtlebot/turtlebot4-hardware/blob/master/TurtleBot%204/Components/USB%20Cables/025943%20-%20Cable%2C%20USB%203.0%20USB-B%2090%C2%B0%20Male%20to%20USB-A%20Male%2C%20200mm/025943_1.pdf) document. I did not make sure that's the required one before i ordered the cable. It almost took 20 days for getting the wire from the day we ordered. Now we received the cable. Plugged in the camera and RPI in the turtlebot4. when i run the `ros2 launch turtlebot4_viz view_model.launch.py ` the camera topic changes from warning and ok status repeatedly. But i can see the image, anyhow it freezes when it shows warning "some messages are lost". This happening because of the cable is not meet the required version? I searched all over the internet, the exact one i can't find. Can you guys help with this one?

RustyCPR, 2023-11-16:
Hi @Vasanth28897 
Thanks for logging this issue. 
Are you running/ powering anything else from USB other than camera and lidar?

Vasanth28897, 2023-11-17:
No, It's solely running on the USB cable... this is happening because of the usb cable is not at rated speed, which is version2.0 - 480MBps..but as per the document mentioned in the question, it shall be minimum 3.0.

roni-kreinin, 2023-11-17:
@Vasanth28897 Could you share the output of `lsusb -t` on the RPi with the camera connected? Also is the RPi connected to a 5GHz network or 2.4GHz?

Vasanth28897, 2023-11-18:
here's the output of lsusb -t on the RPI with camera cable connected(the default cable comes with the turtlebot4). 
```
ubuntu@ubuntu:~$ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 9, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
        |__ Port 4: Dev 8, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M

```

The RPI is connected to a 5GHz network . And when i plug the camera usb cable into the rpi, the /scan topic did not publish any messages(using echo /scan command) and there is no scan points in the rviz(image attached below)
[image]

roni-kreinin, 2023-11-20:
@Vasanth28897 From the `lsusb` output we can see that the camera is running on USB 3.0 (`Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M`). Given that you are receiving images, I suspect that the issue is with networking. I would recommend you try connecting your PC to the Rpi4 directly with an ethernet cable and see if performance improves.

Regarding the lidar issue, can you share the logs from the turtlebot4 service? `sudo journalctl -u turtlebot4 -r`

Vasanth28897, 2023-11-23:
@roni-kreinin  sorry for the delay, we were testing. Since your claim is on the networking issue, we connected our pc and the router directly with ethernet cable, and the turtlebot4 connected with the wifi(from the router), i cannot see the "warning" topic in the rviz and the frequency rate of the topic also as 30. 
```
ros2 topic hz /oakd/rgb/preview/image_raw
WARNING: topic [/oakd/rgb/preview/image_raw] does not appear to be published yet
average rate: 30.246
	min: 0.027s max: 0.036s std dev: 0.00159s window: 32
average rate: 30.127
	min: 0.017s max: 0.057s std dev: 0.00397s window: 63
average rate: 30.065
	min: 0.011s max: 0.057s std dev: 0.00465s window: 94
average rate: 30.059
	min: 0.011s max: 0.057s std dev: 0.00409s window: 125
average rate: 29.959
	min: 0.011s max: 0.057s std dev: 0.00422s window: 155
average rate: 30.039
	min: 0.011s max: 0.057s std dev: 0.00461s window: 186
```
and also we tested when generating a map using `slam` and set the goal using `nav2` all the time the camera topic is enabled in the rviz. we cannot get a warning message now like when we received before. So as i mentioned above how the connection should be, the data is transmitting without any loss. You were right, the issue is in the networking.
So, even tho the cable(we bought from amazon version-2.0 - the support team suggested) is not the required version(3.0), Its working good.
And you mentioned the wrong cable tho...here's the output of `lsusb -t `
when cable plugged:
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
        |__ Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M
```
when cable unplugged:
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M
```

closed 2023-11-23 by Vasanth28897