turtlebot/turtlebot4 #311: Failed to subscribe topic after RasPi4 reboot

atinfinity, 2023-11-27:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 22.04)
 - ROS Distro: (e.g. Humble)
 - Built from source or installed: installed
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Expected behaviour**

After user reboot RasPi4, user run `ros2 topic echo /odom`.
I expect that user can subscribe `/odom` topic.

 **Actual behaviour**

After user reboot RasPi4, user run `ros2 topic echo /odom`.
I expect that user can subscribe `/odom` topic.

```shell
$ ros2 topic echo /odom
WARNING: topic [/odom] does not appear to be published yet
Could not determine the type for the passed topic
```

**To Reproduce**

1. boot TurtleBot4
2. reboot RasPi4
3. do `ros2 topic echo /odom`

**Other notes**

### reboot RasPi4

|command|result|
|:---|---|
|ros2 topic echo /ip | OK |
| ros2 topic echo /scan | OK |
| ros2 topic echo /oakd/rgb/preview/image_raw | OK |
| ros2 topic echo /imu | NG |
| ros2 topic echo /dock_status | NG |
| ros2 topic echo /battery_state | NG |
| ros2 topic echo /hazard_detection | NG |

```shell
ros2 topic echo /odom
WARNING: topic [/odom] does not appear to be published yet
Could not determine the type for the passed topic
```

```shell
$ ros2 node list
/analyzers
/joint_state_publisher
/joy_linux_node
/launch_ros_1022
/oakd
/oakd_container
/robot_state_publisher
/rplidar_composition
/teleop_twist_joy_node
/turtlebot4_base_node
/turtlebot4_diagnostics
/turtlebot4_nod
```

```shell
$ ros2 topic list
/battery_state
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/interface_buttons
/ip
/joy
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/parameter_events
/rosout
/scan
/wheel_status
```

### reboot TurtleBot4 via Web browser

|command|result|
|:---|---|
| ros2 topic echo /ip | OK |
| ros2 topic echo /scan | OK |
| ros2 topic echo /oakd/rgb/preview/image_raw | OK |
| ros2 topic echo /imu | OK |
| ros2 topic echo /dock_status | OK |
| ros2 topic echo /battery_state | OK |
| ros2 topic echo /hazard_detection | OK |
| ros2 topic echo /odom | OK |

```shell
$ ros2 node list
/_internal/composite_hazard
/_internal/kinematics_engine
/_internal/mobility
/_internal/stasis
/analyzers
/joint_state_publisher
/joy_linux_node
/launch_ros_1022
/mobility_monitor
/motion_control
/oakd
/oakd_container
/robot_state
/robot_state_publisher
/rplidar_composition
/static_transform
/system_health
/teleop_twist_joy_node
/turtlebot4_base_node
/turtlebot4_diagnostics
/turtlebot4_node
/ui_mgr
```

```shell
$ ros2 topic list
/battery_state
/cliff_intensity
/cmd_audio
/cmd_lightring
/cmd_vel
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock_status
/function_calls
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/joy/set_feedback
/kidnap_status
/mobility_monitor/transition_event
/mouse
/oakd/rgb/preview/camera_info
/oakd/rgb/preview/image_raw
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels
```

atinfinity, 2023-11-27:
I read <https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_robot.html#robot-upstart> and I checked the status of `Robot Upstart`.

## reboot RasPi4

```
● turtlebot4.service - "bringup turtlebot4"
     Loaded: loaded (/lib/systemd/system/turtlebot4.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-09-19 16:59:51 UTC; 2 months 7 days ago
   Main PID: 747 (turtlebot4-star)
      Tasks: 134 (limit: 4423)
     Memory: 363.9M
        CPU: 5min 58.120s
     CGroup: /system.slice/turtlebot4.service
             ├─ 747 /bin/bash /usr/sbin/turtlebot4-start
             ├─1019 /usr/bin/python3 /opt/ros/humble/bin/ros2 launch /tmp/turtlebot4.launch.py
             ├─1035 /opt/ros/humble/lib/turtlebot4_node/turtlebot4_node --ros-args -r __ns:=/ --params-file /tmp/tmp0_5ln1xe --params-file /tmp/launch_params_j0_i5zyh
             ├─1037 /opt/ros/humble/lib/turtlebot4_base/turtlebot4_base_node --ros-args -r __ns:=/ --params-file /tmp/tmp0_5ln1xe
             ├─1039 /opt/ros/humble/lib/joy_linux/joy_linux_node --ros-args -r __node:=joy_linux_node -r __ns:=/ -r /diagnostics:=diagnostics
             ├─1041 /opt/ros/humble/lib/teleop_twist_joy/teleop_node --ros-args -r __node:=teleop_twist_joy_node -r __ns:=/ --params-file /tmp/tmpmy9dpbb1
             ├─1043 /opt/ros/humble/lib/rplidar_ros/rplidar_composition --ros-args -r __node:=rplidar_composition -r __ns:=/ --params-file /tmp/launch_params_8o8l5qyj
             ├─1045 /opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=oakd_container -r __ns:=/
             ├─1047 /opt/ros/humble/lib/robot_state_publisher/robot_state_publisher --ros-args -r __node:=robot_state_publisher -r __ns:=/ --params-file /tmp/launch_params_u30aby0o --params-file /tmp/launch>
             ├─1049 /usr/bin/python3 /opt/ros/humble/lib/joint_state_publisher/joint_state_publisher --ros-args -r __node:=joint_state_publisher -r __ns:=/ --params-file /tmp/launch_params_gq9ksex1 -r /tf:=>
             ├─1051 /opt/ros/humble/lib/diagnostic_aggregator/aggregator_node --ros-args -r __ns:=/ --params-file /tmp/tmph6rmoy0_ -r /diagnostics:=diagnostics -r /diagnostics_agg:=diagnostics_agg -r /diagn>
             └─1053 /usr/bin/python3 /opt/ros/humble/lib/turtlebot4_diagnostics/diagnostics_updater --ros-args -r __ns:=/ -r /diagnostics:=diagnostics -r /diagnostics_agg:=diagnostics_agg -r /diagnostics_to>

Nov 27 04:36:06 ubuntu turtlebot4-start[1019]: [rplidar_composition-5] [INFO] [1701059766.238966026] [rplidar_composition]: Start
Nov 27 04:36:06 ubuntu turtlebot4-start[1019]: [rplidar_composition-5] [INFO] [1701059766.935511386] [rplidar_composition]: current scan mode: Sensitivity, sample rate: 8 Khz, max_distance: 12.0 m, scan fre>
Nov 27 04:36:08 ubuntu turtlebot4-start[1019]: [component_container-6] [INFO] [1701059768.599553676] [oakd]: Camera with MXID: 18443010E13B990F00 and Name: 1.1.2 connected!
Nov 27 04:36:08 ubuntu turtlebot4-start[1019]: [component_container-6] [INFO] [1701059768.601324354] [oakd]: USB SPEED: SUPER
Nov 27 04:36:08 ubuntu turtlebot4-start[1019]: [component_container-6] [INFO] [1701059768.623850247] [oakd]: Device type: OAK-D-PRO-AF
Nov 27 04:36:08 ubuntu turtlebot4-start[1019]: [component_container-6] [INFO] [1701059768.629575941] [oakd]: Pipeline type: RGB
Nov 27 04:36:08 ubuntu turtlebot4-start[1019]: [component_container-6] [WARN] [1701059768.670707061] [oakd]: Resolution 1080 not supported by sensor IMX378. Using default resolution 1080P
Nov 27 04:36:08 ubuntu turtlebot4-start[1019]: [component_container-6] [INFO] [1701059768.686859139] [oakd]: Finished setting up pipeline.
Nov 27 04:36:08 ubuntu turtlebot4-start[1019]: [component_container-6] [INFO] [1701059768.908833242] [oakd]: Camera ready!
Nov 27 04:36:08 ubuntu turtlebot4-start[1019]: [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oakd' in container '/oakd_container'
```

## reboot TurtleBot4 via Web browser

```
● turtlebot4.service - "bringup turtlebot4"
     Loaded: loaded (/lib/systemd/system/turtlebot4.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-09-19 16:59:51 UTC; 2 months 7 days ago
   Main PID: 747 (turtlebot4-star)
      Tasks: 125 (limit: 4423)
     Memory: 380.4M
        CPU: 10min 30.829s
     CGroup: /system.slice/turtlebot4.service
             ├─ 747 /bin/bash /usr/sbin/turtlebot4-start
             ├─1019 /usr/bin/python3 /opt/ros/humble/bin/ros2 launch /tmp/turtlebot4.launch.py
             ├─1035 /opt/ros/humble/lib/turtlebot4_node/turtlebot4_node --ros-args -r __ns:=/ --params-file /tmp/tmp0_5ln1xe --params-file /tmp/launch_params_j0_i5zyh
             ├─1037 /opt/ros/humble/lib/turtlebot4_base/turtlebot4_base_node --ros-args -r __ns:=/ --params-file /tmp/tmp0_5ln1xe
             ├─1039 /opt/ros/humble/lib/joy_linux/joy_linux_node --ros-args -r __node:=joy_linux_node -r __ns:=/ -r /diagnostics:=diagnostics
             ├─1041 /opt/ros/humble/lib/teleop_twist_joy/teleop_node --ros-args -r __node:=teleop_twist_joy_node -r __ns:=/ --params-file /tmp/tmpmy9dpbb1
             ├─1043 /opt/ros/humble/lib/rplidar_ros/rplidar_composition --ros-args -r __node:=rplidar_composition -r __ns:=/ --params-file /tmp/launch_params_8o8l5qyj
             ├─1045 /opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=oakd_container -r __ns:=/
[... cut ...]
Nov 27 04:51:04 ubuntu turtlebot4-start[1019]: [rplidar_composition-5] [INFO] [1701060664.341978942] [rplidar_composition]: Ingnoring stop_motor request because rplidar_node is in 'auto standby' mode
Nov 27 04:51:04 ubuntu turtlebot4-start[1019]: [turtlebot4_node-1] [INFO] [1701060664.344479795] [turtlebot4_node]: stop_motor service completed.
Nov 27 04:51:05 ubuntu turtlebot4-start[1019]: [turtlebot4_node-1] [INFO] [1701060665.767139872] [turtlebot4_node]: oakd/stop_camera service got results: Success
Nov 27 04:51:05 ubuntu turtlebot4-start[1019]: [turtlebot4_node-1] [INFO] [1701060665.767294592] [turtlebot4_node]:
```

atinfinity, 2023-12-12:
@hilary-luo There is no response. Could you give me your comment?

hilary-luo, 2023-12-13:
Can you please provide the full create3 log?

atinfinity, 2023-12-14:
@hilary-luo I encountered other problem(<https://github.com/turtlebot/turtlebot4/issues/325>). So, it is difficult to get a log of this issue.

atinfinity, 2023-12-15:
@hilary-luo https://github.com/turtlebot/turtlebot4/issues/325 was resolved. And, I reproduced this problem. So, I collect iRobot Create3 log via Web server.
I power off Create 3 and power on Create 3. This problem was resolved.

### iRobot Create3

```
Dec 15 00:50:54 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [WARN] [1702601454.015152973] [motion_control]: Disabling reflexes
Dec 15 00:50:54 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601454.019990063] [mobility_monitor]: New kidnap status: kidnapped
Dec 15 00:50:54 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601454.022490454] [_internal.mobility]: Received enable wheels request
Dec 15 00:50:54 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601454.023570064] [mobility_monitor]: Requested to disable wheels!
Dec 15 00:50:54 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601454.048117261] [mobility_monitor]: Updating ready to drive: false
Dec 15 00:50:54 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [WARN] [1702601454.048539998] [mobility_monitor]: Robot is kidnapped! Temporarily disabling wheels!
Dec 15 00:50:54 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601454.769931909] [motion_control]: Kidnap status changed: temporarily enabling reflexes
Dec 15 00:50:54 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [WARN] [1702601454.770921060] [motion_control]: Enabling reflexes
Dec 15 00:50:54 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601454.774118227] [mobility_monitor]: New kidnap status: kidnap cleared
Dec 15 00:50:55 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601455.376746039] [_internal.mobility]: Received enable wheels request
Dec 15 00:50:55 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601455.377904598] [mobility_monitor]: Requested to enable wheels!
Dec 15 00:50:55 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601455.406697569] [mobility_monitor]: Updating ready to drive: true
Dec 15 00:50:55 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601455.407116545] [mobility_monitor]: Kidnap cleared! Wheels are now enabled!
Dec 15 00:50:55 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 32s
Dec 15 00:51:08 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601468.099074811] [system_health]: CPU usage: max 78 [%] mean 69 [%] RAM usage: 35/59 [MB]
Dec 15 00:51:44 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 64s
Dec 15 00:52:08 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601528.092899580] [system_health]: CPU usage: max 86 [%] mean 76 [%] RAM usage: 35/59 [MB]
Dec 15 00:52:58 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 kern.info kernel: [  894.938863] usb 1-1: USB disconnect, device number 2
Dec 15 00:52:58 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 kern.info kernel: [  894.947706] cdc_ether 1-1:1.0 usb0: unregister 'cdc_ether' usb-101c0000.ehci-1, CDC Ethernet Device
Dec 15 00:52:59 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 daemon.info avahi-daemon[1217]: Interface usb0.IPv4 no longer relevant for mDNS.
Dec 15 00:52:59 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 daemon.info avahi-daemon[1217]: Leaving mDNS multicast group on interface usb0.IPv4 with address 192.168.186.2.
Dec 15 00:52:59 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 daemon.info avahi-daemon[1217]: Withdrawing address record for 192.168.186.2 on usb0.
Dec 15 00:53:05 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice ntpd: ntpd: timed out waiting for 192.168.186.1, reach 0x00, next query in 66s
Dec 15 00:53:08 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702601588.089692347] [system_health]: CPU usage: max 99 [%] mean 69 [%] RAM usage: 35/59 [MB]
Dec 15 00:53:11 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 kern.info kernel: [  908.601320] usb 1-1: new high-speed USB device number 3 using ehci-platform
Dec 15 00:53:17 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 kern.err kernel: [  914.033931] usb 1-1: device descriptor read/64, error -71
Dec 15 00:53:17 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 kern.info kernel: [  914.354720] usb 1-1: New USB device found, idVendor=0525, idProduct=a4a2
Dec 15 00:53:17 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 kern.info kernel: [  914.354742] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Dec 15 00:53:17 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 kern.info kernel: [  914.354754] usb 1-1: Product: RNDIS/Ethernet Gadget
Dec 15 00:53:17 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 kern.info kernel: [  914.354764] usb 1-1: Manufacturer: Linux 5.15.0-1044-raspi with fe980000.usb
Dec 15 00:53:17 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 kern.info kernel: [  914.359457] cdc_ether 1-1:1.0 usb0: register 'cdc_ether' at usb-101c0000.ehci-1, CDC Ethernet Device, 22:d8:59:48:d2:0d
Dec 15 00:53:17 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 kern.err kernel: [  914.694021] cdc_ether 1-1:1.0 usb0: kevent 12 may have been dropped
[... cut ...]
Dec 15 01:00:08 iRobot-717F1994F3A3457ABAF0EBF2CE9DB939 user.notice create-platform: [INFO] [1702602008.090496034] [system_health]: CPU usage: max 53 [%] mean 45 [%] RAM usage: 35/59 [MB]
```

### RasPi4

[... 24 lines cut ...]
/oakd/rgb/preview/image_raw
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static
/wheel_status
```

```
$ ros2 node list
/analyzers
/joint_state_publisher
/joy_linux_node
/launch_ros_963
/oakd
/oakd_container
/robot_state_publisher
/rplidar_composition
/teleop_twist_joy_node
/turtlebot4_base_node
/turtlebot4_diagnostics
/turtlebot4_node
```

hilary-luo, 2023-12-17:
My understanding then is that this issue is also now solved. There was a networking issue between the raspberry pi and the create3 that seems to have resolved itself.

atinfinity, 2023-12-18:
@hilary-luo 

> My understanding then is that this issue is also now solved. 

No. 

> Failed to subscribe topic after RasPi4 reboot 

This problem is NOT resolved.
<https://github.com/turtlebot/turtlebot4/issues/311#issuecomment-1857122945>

1. power on TurtleBot4
2. reboot RasPi4
3. do `ros2 topic echo /odom`. But, I could not subscribe odom topic.

Is this expected behavior?

hilary-luo, 2023-12-22:
@atinfinity my apologies, when you said "this problem is resolved" I thought you were referring to this ticket.

It does seem like you have a networking issue between the create3 and the raspberry pi. Are you using simple discovery or discovery server?

That doesn't seem like the entire create3 log, it would be more helpful if you could provide the full log.

hilary-luo, 2023-12-22:
Based on your other ticket it would seem that you are using discovery server. Can you please verify that the create3 webserver shows that it is set to discovery server and is showing the correct IP address and port for the discovery server? While you are there please also verify that the rmw implementation is set to fastrtps and that the ros domain ID is correct for your robot. Still please provide the full create3 log.

atinfinity, 2023-12-22:
> Based on your other ticket it would seem that you are using discovery server.

Yes. 

> Can you please verify that the create3 webserver shows that it is set to discovery server and is showing the correct IP address and port for the discovery server? While you are there please also [...]

I'll provide other information later. Please wait a little.

atinfinity, 2023-12-25:
> Can you please verify that the create3 webserver shows that it is set to discovery server and is showing the correct IP address and port for the discovery server? While you are there please also [...]

I checked create3 webserver. As a result, there is no problem.

[image]

And, I reboot RasPi4. I could subscribe `odom` topic from RasPi4. So, I closed this issue.
But, I could NOT subscribe `odom` topic from Other PC(NOT RasPi4). I will discuss this point in another ticket(<https://github.com/turtlebot/turtlebot4/issues/329>).

closed 2023-12-25 by atinfinity