turtlebot/turtlebot4 #314: RPi Does not connect to Wi-Fi from the first shot

myamo1, 2023-12-01:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Installed:

 **Expected behaviour**
 Turn on turtlebot4 and RPi connects to Wi-Fi (screen shows IP address).
 **Actual behaviour**
 The screen shows unknown instead of IP address. Generally, after the 5th trial, RPi connects to Wi-Fi.

**To Reproduce**

**Other notes**
-I reflashed the RPi, changed another Wi-Fi but always the same behaviour.
-If I connect RPi to amonitor via a (usb-HDMI)-cable, the RPi connects.
-Screen shows IP address on Access point mode.

hilary-luo, 2023-12-11:
@myamo1 Are you connecting to a 2.4GHz network or 5GHz?

myamo1, 2023-12-18:
I am connecting to 2.4GHz network

hilary-luo, 2024-02-13:
@myamo1 apologies that I missed seeing your response earlier. Are you still experiencing this issue?

hilary-luo, 2024-02-20:
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.

myamo1, 2024-03-06:
Hello again,
I am still having the same issue. Few days it was connecting from the first shot but yesterday night it did not work even after several trials. This morning, it worked from the second shot.

I am using a router connected to the university network.

hilary-luo, 2024-03-14:
Does that router support 5 GHz? if so I would suggest trying to put the Raspberry Pi on the 5GHz network. 

Another thing that I would suggest is to reserve an IP address for the Raspberry Pi on the router. This seems like it is an issue with the network configuration so that could potentially help.

If the issue persists then connect to the pi when you are experiencing the issue and share the output of `journalctl -b | grep wlan0`

myamo1, 2024-03-22:
The router does not support 5 GHz.
I reserved an IP address for the RPi but I experienced the same issue.
Here is the output you are asking for:

Nov 21 20:59:33 ubuntu cloud-init[657]: ci-info: | wlan0  | False |              .               |       .       |   .    | d8:3a:dd:08:60:35 |
Nov 21 20:59:36 ubuntu NetworkManager[663]: <info>  [1700600376.0408] device (wlan0): driver supports Access Point (AP) mode
Nov 21 20:59:36 ubuntu NetworkManager[663]: <info>  [1700600376.0436] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/4)
Nov 21 20:59:36 ubuntu NetworkManager[663]: <info>  [1700600376.0447] device (wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Nov 21 20:59:36 ubuntu systemd-networkd[650]: wlan0: Link UP
Nov 21 20:59:36 ubuntu NetworkManager[663]: <info>  [1700600376.5015] device (wlan0): supplicant interface state: internal-starting -> disconnected
Nov 21 20:59:36 ubuntu NetworkManager[663]: <info>  [1700600376.5019] Wi-Fi P2P device controlled by interface wlan0 created
Nov 21 20:59:36 ubuntu NetworkManager[663]: <info>  [1700600376.5026] manager: (p2p-dev-wlan0): new 802.11 Wi-Fi P2P device (/org/freedesktop/NetworkManager/Devices/5)
Nov 21 20:59:36 ubuntu NetworkManager[663]: <info>  [1700600376.5033] device (p2p-dev-wlan0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Nov 21 20:59:36 ubuntu NetworkManager[663]: <info>  [1700600376.5044] device (wlan0): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
Nov 21 20:59:36 ubuntu NetworkManager[663]: <info>  [1700600376.5062] device (p2p-dev-wlan0): state change: unavailable -> disconnected (reason 'none', sys-iface-state: 'managed')
Nov 21 20:59:38 ubuntu networkd-dispatcher[672]: ERROR:Unable to retrieve ESSID for wireless interface wlan0: no supported wireless tool installed
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2446] policy: auto-activating connection 'netplan-wlan0-Lab1012' (38462bcd-a2f5-3da7-a029-d760d96dcf66)
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2488] device (wlan0): Activation: starting connection 'netplan-wlan0-Lab1012' (38462bcd-a2f5-3da7-a029-d760d96dcf66)
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2500] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2538] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2566] device (wlan0): Activation: (wifi) access point 'netplan-wlan0-Lab1012' has security, but secrets are required.
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2569] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2575] sup-iface[c1c960adb145e673,0,wlan0]: wps: type pbc start...
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2599] device (wlan0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2609] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2623] device (wlan0): Activation: (wifi) connection 'netplan-wlan0-Lab1012' has security, and secrets exist.  No new secrets needed.
Nov 21 20:59:39 ubuntu wpa_supplicant[682]: wlan0: WPS-CANCEL
Nov 21 20:59:39 ubuntu wpa_supplicant[682]: wlan0: Trying to associate with SSID 'Lab1012'
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2986] device (wlan0): supplicant interface state: disconnected -> associating
Nov 21 20:59:39 ubuntu NetworkManager[663]: <info>  [1700600379.2996] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> associating
Nov 21 20:59:42 ubuntu wpa_supplicant[682]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Nov 21 20:59:42 ubuntu NetworkManager[663]: <info>  [1700600382.3160] device (wlan0): supplicant interface state: associating -> disconnected
Nov 21 20:59:42 ubuntu NetworkManager[663]: <info>  [1700600382.3162] device (p2p-dev-wlan0): supplicant management interface state: associating -> disconnected
Nov 21 20:59:42 ubuntu NetworkManager[663]: <info>  [1700600382.4142] device (wlan0): supplicant interface state: disconnected -> scanning
Nov 21 20:59:42 ubuntu NetworkManager[663]: <info>  [1700600382.4145] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Nov 21 20:59:45 ubuntu wpa_supplicant[682]: wlan0: Trying to associate with SSID 'Lab1012'
Nov 21 20:59:45 ubuntu NetworkManager[663]: <info>  [1700600385.0986] device (wlan0): supplicant interface state: scanning -> associating
Nov 21 20:59:45 ubuntu NetworkManager[663]: <info>  [1700600385.0989] device (p2p-dev-wlan0): supplicant management interface state: scanning -> associating
Nov 21 20:59:47 ubuntu wpa_supplicant[682]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Nov 21 20:59:47 ubuntu NetworkManager[663]: <info>  [1700600387.7670] device (wlan0): supplicant interface state: associating -> disconnected
[... cut ...]
Nov 21 21:00:54 ubuntu NetworkManager[663]: <warn>  [1700600454.9285] device (wlan0): Activation: failed for connection 'netplan-wlan0-Lab1012'
Nov 21 21:00:54 ubuntu NetworkManager[663]: <info>  [1700600454.9301] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Nov 21 21:01:01 ubuntu NetworkManager[663]: <info>  [1700600461.2041] device (wlan0): supplicant interface state: disconnected -> inactive
Nov 21 21:01:01 ubuntu NetworkManager[663]: <info>  [1700600461.2043] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> inactive

hilary-luo, 2024-03-25:
The repeated error seems to be this one: 

> CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
> wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="Lab1012" auth_failures=1 duration=10 reason=CONN_FAILED

I have not directly had experience with this type of failure but through some searching people seem to run into this issue with raspberry pi's in several different types of scenarios including incorrect bands / channels and general interference from other physical connections.

First thing to check is when you run the `turtlebot4-setup` tool, in the `Wi-Fi Settings` do you have the band set to 2.4 GHz? (or Any). 

If not that, some people have reported incompatibilities with the raspberry pi WIFi driver and specific access points. The solution people seem to have is to disable the fast-roam feature on the pi. See https://bugs.launchpad.net/raspbian/+bug/1929746 and https://raspberrypi.stackexchange.com/questions/77144/rpi3-wireless-issue-ctrl-event-assoc-reject-status-code-16

If that still doesn't work, try searching up that specific error, as people have reported different solutions for it. This is not a Turtlebot4 specific error but a general raspberry pi issue that seems to apply to certain routers. If you are still unable to get it connected, it may be worth posting this issue into a raspberry pi forum and see if the folks there can help further. Alternatively, you could also try a different type of router.