turtlebot/turtlebot4 #316: turtlebot4 not showing up on Rviz and not connecting to wifi

Akalpita30, 2023-12-05:
Please provide the following information:
OS: Ubuntu version 22.04
ROS Distro: Humble
Built from source
Package version: H.2.2

Expected behaviour
Turtlebot4 shows up on rviz for creating a new map.

Actual behaviour
on running command-ros2 launch turtlebot4_viz view_robot.launch.py, ros2 launch turtlebot4_viz view_model.launch.py  : the bot doesnt show up on the window i tried running the other command too the results were same. on checking the wi-fi connection the irobot doesn't show connected. i tried to reboot the RPi and unplug it for a day, charging the robot, changing the battery but it doesn't change anything. i tried to connect to the icreate logs for ntp configs but was unable to connect to the server.
This is the screenshot of the rviz window,
[image]

Turtlebot4 isn't showing up on the wi-fi clients list.
[image]

Screenshot from icreate server
[image]

After running diagnostics, it shows a lot of hazards
[image]

screenshot from terminals
ros2 launch turtlebot4_navigation slam.launch.py
[image]

ros2 launch turtlebot4_viz view_robot.launch.py
[image]

hilary-luo, 2023-12-11:
@Akalpita30 when you say that the turtlebot is not showing as connected, do you mean the create3 base or the raspberry pi? In your wifi client list one does show up as 'ubuntu' which I would assume is the raspberry pi.

Can you ssh into the robot and/or are you receiving any topics? It sounds like you have networking issues that need to be resolved first before you can run any navigation / rviz windows.

MikitaSaladukha, 2024-05-23:
The above problem seems to appear after I've tried to switch to lite model of robot