turtlebot/turtlebot4 #318: It seems that the rplidar and create3 times do not match. local_costmap is not visible.rplidar

gmchoi96, 2023-12-06:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: humble
 - H.2.3 펌웨어
 - FastDDS

 **Expected behaviour**

When running nav2, local_costmap should appear so turtlebot can avoid objects.

 **Actual behaviour**

When running nav2, global_costmap can be moved, but local_hostmap does not appear. 2 out of 5 robots. 
This symptom is very important, and it is highly likely that it will also appear in my other Turtlebots, so I must provide a solution.

**To Reproduce**

Provide the steps to reproduce:

The commands executed are as follows.

ros2 launch turtlebot4_navigation localization.launch.py map:=test1.yaml
ros2 launch turtlebot4_navigation nav2.launch.py 
ros2 launch turtlebot4_viz view_robot.launch.py
-----------  terminal  -------------------
$ ros2 launch turtlebot4_navigation localization.launch.py map:=test1.yaml
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-12-06-04-40-15-733801-ubuntu-1660
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [1661]
[INFO] [amcl-2]: process started with pid [1663]
[INFO] [lifecycle_manager-3]: process started with pid [1665]
[lifecycle_manager-3] [INFO] [1701837616.803464953] [lifecycle_manager_localization]: Creating
[lifecycle_manager-3] [INFO] [1701837616.837421715] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[amcl-2] [INFO] [1701837616.889747046] [amcl]: 
[amcl-2] 	amcl lifecycle node launched. 
[amcl-2] 	Waiting on external lifecycle transitions to activate
[amcl-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[amcl-2] [INFO] [1701837616.910092204] [amcl]: Creating
[map_server-1] [INFO] [1701837616.964879525] [map_server]: 
[map_server-1] 	map_server lifecycle node launched. 
[map_server-1] 	Waiting on external lifecycle transitions to activate
[map_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1701837616.965506111] [map_server]: Creating
[lifecycle_manager-3] [INFO] [1701837618.851586001] [lifecycle_manager_localization]: Waiting for service map_server/get_state...
[lifecycle_manager-3] [INFO] [1701837620.852035035] [lifecycle_manager_localization]: Waiting for service map_server/get_state...
[lifecycle_manager-3] [INFO] [1701837620.877623041] [lifecycle_manager_localization]: Starting managed nodes bringup...
[lifecycle_manager-3] [INFO] [1701837620.877796742] [lifecycle_manager_localization]: Configuring map_server
[map_server-1] [INFO] [1701837620.879081154] [map_server]: Configuring
[map_server-1] [INFO] [map_io]: Loading yaml file: test1.yaml
[map_server-1] [DEBUG] [map_io]: resolution: 0.05
[map_server-1] [DEBUG] [map_io]: origin[0]: -6.64
[map_server-1] [DEBUG] [map_io]: origin[1]: -6.26
[map_server-1] [DEBUG] [map_io]: origin[2]: 0
[map_server-1] [DEBUG] [map_io]: free_thresh: 0.25
[map_server-1] [DEBUG] [map_io]: occupied_thresh: 0.65
[map_server-1] [DEBUG] [map_io]: mode: trinary
[map_server-1] [DEBUG] [map_io]: negate: 0
[map_server-1] [INFO] [map_io]: Loading image_file: ./test1.pgm
[map_server-1] [DEBUG] [map_io]: Read map ./test1.pgm: 226 X 334 map @ 0.05 m/cell
[lifecycle_manager-3] [INFO] [1701837621.036223330] [lifecycle_manager_localization]: Configuring amcl
[amcl-2] [INFO] [1701837621.959216207] [amcl]: Configuring
[amcl-2] [INFO] [1701837621.959542574] [amcl]: initTransforms
[amcl-2] [INFO] [1701837622.029746757] [amcl]: initPubSub
[amcl-2] [INFO] [1701837622.083357240] [amcl]: Subscribed to map topic.
[lifecycle_manager-3] [INFO] [1701837622.120085396] [lifecycle_manager_localization]: Activating map_server
[map_server-1] [INFO] [1701837622.123092733] [map_server]: Activating
[map_server-1] [INFO] [1701837622.124426958] [map_server]: Creating bond (map_server) to lifecycle manager.
[amcl-2] [INFO] [1701837622.126428065] [amcl]: Received a 226 X 334 map @ 0.050 m/pix
[lifecycle_manager-3] [INFO] [1701837622.257833219] [lifecycle_manager_localization]: Server map_server connected with bond.
[lifecycle_manager-3] [INFO] [1701837622.257991643] [lifecycle_manager_localization]: Activating amcl
[... 319 lines cut ...]
[rviz2-1] [INFO] [1701837683.163595354] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837674.189 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837683.411797490] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837674.378 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837683.577359160] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837674.539 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837683.708280235] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837674.640 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837683.811908478] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837674.780 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837683.964162667] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837674.990 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837684.217591012] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837675.178 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837684.217717413] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837675.240 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837684.359285182] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837675.389 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837684.612346436] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837675.578 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837684.762648920] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837675.790 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[rviz2-1] [INFO] [1701837684.905599105] [rviz2]: Message Filter dropping message: frame 'odom' at time 1701837675.840 for reason 'the timestamp on the message is earlier than all the data in the transform cache'

Similar symptoms referred to,
github.com/iRobotEducation/create3_examples/issues/44

These are rivz photos and create3 configuration photos and logs.

**Other notes**
Add anything else you thing is important.
[messages.txt](https://github.com/turtlebot/turtlebot4/files/13576062/messages.txt)
[image]
[image]
[image]
[image]

gmchoi96, 2023-12-12:
When I open the navigation package, it says the following:

<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>turtlebot4_navigation</name>
  <version>1.0.1</version>
  <description>Turtlebot4 Navigation</description>
  <maintainer email="rkreinin@clearpathrobotics.com">rkreinin</maintainer>
  <license>Apache 2.0</license>

My robot S/N
RCI3099J 221122 N101901

hilary-luo, 2023-12-13:
Please update all of your turtlebot4 packages on the robots to the latest, this sounds like it could be a bug that was already fixed.

gmchoi96, 2023-12-15:
Thanks, hilary-luo. it's resolved.

However, intermittently, depending on the saved map, the global cost map does not appear and Nav2goal does not work. Have you checked for bugs regarding this?

In some cases, redrawing the map and saving it may solve the problem. When I just updated the Nav, I found that the local cost map came up but the global cost map did not and the nav didn't work, so I redrawed the map through slam, rebooted, and it worked normally.

This problem was also seen often.

hilary-luo, 2023-12-17:
@gmchoi96 Are there specific instructions that I could follow to try and replicate this issue? or is it too intermittent for that?

When you run into this issue, can you pull and share the create3 log? Also, are you using simple discovery or discovery server?

gmchoi96, 2023-12-19:
I use all robots connected to the Discovery server. Simple discovery has too slow communication.

Since this is a symptom that appears once in a while when using the robot, I will upload the Create3 log again when the symptom occurs.

hilary-luo, 2024-02-13:
Feel free to reopen this ticket when you have a log to share.

closed 2024-02-13 by hilary-luo