turtlebot/turtlebot4 #322: Nav2 does not work

myamo1, 2023-12-06:
**Please provide the following information:**
 - OS: Ubuntu 22.04
 - ROS Distro: Humble
 - Installed
 
 **Expected behaviour**
 Visualize the robot on the map (Rviz). Provide the initial pose estimate and the goal. Finally, the robot moves toward the destination.

 **Actual behaviour**
 The map shows on RViz but I have the errors shown on the screenshot.
[image]

**Other notes**
Similar errors appear when I try it with gazebo.

hilary-luo, 2023-12-17:
@myamo1 please provide the exact commands that you are running and the full terminal output. Overall I don't see anything specifically alarming here, it looks like the normal status as different nodes are waiting for each other. How long have you given it in this state for everything to get started? Can you also provide your create3 logs from when you run this? If you wish to also troubleshoot the simulation then also give the exact commands and terminal outputs from when you ran it with simulation.

myamo1, 2023-12-18:
It worked, finally. I just needed to give it a longer time. Sometimes, it works quickly, but other times, I have to wait longer.
Thanks.

closed 2024-01-15 by hilary-luo