turtlebot/turtlebot4 #323: Trouble Connecting Turtlebot4 to Institutional Wi-Fi Network.

Tauhid-Tanjim13, 2023-12-07:
I'm working with a Turtlebot4 running **Galactic Distro** and using a Create3 base with **firmware G.5.3**. **My goal is to connect the Turtlebot4 to my institutional Wi-Fi network named "RedRover"**. I'm attempting this connection using **Discovery Server Mode**, so I don't have to directly connect the Create3 to RedRover.

The problem arises when I execute the command sudo wifi.sh -s 'RedRover' -p '' -r 'US' && sudo reboot. This command successfully changes my IP address to a 10.X.X.X subnet. However, despite this, I'm unable to ping or access the Turtlebot4 when connected to RedRover. Interestingly, it remains accessible when connected to my personal Wi-Fi network (TP-Link_DXX_5G).

Additional steps I've taken include providing the Raspberry Pi's MAC address to my institution to allow network access. Despite this, the connectivity issue persists on the RedRover network.

I'm seeking advice on how to resolve this connectivity issue. Are there specific settings or considerations for connecting a Turtlebot4 to an institutional network like RedRover? Any insights or suggestions would be greatly appreciated.

[image]
[image]

smatar22, 2023-12-18:
Hello Tauhid,
Thank you for opening up this issue. I have a question regarding your personal network TP-Link_DXX_5G, is the network password protected or does it also use no password? If connecting to RedRover, try running the command without the -p flag: `sudo wifi.sh -s 'RedRover' -p '' -r 'US' && sudo reboot` and see if the TB4 is able to connect to the network in this manner.

Looking forward to hearing from you soon.

Tauhid-Tanjim13, 2023-12-19:
@smatar22, TP-Link_DXX_5G has its own password. My robot was previously connected to TP-Link_DXX_5G. But I tried to switch my TB4 to 'RedRover'. 
I have already tried this: sudo wifi.sh -s 'RedRover' -p '' -r 'US' && sudo reboot. I saw the IP address of TB4 changes to something from 192.168.X.X to something 10.X.X.X. But I cannot ping the TB4 from RedRover, I can ping the robot when I'm connected to TP-Link_DXX_5G this router. Although my robot is showing inside it is connected to something name RedRover (by ssh to TB4).

Tauhid-Tanjim13, 2023-12-20:
@smatar22 , No, after connecting the TB4 to Redrover, is the address: 10.52.16.4 and one of the laptop's ip address is 10.52.16.7. I can ssh with that laptop. Another laptop's IP address is different doesn't match last two numbers. I cannot ping TB4 with that laptop. Thanks for the information. I'll talk to the IT team. We are planning to put 6-8 TB4s under our institutional wifi network. I hope it will work.

smatar22, 2024-01-04:
Hello Tauhid,
From the description it appears that due to the laptop being on a different subnet, it is unable to communicate with the TB4 team. If possible, try changing the adapter settings for the laptop with the different IP address to something in line with the format 10.52.16.X and see if it is now able to ssh and ping the Turtlebot.

In terms of the command, if the original command works, then there is no need to try with the new command.

Looking forward to hearing from you soon.

Tauhid-Tanjim13, 2024-01-05:
@smatar22 , Thank you so much. It worked after talking to our institutional IT folks. I'll work on other TB4s. Thank you for the constant support. I'm closing this for now. If I face any problem regarding other TB4s. I'll open the issue again.

closed 2024-01-05 by Tauhid-Tanjim13