turtlebot/turtlebot4 #325: Could not subscribe odom topic after firmware update

atinfinity, 2023-12-12:
**Please provide the following information:**
 - OS: (e.g. Ubuntu 22.04)
 - ROS Distro: (e.g. Humble)
 - Built from source or installed: installed
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Expected behaviour**

After user updated the latest firmware, user run ros2 topic echo /odom.
I expect that user can subscribe /odom topic.

 **Actual behaviour**

After user updated the latest firmware, user run `ros2 topic echo /odom`.
But, I could not subscribe `/odom` topic.

**To Reproduce**
Provide the steps to reproduce:

1. updated the latest firmware(<https://iroboteducation.github.io/create3_docs/releases/h_2_3/>)
2. boot TurtleBot4
3. do `ros2 topic echo /odom`

**Other notes**

> * Update to the latest firmware using the webserver.
> * Once updated, perform a factory reset to disconnect the Create® 3 from any Wi-Fi networks.

https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html

I read the above description.
So, I updated firmware to [iRobot Create 3 Release H.2.3](https://iroboteducation.github.io/create3_docs/releases/h_2_3/).
And, I performed a factory reset. As a result, I could not subscribe `odom` topic.

```
$ ros2 node list
/analyzers
/joint_state_publisher
/joy_linux_node
/launch_ros_1022
/oakd
/oakd_container
/robot_state_publisher
/rplidar_composition
/teleop_twist_joy_node
/turtlebot4_base_node
/turtlebot4_diagnostics
/turtlebot4_node
```

atinfinity, 2023-12-12:
I downgrade to <https://iroboteducation.github.io/create3_docs/releases/h_2_2/>. This problem was not resolved.

atinfinity, 2023-12-12:
```shell
$ ros2 node list
/analyzers
/joint_state_publisher
/joy_linux_node
/launch_ros_1022
/oakd
/oakd_container
/robot_state_publisher
/rplidar_composition
/teleop_twist_joy_node
/turtlebot4_base_node
/turtlebot4_diagnostics
/turtlebot4_node
```

It seem that iRobot Create 3 node failed to launch.

atinfinity, 2023-12-12:
I checked log via Web server of Create 3.

```
Dec 12 07:18:54 user.notice create-platform: [INFO] [1702365534.202752691] [mobility_monitor]: Lifecycle callback: on_configure
Dec 12 07:18:58 user.notice create-platform: [INFO] [1702365538.432919320] [create-platform]: Activating mobility_monitor node
Dec 12 07:18:58 user.notice create-platform: [INFO] [1702365538.433294812] [mobility_monitor]: Lifecycle callback: on_activate
Dec 12 07:18:58 user.notice create-platform: [INFO] [1702365538.435663999] [mobility_monitor]: New kidnap status: kidnap cleared
Dec 12 07:18:59 user.notice create-platform: [INFO] [1702365539.040102439] [_internal.mobility]: Received enable wheels request
Dec 12 07:18:59 user.notice create-platform: [INFO] [1702365539.102631961] [mobility_monitor]: Requested to enable wheels!
Dec 12 07:18:59 user.notice create-platform: [INFO] [1702365539.389320450] [mobility_monitor]: Updating ready to drive: true
Dec 12 07:18:59 user.notice create-platform: [INFO] [1702365539.390442063] [mobility_monitor]: Kidnap cleared! Wheels are now enabled!
Dec 12 07:18:59 user.notice create-platform: [INFO] [1702365539.433863371] [create-platform]: Started mobility_monitor node
Dec 12 07:18:59 user.notice create-platform: [INFO] [1702365539.434053600] [create-platform]: Configuring robot_state node
Dec 12 07:18:59 user.notice create-platform: [INFO] [1702365539.440272596] [robot_state]: Lifecycle callback: on_configure
Dec 12 07:19:10 user.notice create-platform: [INFO] [1702365550.630026340] [create-platform]: Activating robot_state node
Dec 12 07:19:10 user.notice create-platform: [INFO] [1702365550.631237922] [robot_state]: Lifecycle callback: on_activate
Dec 12 07:19:11 user.notice create-platform: [INFO] [1702365551.633011869] [create-platform]: Started robot_state node
Dec 12 07:19:11 user.notice create-platform: [INFO] [1702365551.633208088] [create-platform]: Configuring static_transform node
Dec 12 07:19:16 user.notice create-platform: [INFO] [1702365556.616494628] [create-platform]: Activating static_transform node
Dec 12 07:19:17 user.notice create-platform: [INFO] [1702365557.617855402] [create-platform]: Started static_transform node
Dec 12 07:19:17 user.notice create-platform: [INFO] [1702365557.651553503] [create-platform]: Ready to go!
Dec 12 07:19:19 user.notice ntpd: ntpd: bad address '3.irobot.pool.ntp.org'
Dec 12 07:19:24 user.notice ntpd: ntpd: bad address '2.irobot.pool.ntp.org'
```

> [mobility_monitor]: Lifecycle callback: on_configure

There is message of mobility_monitor. But, mobility_monitor is not displayed as the result of "ros2 node list".

> ntpd: bad address '3.irobot.pool.ntp.org'

This setting is default of Create 3. But this message was displayed.

atinfinity, 2023-12-12:
@hilary-luo I'm investigating this problem. Could you give me your comment?

hilary-luo, 2023-12-13:
It sounds like you haven't configured the robot after the factory reset, did you reconfigure all of the networking etc? if so, did you reconfigure as simple discovery or discovery server?

atinfinity, 2023-12-14:
I reconfigured network setting. As a result, I could not connect to RasPi4 via ssh. So, I try to install Raspberry Pi image.  
<https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#install-latest-raspberry-pi-image>

atinfinity, 2023-12-14:
> So, I try to install Raspberry Pi image.

I installed Raspberry Pi image. As a result, this issue is resolved. So, I close this issue.

closed 2023-12-14 by atinfinity