turtlebot/turtlebot4 #327: Wifi and IP adress problem

ValentinLeglise, 2023-12-14:
OS: Ubuntu 22.04
ROS Distro: Humble
Installed)

Hello ClearPath.

My customer has some problem with the Wifi set up of his TB4 and his actual behavior is as follows :

"
When the RPi is configured and you set the desired network, it connects correctly and has a stable connection. Same for the create 3.
It remains OK for 3 or 4 power-offs of the robot. Suddenly, when you turn on the robot again it does not get an IP anymore. Curiously, both the RPi and the Create lose the IP, none of both connects again. The problem persists until a new configuration is loaded using turtlebot4-setup.
"
Do you know where does it come from ?

Best regards.

smatar22, 2023-12-20:
Hello Valentin,
Thank you for reaching out on the Turtlebot 4 Github. How much time pasess between the initial setup and the TB4 losing it's IP address. When the TB4 network is setup again, is the address the same as before it was lost or is a new address given? Is this a standard Wi-Fi network or an institutional network? From the description it seems the lease on the original expires, and the address is no longer available when the TB4 reconnects to the network. Let me know if this is not the case. Looking forward to hearing from you soon.

ValentinLeglise, 2023-12-22:
Hello,
Thank you for your response. There are the answers of the customer:

1. There is not a specific time amount between those events, in fact when the IP is set, until there are like 2 or 3 restarts of the robot the problem does not appear again.
2. Is the same as before, being configured with DHCP
3. It is an standard wifi, set-up with a 2.4Ghz Access Point.

Regarding your last sentence ("t seems the lease on the original expires, and the address is no longer available when the TB4 reconnects to the network"), Can you tell me more about how the customer can verify this?

Regards.

ValentinLeglise, 2024-01-02:
Hello ClearPath

Is it possible to have an answer for this request please ?

Thank you. Best regards.

smatar22, 2024-01-03:
Hello Valentin,
Apologies for the late reply our offices were closed for the holidays. 

Would it be possible to share the Create3 log files found when connecting to the Create3's webserver: https://iroboteducation.github.io/create3_docs/webserver/logs/

How many Turtlebot's are being used on this network? If it is only one, would it be possible to have the client switch to a discovery server setup: https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html After doing so, restart the TB4 a few times and see if it exhibits this same behaviour.

If there are multiple TB4 units using this network, is this behaviour common across all the units or it is restricted to a single unit?

Looking forward to hearing from you soon.

ValentinLeglise, 2024-01-11:
Hello ClearPath,

Attached, you can see the logs.
[messages.txt](https://github.com/turtlebot/turtlebot4/files/13905511/messages.txt)

They have a network that is composed by the access point (provided by an Ubuntu machine) and 3 Turtlebots connected to it using the simple discovery setup. The other 2 work properly as expected, and it is always the same unit the one that has the problem.

Regards

smatar22, 2024-01-16:
Hello Valentin,
Has the client tried reimaging the Raspberry Pi on the issue unit? Since it is isolated to the single unit perhaps a reset will clear any initial configuration issues present on the unit. 

Let me know how this recommendation works for your client. Looking forward to hearing from you soon.

ValentinLeglise, 2024-01-17:
Hello smatar,

Yes he tried multiple times weeks ago and the problem persists. Do we have to repatriate the unit ?

Best regards.

smatar22, 2024-01-22:
Hello Valentin,
Would it be possible to have the client boot the Turtlebot4 unit a few times (until it gets removed from the network again). Once this occurs have the SSH into the Turtlebot4 using an ethernet connection, the ethernet IP address for the Turtlebot4 should be `192.168.185.3`. After doing so have them provide the output of the following two commands: `journalctl -u NetworkManager.service` and `sudo dmesg` The updated Create3 logs may also prove to be useful. The server should be accessible using the ethernet address of the Turtlebot4.

Lastly, they may also want to check the logs of their network to see if they can find any information as to way the Turtlebot4 is being removed.

Looking forward to hearing from you soon.

smatarCPR, 2024-02-05:
Hello Valentin,
Thank you for sending the logs. Reviewing the contents we found the following excerpt:
```
Dec 15 15:14:27 ubuntu NetworkManager[2939]: <info>  [1702653267.8858] Config: added 'ssid' value 'charm_pc_wifi'
Dec 15 15:14:27 ubuntu NetworkManager[2939]: <info>  [1702653267.8862] Config: added 'scan_ssid' value '1'
Dec 15 15:14:27 ubuntu NetworkManager[2939]: <info>  [1702653267.8866] Config: added 'freq_list' value '2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472 2484'
Dec 15 15:14:27 ubuntu NetworkManager[2939]: <info>  [1702653267.8867] Config: added 'bgscan' value 'simple:30:-70:86400'
Dec 15 15:14:27 ubuntu NetworkManager[2939]: <info>  [1702653267.8869] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256 FT-PSK'
Dec 15 15:14:27 ubuntu NetworkManager[2939]: <info>  [1702653267.8870] Config: added 'psk' value '<hidden>'
Dec 15 15:14:27 ubuntu NetworkManager[2939]: <info>  [1702653267.8893] device (wlan0): supplicant interface state: disconnected -> scanning
Dec 15 15:14:27 ubuntu NetworkManager[2939]: <info>  [1702653267.8894] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Dec 15 15:14:27 ubuntu NetworkManager[2939]: <info>  [1702653267.9460] device (wlan0): supplicant interface state: scanning -> associating
Dec 15 15:14:27 ubuntu NetworkManager[2939]: <info>  [1702653267.9462] device (p2p-dev-wlan0): supplicant management interface state: scanning -> associating
Dec 15 15:14:30 ubuntu NetworkManager[2939]: <info>  [1702653270.6977] device (wlan0): supplicant interface state: associating -> disconnected
Dec 15 15:14:30 ubuntu NetworkManager[2939]: <info>  [1702653270.6979] device (p2p-dev-wlan0): supplicant management interface state: associating -> disconnected
Dec 15 15:14:30 ubuntu NetworkManager[2939]: <info>  [1702653270.7964] device (wlan0): supplicant interface state: disconnected -> scanning
Dec 15 15:14:30 ubuntu NetworkManager[2939]: <info>  [1702653270.7966] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Dec 15 15:14:33 ubuntu NetworkManager[2939]: <info>  [1702653273.2365] device (wlan0): supplicant interface state: scanning -> associating
Dec 15 15:14:33 ubuntu NetworkManager[2939]: <info>  [1702653273.2366] device (p2p-dev-wlan0): supplicant management interface state: scanning -> associating
Dec 15 15:14:36 ubuntu NetworkManager[2939]: <info>  [1702653276.0204] device (wlan0): supplicant interface state: associating -> disconnected
Dec 15 15:14:36 ubuntu NetworkManager[2939]: <info>  [1702653276.0206] device (p2p-dev-wlan0): supplicant management interface state: associating -> disconnected
Dec 15 15:14:36 ubuntu NetworkManager[2939]: <info>  [1702653276.5383] device (wlan0): supplicant interface state: disconnected -> scanning
Dec 15 15:14:36 ubuntu NetworkManager[2939]: <info>  [1702653276.5385] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Dec 15 15:14:38 ubuntu NetworkManager[2939]: <info>  [1702653278.9765] device (wlan0): supplicant interface state: scanning -> associating
Dec 15 15:14:38 ubuntu NetworkManager[2939]: <info>  [1702653278.9767] device (p2p-dev-wlan0): supplicant management interface state: scanning -> associating
Dec 15 15:14:41 ubuntu NetworkManager[2939]: <info>  [1702653281.4235] device (wlan0): supplicant interface state: associating -> disconnected
Dec 15 15:14:41 ubuntu NetworkManager[2939]: <info>  [1702653281.4237] device (p2p-dev-wlan0): supplicant management interface state: associating -> disconnected
Dec 15 15:14:42 ubuntu NetworkManager[2939]: <info>  [1702653282.4418] device (wlan0): supplicant interface state: disconnected -> scanning
Dec 15 15:14:42 ubuntu NetworkManager[2939]: <info>  [1702653282.4420] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Dec 15 15:14:44 ubuntu NetworkManager[2939]: <info>  [1702653284.8100] device (wlan0): supplicant interface state: scanning -> associating
Dec 15 15:14:44 ubuntu NetworkManager[2939]: <info>  [1702653284.8101] device (p2p-dev-wlan0): supplicant management interface state: scanning -> associating
Dec 15 15:14:47 ubuntu NetworkManager[2939]: <info>  [1702653287.4907] device (wlan0): supplicant interface state: associating -> disconnected
Dec 15 15:14:47 ubuntu NetworkManager[2939]: <info>  [1702653287.4908] device (p2p-dev-wlan0): supplicant management interface state: associating -> disconnected
Dec 15 15:14:52 ubuntu NetworkManager[2939]: <info>  [1702653292.5141] device (wlan0): supplicant interface state: disconnected -> scanning
Dec 15 15:14:52 ubuntu NetworkManager[2939]: <info>  [1702653292.5144] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Dec 15 15:14:53 ubuntu NetworkManager[2939]: <warn>  [1702653293.5624] device (wlan0): Activation: (wifi) association took too long
Dec 15 15:14:53 ubuntu NetworkManager[2939]: <info>  [1702653293.5625] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Dec 15 15:14:53 ubuntu NetworkManager[2939]: <info>  [1702653293.5632] sup-iface[0a2a91855948806a,2,wlan0]: wps: type pbc start...
Dec 15 15:14:53 ubuntu NetworkManager[2939]: <warn>  [1702653293.5637] device (wlan0): Activation: (wifi) asking for new secrets
Dec 15 15:14:53 ubuntu NetworkManager[2939]: <warn>  [1702653293.5645] device (wlan0): no secrets: No agents were available for this request.
[... cut ...]
From these logs it appears that Turtlebot4 unit is attempting to connect to the network multiple but fails. Unfortunately it is not reporting the reason for this failed connection.

This may be a case of too many devices being connected to the network at the same time. Has the client set a limit on the number of devices that can be connect to this network? Does the issue persist if all other TB4 units (and any other devices connected to the network) are turned off/have their Wi-Fi disabled?

Looking forward to hearing from you soon.
Best Regards,
Saif

ValentinLeglise, 2024-02-09:
According the customer, he problem persists even with only the unit attempting to connect, and is independent of the number of devices connected.

ValentinLeglise, 2024-02-16:
Hello,

Please can I have a response for this ?

Thanks.
Regards.

smatarCPR, 2024-02-23:
Hello,
Apologies for the late reply on this I am consulting with senior TB4 engineers with regards to this matter. I should have an update for you very soon.
Best Regards,
Saif

smatarCPR, 2024-02-28:
Hello,
Apologies for the late reply. Would it be check one last item? Should this check fail we'll begin making arrangements to ship the unit to our head offices and repair/replace the unit.

The item to check is the ribbon cable connecting the Raspberry Pi with the User Interface PCBA. Ensure that this cable is not covering the Raspbery PI or its wireless adapter:
[image]
[image]
To gain access the Pi and the Ribbon cable:
Remove Top Integration Plate:
The plate is attached to the robot by four Round head M4x0.7 screws. Remove the screws by using a 2.5mm hex key.
[image]
Remove the PCBA:
After removing the top plate, you can remove the PCBA by removing the four standoffs:
[image]
After this you can carefully lift the PCBA, such that none of the cables are pulled taught, and you should be able to access the Raspberry Pi.

Let me know how this recommendation works for you. Looking forward to hearing from you .
Best Regards,
Saif

ValentinLeglise, 2024-02-29:
Hello,

Thank you for this information.
We really hope that will be helpful. Indeed, if not, we will replace the robot.

Best regards, Valentin.

smatarCPR, 2024-03-18:
Hello @ValentinLeglise ,
Any updates regarding this? Looking forward to hearing from you soon.
Best Regards,
Saif

ValentinLeglise, 2024-03-20:
Hello,

We are waiting for a response from our client. We will let you know.
Best regards.

smatarCPR, 2024-04-22:
Hello @ValentinLeglise ,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Saif

closed 2024-04-22 by smatarCPR