turtlebot/turtlebot4 #328: TB4 WiFi connection problem

ValentinLeglise, 2023-12-14:
Informations:

- Serial number: RCI3099J221122N100079
- Operating system: Firmware version H2.3 (irobot create3) 
- ROS Distro: Humble

Hello ClearPath.

A customer tried to connect his robot to his Wifi and it behaves as follows: 

"I managed to connect the create to WiFi in general, but it is still not working as expected. 
First, updating the irobot create 3 to the newest firmware always results in an error message “command [“/usr/bin/ota_install.sh”, “-f”, “/data/latest_ota.swu”] exited with code 66”. 
It does not directly connect to the WiFi but I have to rescan the networks and connect afterward. When connected, the date is not updated and stays to Sep 29. All other robots (we have ordered 8 in total) did update the date automatically after connecting to WiFi."

Attached you can find the robot's log after the customer booted it, rescanned networks and connected it to WiFi. 
He also says "There are several error messages related to the WiFi connection. I tried reinstalling the firmware several times which did not help."

[logs.pdf](https://github.com/turtlebot/turtlebot4/files/13673477/logs.pdf)

smatar22, 2023-12-22:
Hello Valentine,
Thank you for reaching out on the Turtlebot 4 Github. From the logs provided I suspect that the connection between the Create3 and the Raspberry Pi is unstable, there are many mentions of the connection timing out in the logs provided. After updating the connection settings on the Create3's webserver and rescanning, does the IP address get updated?

If the option is available, try factory resetting the Create3 to clear all the previously used connection settings. Afterwards, try connecting to the Wi-Fi network and see if it can connect to the network and connect to the internet.

Is the behaviour replicated when a hotspot, or alternative network is being used?

I also wanted to note that according to the software releases page https://iroboteducation.github.io/create3_docs/releases/overview/ firmware version H2.3 is the latest firmware for ROS Humble. Where are they receiving this firmware update? Have they downloaded the file from the software releases and are attempting to   upload it to the Create3 or are they using the drop down options?
[image]

Looking forward to hearing from you soon.

ValentinLeglise, 2024-01-02:
Hello.

Thank you for this response. I will get back to you when we will have received the answer to all these questions from our customer.

Regards.

ValentinLeglise, 2024-01-11:
Hello,

Our customer did a factory reset of the create 3 and reconnected to WIFI. It seems to solve all issues he had.
Thank you for your help.

Best regards

smatar22, 2024-01-11:
Hello Valentin,
Glad to hear that the issue was resolved. I will go ahead and close this ticket. Should you need any further assistance, please feel free to open a new ticket.
Best Regards,
Saif

closed 2024-01-11 by smatar22