turtlebot/turtlebot4 #329: Could not find a connection between 'map' and 'base_link'

atinfinity, 2023-12-20:
I enabled Discovery Server(<https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html>).
I tried to run `ros2 topic list` on **other PC(NOT RasPi4)**. And, I could find TurtleBot4 topic.
But, I could not resolved transform between `map` and `base_link`.

**Please provide the following information:**
 - OS: (e.g. Ubuntu 22.04)
 - ROS Distro: (e.g. Humble)
 - Built from source or installed: installed
 - Package version: (if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-turtlebot4-PACKAGE_WITH_ISSUE`, if from source, give commit hash)

 **Expected behaviour**

When user run  `slam_toolbox `, user can get transform between `map` and `base_link`. 

 **Actual behaviour**

When user run  `slam_toolbox `, user could not get transform between `map` and `base_link`. 

**To Reproduce**
1. enabled Discovery Server. 
2. launch `ros2 launch turtlebot4_navigation slam.launch.py ` on RasPi4
3. launch `ros2 run tf2_ros tf2_echo map base_link` on **other PC(NOT RasPi4)**

**Other notes**

### RasPi4

```shell
$ ros2 topic echo /map --no-arr
header:
  stamp:
    sec: 1703034078
    nanosec: 640259523
  frame_id: map
info:
  map_load_time:
    sec: 0
    nanosec: 0
  resolution: 0.05000000074505806
  width: 147
  height: 108
  origin:
    position:
      x: -5.175382561534499
      y: -3.2515461374978822
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
data: '<sequence type: int8, length: 15876>'
```

```shell
$ rosrun tf2_ros tf2_echo map base_link
At time 1703034108.30666946
- Translation: [-0.000, -0.000, 0.000]
- Rotation: in Quaternion [-0.000, -0.000, 0.000, 1.000]
- Rotation: in RPY (radian) [-0.000, -0.000, 0.001]
- Rotation: in RPY (degree) [-0.000, -0.000, 0.048]
- Matrix:
  1.000 -0.001 -0.000 -0.000
  0.001  1.000  0.000 -0.000
  0.000 -0.000  1.000  0.000
  0.000  0.000  0.000  1.000
At time 1703034109.39461181
- Translation: [-0.000, -0.000, 0.000]
- Rotation: in Quaternion [-0.000, -0.000, 0.000, 1.000]
- Rotation: in RPY (radian) [-0.000, -0.000, 0.001]
- Rotation: in RPY (degree) [-0.000, -0.000, 0.048]
- Matrix:
  1.000 -0.001 -0.000 -0.000
  0.001  1.000  0.000 -0.000
  0.000 -0.000  1.000  0.000
  0.000  0.000  0.000  1.000
```

It seems that there is no problem.

### Other PC

```shell
$ ros2 topic echo /map --no-arr
header:
  stamp:
    sec: 1703034163
    nanosec: 190597969
  frame_id: map
info:
  map_load_time:
    sec: 0
    nanosec: 0
  resolution: 0.05000000074505806
  width: 148
  height: 108
  origin:
    position:
      x: -5.195460385557224
      y: -3.2515461374978916
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
data: '<sequence type: int8, length: 15984>'
```

```shell
$ ros2 run tf2_ros tf2_echo map base_link
[INFO] [1703034195.754294720] [tf2_echo]: Waiting for transform map ->  base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
[INFO] [1703034196.755489788] [tf2_echo]: Waiting for transform map ->  base_link: Could not find a connection between 'map' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.
```

I could not resolved transform between `map` and `base_link`. As a result, I could not visualize using RViz2.

atinfinity, 2023-12-22:
@hilary-luo Could you give me your comment?

hilary-luo, 2023-12-22:
If you are not able to establish communication between the raspberry pi and create3 base (from your other ticket) then that will cause problems trying to run slam.

atinfinity, 2023-12-22:
> If you are not able to establish communication between the raspberry pi and create3 base

- RasPi4 and Create 3 were able to communicate with `ping`.
- RasPi4 could subscribed `odom` topic. This topic is published from Create 3.
- I published `cmd_vel` on RasPi4. As a result, TurtleBot4 moved.
- After I launch `slam_toolbox` node, RasPi4 could subscribed `map` topic.
- I recorded rosbag. And, I checked `map` topic. The `slam_toolbox` successfully generated a map.
- I could get transform between `map` and `base_link` on RasPi4.

So, I think that there is no problem with communication between RasPi4 and Create 3.

atinfinity, 2023-12-25:
I have summarized the experimental results(<https://github.com/turtlebot/turtlebot4/issues/329#issuecomment-1868086789>) in the following figure.

[image]

I configured the network using <https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html> as a reference.
But, it looks like there is a problem with the network connection between **User PC** and **Create 3**.

atinfinity, 2023-12-25:
@hilary-luo I provided additional information. Could you give me your comment?

hilary-luo, 2023-12-27:
Are you now able to see all of the create3 topics on the raspberry pi? If you are now able to do that then slam / nav2 should work properly. 

Are you running the slam / nav2 nodes on the raspberry pi or on your PC? I would generally suggest running them on your PC (assuming your PC should have more processing power than the pi). Are you able to list or subscribe to any other create3 topics?

atinfinity, 2023-12-28:
@hilary-luo 

> Are you now able to see all of the create3 topics on the raspberry pi?

Yes. I checked to subscribe "create3 topics" on RasPi4.
On the other hands, I could not subscribe "create3 topics" on Other PC.

|Topic Name|RasPi4|
|---|---|
|/battery_state|OK|
|/cliff_intensity|OK|
|/dock_status|OK|NG|
|/hazard_detection|OK|NG|
|/hmi/buttons|OK|
|/hmi/display|OK|
|/imu|OK|NG|
|/interface_buttons|OK|
|/ir_intensity|OK|
|/kidnap_status|OK|
|/mouse|OK|
|/odom|OK|
|/slip_status|OK|
|/stop_status|OK|
|/wheel_status|OK|
|/wheel_ticks|OK|
|/wheel_vels|OK|

> Are you running the slam / nav2 nodes on the raspberry pi or on your PC?

Now, I run slam / nav2 nodes on RasPi4. It seems that these nodes are working normally.
But, I could not subscribe "create3 topics" on Other PC. So, I could not run slam / nav2 nodes on other PC.

atinfinity, 2023-12-28:
@hilary-luo

> I could not subscribe "create3 topics" on Other PC. 

I guess that there is problem in network routing between Create3 and Other PC.
But, I could not find the information from <https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html>.

When you enabled "Discovery Server", could you subscribe "create3 topics" from Other PC on your environment?

atinfinity, 2023-12-28:
@hilary-luo I checked network configuration.

### RasPi4

I enabled Discovery Server via `turtlebot4-setup`.

```
$ ip route
192.168.186.0/24 dev usb0 proto kernel scope link src 192.168.186.3
```

It seems that there is no problem in connection between RasPi4 and Create 3.

```
$ ping 192.168.186.2
PING 192.168.186.2 (192.168.186.2) 56(84) bytes of data.
64 bytes from 192.168.186.2: icmp_seq=1 ttl=64 time=0.460 ms
64 bytes from 192.168.186.2: icmp_seq=2 ttl=64 time=2.67 ms
```

### Other PC

I run `turtlebot4_discovery/configure_discovery.sh`.

```
$ ip route
192.168.186.0/24 via $RASPI4_IP dev en0
```

It seems that there is no problem in connection between Other PC and RasPi4.

```
ping $RASPI4_IP
64 bytes from $RASPI4_IP: icmp_seq=0 ttl=64 time=12.359 ms
64 bytes from $RASPI4_IP: icmp_seq=1 ttl=64 time=13.154 ms
```

But, there is problem in connection between Other PC and Create 3.

```
$ ping 192.168.186.2
PING 192.168.186.2 (192.168.186.2): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
```

### Create 3

I guess that Create 3 also requires routing settings.
But, I could not find the network configuration from Create 3 Web Server.

hilary-luo, 2023-12-29:
You have tested well that you have communication between the create3 and raspberry pi so it shouldn't be a problem with the settings on the create3.

> I enabled Discovery Server via turtlebot4-setup.
`$ ip route`
`192.168.186.0/24 via $RASPI4_IP dev en0`

I don't think that the routing table can support the environment variables. Try setting up the routing on the remote PC with the raspberry pi IP address hard coded.

jordanlv, 2024-02-09:
Hello @atinfinity ,

Have you solved this issue? I believe I'm encountering a similar problem.
Is your discovery server the hotspot for the RPI4?"

atinfinity, 2024-02-13:
@jordanlv 

> Have you solved this issue?

No.

> Is your discovery server the hotspot of the RPI4?

Yes.

> And are you able to echo other create3 topics like /interfaces_button ?

I could subscribe `/interfaces_button` topic on RasPi4.
But, I could NOT subscribe `/interfaces_button` topic on User PC(NOT RasPi4).

jordanlv, 2024-02-13:
So I think we have the same issue.

In my case, when the hotspot is not the RPI (for example the user PC), I can read /odom from the user PC but not /scan....

I decided to switch to simple discovery even if it's not a convenient option for me.

hilary-luo, 2024-02-13:
@atinfinity did you try switching your routing table to use a hard coded IP address as requested?

@jordanlv Your issue sounds different because you are reporting having issues echoing a raspberry pi topic not a create3 topic. If you would like aid troubleshooting it then please make your own ticket.

atinfinity, 2024-02-15:
@hilary-luo 
> did you try switching your routing table to use a hard coded IP address as requested?

I tried to use a hard coded IP address.
But, I could NOT subscribe create3 topics on User PC(NOT RasPi4).

jordanlv, 2024-02-15:
@hilary-luo I'm having the same issue. I was just mentioning what I've tried to solve it (not using the RPI as a hotspot).

atinfinity, 2024-02-15:
@hilary-luo I'll check this point. 

But, I read <https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html> to setup discovery server.
And I configured it by following the steps on this page. Is there a problem with this page or script?

hilary-luo, 2024-02-20:
@atinfinity There are no known bugs with that page or script but whether or not the system works does depend on the settings that you have configured. Verifying the final output of the script is a fast way to check and make sure a subset of the settings are correct and that they were applied correctly.

closed 2024-03-11 by hilary-luo