turtlebot/turtlebot4 #330: DepthAI Packages not working

sonnetx, 2023-12-20:
**Describe the bug**
When I type "ros2 launch depthai_examples rgb_stereo_node.launch.py camera_model:=OAK-D-LITE" into my terminal, I get the output below.

[rgb_stereo_node-1] terminate called after throwing an instance of 'std::runtime_error'
[rgb_stereo_node-1]   what():  Failed to connect to device, error message: X_LINK_DEVICE_ALREADY_IN_USE
[ERROR] [rgb_stereo_node-1]: process has died [pid 4121, exit code -6, cmd '/opt/ros/humble/lib/depthai_examples/rgb_stereo_node --ros-args --params-file /tmp/launch_params_16cs7983 --params-file /tmp/launch_params_jplihort --params-file /tmp/launch_params_9e_51gy4 --params-file /tmp/launch_params_gfatjr5q --params-file /tmp/launch_params_9lvklc37 --params-file /tmp/launch_params_zlyuspcs --params-file /tmp/launch_params_phhohrk1 --params-file /tmp/launch_params_ifra5ix4 --params-file /tmp/launch_params_k06bfwnf --params-file /tmp/launch_params_buov9yvl --params-file /tmp/launch_params_zitpsl5b --params-file /tmp/launch_params_cqvm9xeq --params-file /tmp/launch_params_bxkeokow --params-file /tmp/launch_params_wh0alsga'].

I am not sure why this is, since "ros2 launch turtlebot4_bringup oakd.launch.py" does not cause any issues. However, for this case, the terminal tells me STEREO is not supported.

If available launch files don't work in your case, please check if you also get errors while running:
- `stereo_inertial_node` in `depthai_examples`
[stereo_inertial_node-2] Listing available devices...
[stereo_inertial_node-2] terminate called after throwing an instance of 'std::runtime_error'
[stereo_inertial_node-2]   what():  " DepthAI Device with MxId  "x" not found.  "
[ERROR] [stereo_inertial_node-2]: process has died [pid 4225, exit code -6, cmd '/opt/ros/humble/lib/depthai_examples/stereo_inertial_node --ros-args --params-file /tmp/launch_params_9ss1hbjb --params-file /tmp/launch_params_a6bsmsvl --params-file /tmp/launch_params_fe8ug6hk --params-file /tmp/launch_params_ko8lgxjd --params-file /tmp/launch_params_l7j53a2b --params-file /tmp/launch_params_l8_x0qo9 --params-file /tmp/launch_params_w1yqcd_k --params-file /tmp/launch_params_55073ttf --params-file /tmp/launch_params_9g3opug3 --params-file /tmp/launch_params_19bu407_ --params-file /tmp/launch_params_odsxc5ul --params-file /tmp/launch_params_mk5_aopd --params-file /tmp/launch_params_ct4ldtf1 --params-file /tmp/launch_params_4wjcgjxq --params-file /tmp/launch_params_n4s88tb_ --params-file /tmp/launch_params_uub408n3 --params-file /tmp/launch_params_a3g0ktae --params-file /tmp/launch_params_i4gmsjd3 --params-file /tmp/launch_params_z963s1rs --params-file /tmp/launch_params_rwnoapae --params-file /tmp/launch_params_20xgb50b --params-file /tmp/launch_params_tdfk16zm --params-file /tmp/launch_params_syjnscss --params-file /tmp/launch_params_kkulfrut --params-file /tmp/launch_params_95fpxnju --params-file /tmp/launch_params_gxkufquw --params-file /tmp/launch_params_3r82ciph --params-file /tmp/launch_params_kl770cav --params-file /tmp/launch_params_sph5qq91 --params-file /tmp/launch_params_oql88zjh --params-file /tmp/launch_params_jo654_3o --params-file /tmp/launch_params_awmnztwq --params-file /tmp/launch_params_74wpjuax --params-file /tmp/launch_params_df65kjlu --params-file /tmp/launch_params_8l_q166k'].

- `camera` in `depthai_ros_driver`
This is my output when I try to run camera:
[component_container-1] [INFO] [1703085608.082415589] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-1] [ERROR] [1703085616.297393302] [oak]: Failed to connect to device, error message: X_LINK_DEVICE_ALREADY_IN_USE

I am using the default installation of depthai that comes on the turtlebot4. Any help would be appreciated!

 - Which ROS version are you using?
ROS 2
 - Which ROS distribution are you using ?
humble

 - Please include versions of following packages - `apt show ros-$ROS_DISTRO-depthai ros-$ROS_DISTRO-depthai-ros ros-$ROS_DISTRO-depthai-bridge ros-$ROS_DISTRO-depthai-ros-msgs ros-$ROS_DISTRO-depthai-ros-driver`
 
Package: ros-humble-depthai
Version: 2.23.0-1jammy.20231122.172543
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 20.5 MB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 4.5), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 12), libopencv-dev, libusb-1.0-0-dev, nlohmann-json3-dev, ros-humble-ros-workspace
Homepage: https://www.luxonis.com/
Download-Size: 10.6 MB
APT-Sources: http://packages.ros.org/ros2-testing/ubuntu jammy/main arm64 Packages
Description: DepthAI core is a C++ library which comes with firmware and an API to interact with OAK Platform

Package: ros-humble-depthai-ros
Version: 2.8.2-1jammy.20231216.022314
Priority: optional
Section: misc
Maintainer: sachin <sachin@luxonis.com>
Installed-Size: 43.0 kB
Depends: ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-depthai-descriptions, ros-humble-depthai-examples, ros-humble-depthai-filters, ros-humble-depthai-ros-driver, ros-humble-depthai-ros-msgs, ros-humble-ros-workspace
Download-Size: 6,130 B
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros2-testing/ubuntu jammy/main arm64 Packages
Description: The depthai-ros package

Package: ros-humble-depthai-bridge
Version: 2.8.2-1jammy.20231205.212807
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 407 kB
Depends: libc6 (>= 2.32), libgcc-s1 (>= 3.3.1), libopencv-calib3d4.5d (>= 4.5.4+dfsg), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgcodecs4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 11), libboost-dev, libopencv-dev, ros-humble-camera-info-manager, ros-humble-composition-interfaces, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-ros-msgs, ros-humble-image-transport, ros-humble-rclcpp, ros-humble-robot-state-publisher, ros-humble-ros-environment, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-stereo-msgs, ros-humble-tf2, ros-humble-tf2-geometry-msgs, ros-humble-tf2-ros, ros-humble-vision-msgs, ros-humble-xacro, ros-humble-ros-workspace
Download-Size: 118 kB
APT-Sources: http://packages.ros.org/ros2-testing/ubuntu jammy/main arm64 Packages
Description: The depthai_bridge package

Package: ros-humble-depthai-ros-msgs
Version: 2.8.2-1jammy.20231205.201618
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 1,675 kB
Depends: libc6 (>= 2.17), libgcc-s1 (>= 3.3.1), libpython3.10 (>= 3.10.0), libstdc++6 (>= 11), ros-humble-fastcdr, ros-humble-builtin-interfaces, ros-humble-geometry-msgs, ros-humble-rclcpp, ros-humble-rosidl-default-generators, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 127 kB
APT-Sources: http://packages.ros.org/ros2-testing/ubuntu jammy/main arm64 Packages
Description: Package to keep interface independent of the driver

Package: ros-humble-depthai-ros-driver
Version: 2.8.2-1jammy.20231216.021048
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 2,542 kB
Depends: libc6 (>= 2.34), libconsole-bridge1.0 (>= 1.0.1+dfsg2), libgcc-s1 (>= 3.3.1), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 12), libtinyxml2-9 (>= 8.0.0), ros-humble-ament-cmake-auto, ros-humble-camera-calibration, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-depthai-descriptions, ros-humble-depthai-examples, ros-humble-depthai-ros-msgs, ros-humble-diagnostic-msgs, ros-humble-diagnostic-updater, ros-humble-image-pipeline, ros-humble-image-transport, ros-humble-image-transport-plugins, ros-humble-pluginlib, ros-humble-rclcpp, ros-humble-rclcpp-components, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-std-srvs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 566 kB
APT-Sources: http://packages.ros.org/ros2-testing/ubuntu jammy/main arm64 Packages
Description: Depthai ROS Monolithic node.

N: There are 4 additional records. Please use the '-a' switch to see them.

hilary-luo, 2023-12-21:
Did you stop the turtlebot4 service prior to launching this? Only one node can connect to the camera at a time so running a second DepthAI node would cause errors.

sonnetx, 2023-12-23:
Thanks for the suggestion.

Now when I run this command (ros2 launch depthai_examples mobile_publisher.launch.py camera_model:=OAK-D-LITE) and launch rviz I get this.

[rviz2-1] [INFO] [1703337643.307606895] [rviz2]: Stereo is NOT SUPPORTED

I can't see any camera information from this node.

When I try to launch another example node with this command: ros2 launch depthai_examples stereo_inertial_node.launch.py

I get the following error:

[stereo_inertial_node-2] Device USB status: SUPER
[stereo_inertial_node-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[stereo_inertial_node-2]   what():  could not create publisher: rcl node's context is invalid, at ./src/rcl/node.c:428
[ERROR] [stereo_inertial_node-2]: process[stereo_inertial_node-2] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [stereo_inertial_node-2]: sending signal 'SIGTERM' to process[stereo_inertial_node-2]
[ERROR] [stereo_inertial_node-2]: process has died [pid 1662, exit code -6, cmd '/opt/ros/humble/lib/depthai_examples/stereo_inertial_node --ros-args --params-file /tmp/launch_params_vf9303ir --params-file /tmp/launch_params_dp82xalj --params-file /tmp/launch_params_xgmvbhrw --params-file /tmp/launch_params_w61388c1 --params-file /tmp/launch_params_sgbegky_ --params-file /tmp/launch_params_3466er18 --params-file /tmp/launch_params_p3txajt3 --params-file /tmp/launch_params_yx1rap94 --params-file /tmp/launch_params_7rrejbdb --params-file /tmp/launch_params_kl27do62 --params-file /tmp/launch_params_y85rz8_w --params-file /tmp/launch_params_6ur2e716 --params-file /tmp/launch_params_kmyx9yrn --params-file /tmp/launch_params_6f21byg2 --params-file /tmp/launch_params_0g_j6515 --params-file /tmp/launch_params_iu0jud_b --params-file /tmp/launch_params_4vii8ehf --params-file /tmp/launch_params_wiaqzig7 --params-file /tmp/launch_params_84bxrll2 --params-file /tmp/launch_params_pqc97tbw --params-file /tmp/launch_params_1rct54iw --params-file /tmp/launch_params_x5eicgnu --params-file /tmp/launch_params_o_jja0k7 --params-file /tmp/launch_params_r9u_wgok --params-file /tmp/launch_params_ut9ow81f --params-file /tmp/launch_params_a3d2c8y8 --params-file /tmp/launch_params_qapvngdg --params-file /tmp/launch_params_75civcy_ --params-file /tmp/launch_params_3f8j_uyl --params-file /tmp/launch_params_3i6pw7oe --params-file /tmp/launch_params_t1m4uygs --params-file /tmp/launch_params__045z1x0 --params-file /tmp/launch_params_goautdz3 --params-file /tmp/launch_params_ev6e72cr --params-file /tmp/launch_params_galk8ysr'].

The only command that works for me is this: ros2 launch turtlebot4_bringup oakd.launch.py
But even with this, stereo is not supported.

slowrunner, 2023-12-23:
I get that warning when successfully running the non-ROS depthai_demo.  Are there non-stereo examples to try?  (I don’t have a TB4, only the Oak-D-Lite on a Create3 and have not started investigating using it with ROS yet.)

sonnetx, 2023-12-24:
I mainly purchased the OAK-D Lite so that I could have depth sensing along with the basic functionalities of a camera, so having stereo working is something that I would love to have assistance on. 

I am wondering if the depth-ai package in ROS is the issue, since nodes launched from this package don't even have a basic camera feed, while the node launched from the turtlebot4_bringup package at least have a camera feed.

hilary-luo, 2023-12-27:
@sonnetx Are you running a turtlebot4? if not then this issue should be on the luxonis depthai repo. I am here to assist in troubleshooting issues with getting the Turtlebot4 oakd functionality working, not the depthai launch files.

That being said, this package [launches the depthai nodes](https://github.com/turtlebot/turtlebot4_robot/blob/humble/turtlebot4_bringup/launch/oakd.launch.py) in a similar way to the [camera.launch.py](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/launch/camera.launch.py) in the depthai repo.

There does seem to be some discrepancy in what you are reporting about whether you are launching the nodes for oak-d or oak-d lite. If this was just a matter of not typing it fully out then ignore this, but if it is something that you were not paying attention to, make sure that when you are comparing that you launch for the same camera each time. The turtlebot4 oakd launch does default to the oakd pro (not lite). 

If you are looking to launch the turtlebot4 oakd node with stereo (instead of just rgb) then that is something that I can help you with.

sonnetx, 2023-12-27:
I am running this on a turtlebot4. I am looking to launch the turtlebot4 oakd node with stereo (instead of just rgb) and I am wondering why this is not working.

Is the depthai_examples package supported by the turtlebot4? I would like to be able to run examples on there from the turtlebot4 but it seems they are not working right now.

sonnetx, 2023-12-27:
Could someone also explain the difference between the lite.launch.py and the oakd.launch.py? Both are able to give my turtlebot4 lite camera feed but don't support stereo. However. the lite.launch.py has many more terminal outputs and talks about the LiDAR as well.

[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-12-27-21-00-50-974761-ubuntu-1683
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [turtlebot4_node-1]: process started with pid [1696]
[INFO] [joy_linux_node-2]: process started with pid [1698]
[INFO] [teleop_node-3]: process started with pid [1700]
[INFO] [rplidar_composition-4]: process started with pid [1702]
[INFO] [component_container-5]: process started with pid [1704]
[INFO] [robot_state_publisher-6]: process started with pid [1714]
[INFO] [joint_state_publisher-7]: process started with pid [1723]
[turtlebot4_node-1] [INFO] [1703710854.424427580] [turtlebot4_node]: Init Turtlebot4 Node Main
[rplidar_composition-4] [INFO] [1703710854.779034236] [rplidar_composition]: RPLidar running on ROS2 package rplidar_ros. RPLIDAR SDK Version:2.0.0
[rplidar_composition-4] *** buffer overflow detected ***: terminated
[teleop_node-3] [INFO] [1703710855.241942063] [TeleopTwistJoy]: Teleop enable button 4.
[teleop_node-3] [INFO] [1703710855.242205357] [TeleopTwistJoy]: Turbo on button 5.
[teleop_node-3] [INFO] [1703710855.242238375] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.200000.
[teleop_node-3] [INFO] [1703710855.242268597] [TeleopTwistJoy]: Turbo for linear axis x is scale 0.500000.
[teleop_node-3] [INFO] [1703710855.242289893] [TeleopTwistJoy]: Angular axis yaw on 0 at scale 0.500000.
[teleop_node-3] [INFO] [1703710855.242308541] [TeleopTwistJoy]: Turbo for angular axis yaw is scale 1.900000.
[robot_state_publisher-6] [WARN] [1703710855.646364941] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-6] [INFO] [1703710855.650441585] [robot_state_publisher]: got segment base_link
[robot_state_publisher-6] [INFO] [1703710855.668143593] [robot_state_publisher]: got segment bump_front_center
[robot_state_publisher-6] [INFO] [1703710855.677845680] [robot_state_publisher]: got segment bump_front_left
[robot_state_publisher-6] [INFO] [1703710855.677904994] [robot_state_publisher]: got segment bump_front_right
[robot_state_publisher-6] [INFO] [1703710855.677933290] [robot_state_publisher]: got segment bump_left
[robot_state_publisher-6] [INFO] [1703710855.677960290] [robot_state_publisher]: got segment bump_right
[robot_state_publisher-6] [INFO] [1703710855.677986790] [robot_state_publisher]: got segment bumper
[robot_state_publisher-6] [INFO] [1703710855.678013604] [robot_state_publisher]: got segment button_1
[robot_state_publisher-6] [INFO] [1703710855.678039567] [robot_state_publisher]: got segment button_2
[robot_state_publisher-6] [INFO] [1703710855.678066048] [robot_state_publisher]: got segment button_power
[robot_state_publisher-6] [INFO] [1703710855.678091919] [robot_state_publisher]: got segment cliff_front_left
[robot_state_publisher-6] [INFO] [1703710855.678117585] [robot_state_publisher]: got segment cliff_front_right
[robot_state_publisher-6] [INFO] [1703710855.678148085] [robot_state_publisher]: got segment cliff_side_left
[robot_state_publisher-6] [INFO] [1703710855.678176658] [robot_state_publisher]: got segment cliff_side_right
[robot_state_publisher-6] [INFO] [1703710855.678202603] [robot_state_publisher]: got segment front_caster_link
[robot_state_publisher-6] [INFO] [1703710855.678228639] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-6] [INFO] [1703710855.678255287] [robot_state_publisher]: got segment ir_intensity_front_center_left
[robot_state_publisher-6] [INFO] [1703710855.678281713] [robot_state_publisher]: got segment ir_intensity_front_center_right
[robot_state_publisher-6] [INFO] [1703710855.678308435] [robot_state_publisher]: got segment ir_intensity_front_left
[robot_state_publisher-6] [INFO] [1703710855.678335120] [robot_state_publisher]: got segment ir_intensity_front_right
[robot_state_publisher-6] [INFO] [1703710855.678361286] [robot_state_publisher]: got segment ir_intensity_left
[robot_state_publisher-6] [INFO] [1703710855.678387082] [robot_state_publisher]: got segment ir_intensity_right
[robot_state_publisher-6] [INFO] [1703710855.678413397] [robot_state_publisher]: got segment ir_intensity_side_left
[robot_state_publisher-6] [INFO] [1703710855.678439378] [robot_state_publisher]: got segment ir_omni
[robot_state_publisher-6] [INFO] [1703710855.678466822] [robot_state_publisher]: got segment left_wheel
[robot_state_publisher-6] [INFO] [1703710855.678492581] [robot_state_publisher]: got segment mouse
[robot_state_publisher-6] [INFO] [1703710855.678517988] [robot_state_publisher]: got segment oakd_camera_bracket
[robot_state_publisher-6] [INFO] [1703710855.678543395] [robot_state_publisher]: got segment oakd_imu_frame
[robot_state_publisher-6] [INFO] [1703710855.678569914] [robot_state_publisher]: got segment oakd_left_camera_frame
[robot_state_publisher-6] [INFO] [1703710855.678596765] [robot_state_publisher]: got segment oakd_left_camera_optical_frame
[robot_state_publisher-6] [INFO] [1703710855.678623913] [robot_state_publisher]: got segment oakd_link
[... cut ...]
[turtlebot4_node-1] [INFO] [1703710895.612463463] [turtlebot4_node]: OAKD stopped
[turtlebot4_node-1] [INFO] [1703710895.612669776] [turtlebot4_node]: RPLIDAR stopped
[turtlebot4_node-1] [INFO] [1703710896.613005069] [turtlebot4_node]: oakd/stop_camera service available, sending request
[turtlebot4_node-1] [ERROR] [1703710925.622876647] [turtlebot4_node]: Service stop_motor unavailable.

I am also confused as to what oakd_example.launch.py does. It doesn't seem to work on my turtlebot4 lite either. Thanks!

hilary-luo, 2023-12-28:
> Could someone also explain the difference between the `lite.launch.py` and the `oakd.launch.py`? Both are able to give my turtlebot4 lite camera feed but don't support stereo. However. the `lite.launch.py` has many more terminal outputs and talks about the LiDAR as well.

You are correct that both `lite.launch.py` and `oakd.launch.py` should give you your camera feed but won't launch all of the same nodes / topics and neither will support stereo out of the box (I will come back to how to set up stereo after answering your other questions). Essentially, the launch files are nested. There are individual launch files that launch each component for the robot (for example the `oakd.launch.py` launching the camera node) and then there are higher level launch files that collect together and call on those individual launch files. So, in this case, `lite.launch.py` is the top level launch file that launches every node for the Turtlebot4 Lite robot. This launch file is launched automatically by the service so that when you turn your robot on, all of those topics will just be there already and the robot will be ready to use. So when this runs, it will in turn run the `oakd.launch.py` file as well as the launch files for the lidar etc. It will also send certain parameters to those launch files for example telling the `oakd.launch.py` file to launch for an Oakd Lite instead of the Pro.

> I am also confused as to what `oakd_example.launch.py` does. It doesn't seem to work on my turtlebot4 lite either. Thanks!

The built in oakd_example is outdated, we should either update it or remove it. 

> Is the depthai_examples package supported by the turtlebot4? I would like to be able to run examples on there from the turtlebot4 but it seems they are not working right now.

It's not a question of the turtlebot4 supporting the depthai packages. It's better to think about the depthai package as a building block that was used to build the turtlebot4 packages. And as soon as you stop the turtlebot4 service, you are just running those examples on a raspberry pi with the camera connected. Most likely the issue that you are seeing are related to not providing the inputs that the package needs in order to successfully find and connect to the camera. 

To enable stereo you will need to modify the [`oakd_lite.yaml`](https://github.com/turtlebot/turtlebot4_robot/blob/humble/turtlebot4_bringup/config/oakd_lite.yaml) parameters that are passed into the launch file. Either you can have a local copy if you want to launch the camera manually and pass it in each time or you can modify the packages that are launched by the services to enable it automatically every time your robot boots up. If you want to do the second option, take a look at the instructions I shared in this issue: https://github.com/turtlebot/turtlebot4/issues/277#issuecomment-1815024838

closed 2024-02-13 by hilary-luo