turtlebot/turtlebot4 #332: Slow Action Server Response

LuHaofan, 2023-12-31:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Not sure which package is causing the problem. It seems that it's the issue with the create 3 firmware.

### Type of issue

Other

### Expected behaviour

I am trying to drive my turtlebot with the action server using the instructions in the documentation. Specifically, I call this command in my CLI:
```
ros2 action send_goal /drive_distance irobot_create_msgs/action/DriveDistance \
"distance: 0.5
max_translation_speed: 0.3"
```

I expect the turtlebot to receive the action request and start conducting the action with very low delay (< 1 sec). As it is the amount of delay for the action server I wrote myself and installed on the raspberry pi on the turtlebot. 

### Actual behaviour

The prompt shows: `Waiting for an action server to become available...`
And then it takes 20~30 seconds before I actually get a response from the robot and it starts to move.
The delay is mostly 20~30 seconds but also varies from time to time and can even go up to several minutes.
This delay is way too long for my application to bear and I wonder what's the reason for such a delay and how to deal with this issue?

### Error messages

```bash
There is no error message. The robot works as expected after this 20~30 sec delay. As shown in the sample log using `time` utility. 

$ time ros2 action send_goal /drive_distance irobot_create_msgs/action/DriveDistance \
"distance: 0.5
max_translation_speed: 0.3"
Waiting for an action server to become available...
Sending goal:
     distance: 0.5
max_translation_speed: 0.3

Goal accepted with ID: 24f6968adcbf45628fe258bfb1c97043

Result:
    pose:
  header:
    stamp:
      sec: 1704053771
      nanosec: 843717783
    frame_id: odom
  pose:
    position:
      x: -1.6398375034332275
      y: 0.0217540692538023
      z: 0.005418745800852776
    orientation:
      x: 5.164551811991156e-05
      y: -0.005196247953536115
      z: 0.9999670114623728
      w: 0.006242781988551551

Goal finished with status: SUCCEEDED
ros2 action send_goal /drive_distance irobot_create_msgs/action/DriveDistance  0.60s user 0.71s system 6% cpu 20.944 total
```
```

### To Reproduce

1. Start up the turtlebot and undock.
2. Open a terminal and run 
```
ros2 action send_goal /drive_distance irobot_create_msgs/action/DriveDistance \
"distance: 0.5
max_translation_speed: 0.3"
```
measure the delay using the time utility 

### Other notes

Here are some additional information regrading my setup and attempts to debug the issue:

1. I am using a WLAN network where only one turtlebot is present.
2. I am using ROS2 Humble and the latest firmware from the irobot create3 website: H.2.3
3. I have sync the clock on my PC and Rasp Pi and icreate 3 using NTP. (But I still see user.notice ntpd: ntpd: reply from [192.168.186.3](http://192.168.186.3/): delay 0.004474 is too high, ignoring in the create 3 log. Not sure how to fix it...)
4. I tried both CLI and rclpy action client. The delays are similar, both 20~30 seconds.
5. I log the time at which the goal request is sent and the create3 receives the goal message. The delay between these two is around 20~30 seconds. After the goal message is received, the robot starts actuating quickly.
6. I tried the other action servers available on create 3, such as rotate_angle, drive_arc, and all of them suffer from this delay issue.
7. I wrote an action server myself and ran it on a Rasp Pi in the same network and interacted with it using the same way. The delay is negligible.
8. I monitor the WiFi traffic and notice that create 3 has a lot of uplink data (0.6Mbps) and downlink data (6 kbps) from the create3 robot. This throughput should not saturate the channel and cause the delay issue.
9. I tried to ssh into raspberry pi and directly run this action client command from the raspberry pi. The resulting delay is similar.

smatar22, 2024-01-02:
Hello,
Thank you for reaching out on the Turtlebot 4 Github page. I am very sorry to hear that you are encountering difficulties with one of our products. Is your current setup using the simple discovery method of connecting the Create3 and Raspberry Pi or the Discovery server method? As you mentioned this being the only Turtlebot on the network, the option to use the discovery server method is available to you. This method allows for faster communication between the Create3 and Raspberry Pi, using the USB-C connection between the two. Information on setting up the discovery server can be found on our online user manual here: https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html  

Let me know how this recommendation works for you. Looking forward to hearing from you soon.

LuHaofan, 2024-01-02:
Hi @smatar22 , 

Thank you so much for the information. I have tried to configure the network settings to discovery server and it helps to reduce the delay significantly. It works for me now.

smatar22, 2024-01-03:
Hello @LuHaofan,
Happy to hear that this recommendation resolved the matter. I will go ahead and close this ticket. Should you need any further assistance, please feel free to open a new ticket.

closed 2024-01-03 by smatar22

LuHaofan, 2024-01-03:
Sorry, I realized that once I put the network setting into discovery server mode, my PC is not able to find other nodes running on other machines than turtlebot. This is a problem for me. Although I have only one turtlebot on in the network. I have other two SBCs running other ros nodes. And I want to be able to talk to them. Is there any recommendations for this setting? Thanks.

scd5027, 2024-05-31:
I'm having this same issue. Using simple discovery shouldn't cause these large delays.. The networks are bridged. There is something using up resources.

scd5027, 2024-07-19:
My setup is the exact same but I can create a duplicate issue if that's preferable