turtlebot/turtlebot4 #333: The display and LED do not turn on, and wired/wireless connections do not work.

gmchoi96, 2024-01-03:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Select One

### Package version

22.04 ros2 humble

### Type of issue

Other

### Expected behaviour

d

### Actual behaviour

When I turn on Turtlebot, the LED light and display do not come on and wired/wireless connection is not possible. It seems to be a hardware problem.

[image]

https://github.com/turtlebot/turtlebot4/assets/149559257/a921921d-0585-42ea-8548-d92e6f3ff8d2

### To Reproduce

s

RustyCPR, 2024-01-09:
Hi @gmchoi96 
Can you please verify if the LCB ribbon cable is securely connected? Its loops underneath the PCB. 

[image]

SangHyun-94, 2024-01-10:
@RustyCPR 
Hi 
I checked  LCB ribbon cable and it does not work even though I disconnected the cable and put it back in
it does not work

RustyCPR, 2024-01-10:
Thanks @SangHyun-94  
I'll process an RMA for a replacement LCD. I have an existing support email with ROAS that I will reply to. 
Regards,
Rusty

closed 2024-01-10 by RustyCPR