turtlebot/turtlebot4 #334: SLAM Stuck at Registering sensor under discovery server mode

LuHaofan, 2024-01-04:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

turtlebot4_navigation

### Type of issue

Navigation (SLAM, Nav2 etc.)

### Expected behaviour

I recently change my networking setting from simple discovery to discovery server for faster communication. After I made the change, I noticed that SLAM stuck at `registering sensor` step for extremely long time (200 seconds), which didn't happen with the simple discovery configuration. Under the simple discovery setting, the slam runs smoothly after launch without any stuck. That's my expected behavior.

### Actual behaviour

After launch slam using the command: 
```
ros2 launch turtlebot4_navigation slam.launch.py params:=./config/slam.yaml
```
The program stuck at the output line:
```
[sync_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
```
for more than 200 seconds. Then it continues to run as expected.

### Error messages

```bash
ros2 launch turtlebot4_navigation slam.launch.py params:=./config/slam.yaml
[INFO] [launch]: All log files can be found below /home/haofan/.ros/log/2024-01-03-17-01-22-124628-icon-13052
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [13053]
[sync_slam_toolbox_node-1] [INFO] [1704330082.229577417] [slam_toolbox]: Node using stack size 80000000
[sync_slam_toolbox_node-1] [INFO] [1704330082.246984131] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1704330082.247627371] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[sync_slam_toolbox_node-1] [INFO] [1704330083.427375852] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1704330083.171 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1704330083.563015511] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1704330083.289 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] [INFO] [1704330083.685930631] [slam_toolbox]: Message Filter dropping message: frame 'rplidar_link' at time 1704330083.424 for reason 'discarding message because the queue is full'
[sync_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[sync_slam_toolbox_node-1] [WARN] [1704330284.289655157] [slam_toolbox]: Queue size has grown to: 46. Recommend stopping until message is gone if online mapping.
[sync_slam_toolbox_node-1] [WARN] [1704330284.298648275] [slam_toolbox]: Queue size has grown to: 45. Recommend stopping until message is gone if online mapping.
[sync_slam_toolbox_node-1] [WARN] [1704330284.307673211] [slam_toolbox]: Queue size has grown to: 44. Recommend stopping until message is gone if online mapping.
[sync_slam_toolbox_node-1] [WARN] [1704330284.316771733] [slam_toolbox]: Queue size has grown to: 43. Recommend stopping until message is gone if online mapping.
[sync_slam_toolbox_node-1] [WARN] [1704330284.325959962] [slam_toolbox]: Queue size has grown to: 42. Recommend stopping until message is gone if online mapping.
[sync_slam_toolbox_node-1] [WARN] [1704330284.335104685] [slam_toolbox]: Queue size has grown to: 41. Recommend stopping until message is gone if online mapping.
[sync_slam_toolbox_node-1] [WARN] [1704330284.344264465] [slam_toolbox]: Queue size has grown to: 40. Recommend stopping until message is gone if online mapping.
[sync_slam_toolbox_node-1] [WARN] [1704330284.354151770] [slam_toolbox]: Queue size has grown to: 39. Recommend stopping until message is gone if online mapping.
[sync_slam_toolbox_node-1] [WARN] [1704330284.363866336] [slam_toolbox]: Queue size has grown to: 38. Recommend stopping until message is gone if online mapping.
```

### To Reproduce

1. Set the turtlebot in to discovery server mode.
2. Run slam to generate map using command
```
ros2 launch turtlebot4_navigation slam.launch.py params:=./config/slam.yaml
```
Monitor delay at the line
```
[sync_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
```

smatar22, 2024-01-05:
Hello Haofan,
Out of curiosity what network are you using to connect you computer to the Turtlebot 4? Are you using a mobile hotspot or do you have a dedicated router? I Are you using a 2.4 GHz network or 5 GHz network? I have seen instances of this error message in the past, and the matter was resolved by using a dedicated router using a 5 GHz network. If you are using a dedicated network, are there other devices on the network causing heavy network traffic? Looking forward to hearing from you soon.

LuHaofan, 2024-01-05:
Greetings,

Thank you for your response. I am using a dedicated router using a 5GHz
WiFi network. There are only limited devices (~6 including the robot)
registered on this router. The delay issue is less severe after I reboot my
PC (reduce to about 1 min at the registering sensor step), but still much
longer than when I was running SLAM under simple discovery mode.

Hope this information helps.

By the way, I am planning to purchase a replacement battery for my
turtlebot. I wonder if I can charge it out of the robot using a separate
charger. The reason is that my robot needs to execute tasks that take a
long time. I want to charge one battery while the robot is executing the
task and swap the battery when the robot is out of battery. I couldn't find
any external charger for the battery only. Any advice would be appreciated.

Best,
Haofan

smatar22, 2024-01-10:
Hello Haofan,
Would it be possible to check the following:

- Which version of the firmware is running on the Create3. For ROS Humble the latest firmware build is vH.2.3 and can be found here - https://iroboteducation.github.io/create3_docs/releases/overview/
- Would it be possible to share the logs generated by the Create3 from its webserver?
- Check the CPU usage on the Raspberry Pi by using HTOP. I want to ensure that the the CPU is not being overloaded with requests.

With regards to the battery - I am not aware of any external charging options, but will verify this with the team and update you.

Looking forward to hearing from you soon.
Best,
Saif

LuHaofan, 2024-01-11:
Hi Saif,

Thank you for your reply. I confirmed that the firmware version is the
latest. The issue somehow disappeared by itself. I have not been in the lab
in the last few days. I will keep you updated.

Best,
Haofan

smatar22, 2024-01-11:
Hello Haofan,
Glad to hear that the issue was resolved. Do reach out if you have any further updates.
Best Regards,
Saif

smatarCPR, 2024-02-12:
Hello Haofan,
It has been some time since we last discussed this matter. I will go ahead and close this ticket. Should you encounter an issue feel free to open a new one on the Github issues page.
Best Regards,
Saif

closed 2024-02-12 by smatarCPR