turtlebot/turtlebot4 #337: Problem in the /scan topic of turtlebot4 Lite

yoraoui, 2024-01-08:
### Robot Model

Turtlebot4 Lite

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

humble

### Expected behaviour

Hello, I expect that the /scan topic returns the data of the Lidar. 

### Actual behaviour

I don't receive any data from the lidar after I run these commands:
ros2 service call /stop_motor std_srvs/srv/Empty {}
ros2 service call /stop_motor std_srvs/srv/Empty {}

Also the Lidar spinning interrupts. 

### Error messages

```bash
Jan 07 12:41:57 ubuntu turtlebot4-start[1074]: [rplidar_composition-4] [ERROR] [1704631317.857308900] [rplidar_composition]: Failed to set scan mode
Jan 07 12:41:59 ubuntu turtlebot4-start[1074]: [rplidar_composition-4] [INFO] [1704631319.858090337] [rplidar_composition]: Start
```

### To Reproduce

[image]

yoraoui, 2024-01-08:
> @yoraoui <https://github.com/yoraoui> If you stop the lidar then it will
> be stopped and will not report data anymore. Start the lidar motor to
> resume getting data. Reassigning this issue as a troubleshooting issue.

I stopped and restarted with

ros2 service call /start_motor std_srvs/srv/Empty {}

However, the issue is still happening.

RustyCPR, 2024-01-16:
Hi @yoraoui,
What occurs when you run the lidar with the `command:` `ros2 launch turtlebot4_bringup rplidar.launch.py`  ? 

If you still don't see anything on the `/scan ` topic can you please verify the RPLIDAR UDEV rules exist and try installing it again by following these instructions: https://turtlebot.github.io/turtlebot4-user-manual/software/sensors.html#connecting

closed 2024-02-09 by RustyCPR