turtlebot/turtlebot4 #340: Turtlebot4 Raspberry Pi wifi connect fail

iyeon915, 2024-01-12:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

sudo apt update && sudo apt install ros-humble-turtlebot4-desktop

### Type of issue

Networking

### Expected behaviour

1 - turn on turtlebot4
2 - connect to "turtlebot4" wifi on PC
3 - "ssh ubuntu@10.42.0.1" to configure wifi setup on PC
4 - "turtlebot4-setup" & (Access point -> client) & change SSID & change Passward & click save, then click Apply setting
5 - Done

### Actual behaviour

1 - turn on turtlebot4
2 - connect to "turtlebot4" wifi on PC
3 - "ssh ubuntu@10.42.0.1" to configure wifi setup on PC
4 - "turtlebot4-setup" & (Access point -> client) & change SSID & change Passward & click save, then click Apply setting
5 (fail point) - return error message "client_loop: send disconnect: Broken pipe" on PC
5.1 - IP: UNKNOWN on Turtlebot
[image]

### Error messages

```bash
client_loop: send disconnect: Broken pipe
```

### To Reproduce

explanated in Actual behaviour

iyeon915, 2024-01-12:
Additional:

After the error have been occured, the turtlebot4 wifi is removed, so I need to connect with LAN cable directly or reinstall Raspberry Pi image.

1. When connecting with LAN cable directly:
Followed the user manual, the error "ssh: connect to host 192.168.185.3 port 22: No route to host" is occured when connect to turtlebot4 using ssh.

2. Reinstall Raspberry Pi image:
The error described above is reproduced.

smatar22, 2024-01-16:
Hello,
Is the wi-fi network you are connecting to an institutional wi-fi network with additional security or is the security limited to a wpa2 password?

Is this behaviour repeated across all networks or this specific network?

Looking forward to hearing from you soon.
Best Regards,
Saif

DDJared, 2024-01-26:
I've just replicated the same behavior with the same configuration. Unknown IP as a result of using turtlebot4-setup and loss of wifi broadcast after using person wpa2 home network

smatarCPR, 2024-03-04:
Hello @iyeon915 and @DDJared ,
Apologies for the late reply on this. A quick note on the error: `client_loop: send disconnect: Broken pipe` Seeing this after saving and applying the settings to Turtlebot4 setup is normal, this pops up as a result of the Turtlebot4 Access Point being disabled when you set the Turtlebot 4 from ap mode to client mode.

Regarding the unknown IP address, are you able to SSH into the raspberry pi using an ethernet connection? Double check to ensure that the computer is set to use the 185 subnet. Alternatively you can also disable Wi-Fi on your computer to ensure that the ethernet interface is used to connect to the Pi.

Looking forward to hearing from you soon.
Best Regards,
Saif

smatarCPR, 2024-03-18:
Hello @iyeon915 and @DDJared ,
Any updates regarding this? Looking forward to hearing from you soon.
Best Regards,
Saif

closed 2024-03-25 by smatarCPR