turtlebot/turtlebot4 #341: My TurtleBot 4 cannot 2D navigation

hhhhhhwww, 2024-01-17:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Built from Source

### Package version

git clone https://github.com/turtlebot/turtlebot4.git -b humble

### Expected behaviour

I am using a discovery server for communication. When I use 'ros2 topic list', Ubuntu can display all the topics, but when I execute 'ros2 launch', I am unable to correctly execute nav2. In rviz2, it shows as not activated.In rviz, the topics for the global map and the local map are showing 'no map received'."，Please tell me what I should do. Is it a problem with my DDS configuration? Thank you very much.

### Actual behaviour

ros2 launch turtlebot4_navigation nav2.launch.py 
[lifecycle_manager-8] [INFO] [1705482718.180098182] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1705482720.180507083] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1705482722.181198584] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1705482724.181589784] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[image]

### Error messages

```bash
ros2 launch turtlebot4_navigation nav2.launch.py 
[lifecycle_manager-8] [INFO] [1705483318.149548081] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[ERROR] [waypoint_follower-6]: process has died [pid 4634, exit code -6, cmd '/opt/ros/humble/lib/nav2_waypoint_follower/waypoint_follower --ros-args --log-level info --ros-args -r __node:=waypoint_follower -r __ns:=/ --params-file /tmp/launch_params_tnpps2vk -r /global_costmap/scan:=/scan -r /local_costmap/scan:=/scan -r /tf:=tf -r /tf_static:=tf_static'].
[ERROR] [controller_server-1]: process has died [pid 4624, exit code -6, cmd '/opt/ros/humble/lib/nav2_controller/controller_server --ros-args --log-level info --ros-args -r __ns:=/ --params-file /tmp/launch_params_tnpps2vk -r /global_costmap/scan:=/scan -r /local_costmap/scan:=/scan -r /tf:=tf -r /tf_static:=tf_static -r cmd_vel:=cmd_vel_nav'].
[lifecycle_manager-8] [INFO] [1705483320.150375534] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[ERROR] [planner_server-3]: process has died [pid 4628, exit code -6, cmd '/opt/ros/humble/lib/nav2_planner/planner_server --ros-args --log-level info --ros-args -r __node:=planner_server -r __ns:=/ --params-file /tmp/launch_params_tnpps2vk -r /global_costmap/scan:=/scan -r /local_costmap/scan:=/scan -r /tf:=tf -r /tf_static:=tf_static'].
```

### To Reproduce

ros2 launch turtlebot4_navigation localization.launch.py map:=532.yaml
ros2 launch turtlebot4_navigation nav2.launch.py
ros2 launch turtlebot4_viz view_robot.launch.py
[image]

### Other notes

no

hhhhhhwww, 2024-01-17:
[image]

hilary-luo, 2024-01-18:
This issue is being reassigned as a troubleshooting issue. Please be advised that for the Turtlebot4, the "bug" template is reserved for when a particular bug has been identified that requires a software patch in this particular repository. Your issue seems to be a request for assistance regarding the configuration and launching of your software. 

To assist our team in helping you please:
1. go back and indicate the package version properly. If you are building from source then please provide the commit hash for the version that you are using.
2. Go through this tutorial: https://turtlebot.github.io/turtlebot4-user-manual/tutorials/navigation.html and confirm that you have done all of the steps indicated there in order including setting your pose estimate. If you are unable to complete the tutorial then indicate where it starts to deviate and what you see in comparison to what you should be seeing so we can understand where things begin to go wrong.
3. Finally, when providing error messages, provide full terminal output from all of the commands involved (rviz and localization as well as nav2).

hhhhhhwww, 2024-01-19:
I use  building from source, the version is 1.0.4.
yes,I complete this tutorial,but always can't implement 2D nav2 in rviz2 ,pose estimate no problem

hhhhhhwww, 2024-01-19:
ros2 launch turtlebot4_navigation localization.launch.py map:=532.yaml

[INFO] [launch]: All log files can be found below /home/lxx/.ros/log/2024-01-19-09-52-41-382236-lxx-vm-19815
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [19816]
[INFO] [amcl-2]: process started with pid [19818]
[INFO] [lifecycle_manager-3]: process started with pid [19820]
[lifecycle_manager-3] [INFO] [1705629161.677503246] [lifecycle_manager_localization]: Creating
[lifecycle_manager-3] [INFO] [1705629161.679955138] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[map_server-1] [INFO] [1705629161.685868611] [map_server]: 
[map_server-1] 	map_server lifecycle node launched. 
[map_server-1] 	Waiting on external lifecycle transitions to activate
[map_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1705629161.686071834] [map_server]: Creating
[amcl-2] [INFO] [1705629161.685959207] [amcl]: 
[amcl-2] 	amcl lifecycle node launched. 
[amcl-2] 	Waiting on external lifecycle transitions to activate
[amcl-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[amcl-2] [INFO] [1705629161.686737734] [amcl]: Creating
[lifecycle_manager-3] [INFO] [1705629163.267259825] [lifecycle_manager_localization]: Starting managed nodes bringup...
[lifecycle_manager-3] [INFO] [1705629163.267386735] [lifecycle_manager_localization]: Configuring map_server
[map_server-1] [INFO] [1705629163.268202032] [map_server]: Configuring
[map_server-1] [INFO] [map_io]: Loading yaml file: /home/lxx/turtlebot4_ws/install/turtlebot4_navigation/share/turtlebot4_navigation/maps/warehouse.yaml
[map_server-1] [DEBUG] [map_io]: resolution: 0.03
[map_server-1] [DEBUG] [map_io]: origin[0]: -15.1
[map_server-1] [DEBUG] [map_io]: origin[1]: -25
[map_server-1] [DEBUG] [map_io]: origin[2]: 0
[map_server-1] [DEBUG] [map_io]: free_thresh: 0.1
[map_server-1] [DEBUG] [map_io]: occupied_thresh: 0.65
[map_server-1] [DEBUG] [map_io]: mode: trinary
[map_server-1] [DEBUG] [map_io]: negate: 0
[map_server-1] [INFO] [map_io]: Loading image_file: /home/lxx/turtlebot4_ws/install/turtlebot4_navigation/share/turtlebot4_navigation/maps/warehouse.pgm
[map_server-1] [DEBUG] [map_io]: Read map /home/lxx/turtlebot4_ws/install/turtlebot4_navigation/share/turtlebot4_navigation/maps/warehouse.pgm: 1006 X 1674 map @ 0.03 m/cell
[lifecycle_manager-3] [INFO] [1705629163.448904555] [lifecycle_manager_localization]: Configuring amcl
[amcl-2] [INFO] [1705629163.532874348] [amcl]: Configuring
[amcl-2] [INFO] [1705629163.533024601] [amcl]: initTransforms
[amcl-2] [INFO] [1705629163.558478657] [amcl]: initPubSub
[amcl-2] [INFO] [1705629163.569745066] [amcl]: Subscribed to map topic.
[lifecycle_manager-3] [INFO] [1705629163.693057794] [lifecycle_manager_localization]: Activating map_server
[map_server-1] [INFO] [1705629163.693528080] [map_server]: Activating
[map_server-1] [INFO] [1705629163.694750555] [map_server]: Creating bond (map_server) to lifecycle manager.
[amcl-2] [INFO] [1705629163.811694387] [amcl]: Received a 1006 X 1674 map @ 0.030 m/pix
[lifecycle_manager-3] [INFO] [1705629164.009767737] [lifecycle_manager_localization]: Server map_server connected with bond.
[lifecycle_manager-3] [INFO] [1705629164.009837448] [lifecycle_manager_localization]: Activating amcl
[amcl-2] [INFO] [1705629164.010242607] [amcl]: Activating
[amcl-2] [INFO] [1705629164.010289796] [amcl]: Creating bond (amcl) to lifecycle manager.
[lifecycle_manager-3] [INFO] [1705629164.312422793] [lifecycle_manager_localization]: Server amcl connected with bond.
[lifecycle_manager-3] [INFO] [1705629164.312551435] [lifecycle_manager_localization]: Managed nodes are active
[lifecycle_manager-3] [INFO] [1705629164.312573327] [lifecycle_manager_localization]: Creating bond timer...
[amcl-2] [INFO] [1705629165.447891952] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1705629163.046 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[amcl-2] [INFO] [1705629165.582237575] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1705629163.194 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[amcl-2] [INFO] [1705629165.729234480] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1705629163.328 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[amcl-2] [INFO] [1705629165.848045522] [amcl]: createLaserObject
[amcl-2] [WARN] [1705629166.231590617] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[amcl-2] [INFO] [1705629166.231766238] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1705629163.475 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[amcl-2] [WARN] [1705629168.364309287] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
[amcl-2] [INFO] [1705629696.200439455] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1705629693.807 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
[amcl-2] [INFO] [1705629696.347382148] [amcl]: Message Filter dropping message: frame 'rplidar_link' at time 1705629693.942 for reason 'the timestamp on the message is earlier than all the data in the transform cache'

hhhhhhwww, 2024-01-19:
ros2 launch turtlebot4_navigation nav2.launch.py

[INFO] [launch]: All log files can be found below /home/lxx/.ros/log/2024-01-19-09-55-51-139659-lxx-vm-20019
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [20022]
[INFO] [smoother_server-2]: process started with pid [20024]
[INFO] [planner_server-3]: process started with pid [20026]
[INFO] [behavior_server-4]: process started with pid [20028]
[INFO] [bt_navigator-5]: process started with pid [20030]
[INFO] [waypoint_follower-6]: process started with pid [20032]
[INFO] [velocity_smoother-7]: process started with pid [20034]
[INFO] [lifecycle_manager-8]: process started with pid [20036]
[smoother_server-2] [ERROR] [1705629351.512110213] [rcl]: Failed to parse global arguments
[smoother_server-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[planner_server-3] [ERROR] [1705629351.516292793] [rcl]: Failed to parse global arguments
[planner_server-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[planner_server-3]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[waypoint_follower-6] [ERROR] [1705629351.517754288] [rcl]: Failed to parse global arguments
[waypoint_follower-6] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[waypoint_follower-6]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[smoother_server-2]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[controller_server-1] [ERROR] [1705629351.519623687] [rcl]: Failed to parse global arguments
[controller_server-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[controller_server-1]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[velocity_smoother-7] [ERROR] [1705629351.532404951] [rcl]: Failed to parse global arguments
[velocity_smoother-7] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[velocity_smoother-7]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[bt_navigator-5] [ERROR] [1705629351.535321218] [rcl]: Failed to parse global arguments
[bt_navigator-5] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[bt_navigator-5]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[behavior_server-4] [ERROR] [1705629351.541072020] [rcl]: Failed to parse global arguments
[behavior_server-4] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[behavior_server-4]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[lifecycle_manager-8] [INFO] [1705629351.544029018] [lifecycle_manager_navigation]: Creating
[lifecycle_manager-8] [INFO] [1705629351.547216361] [lifecycle_manager_navigation]: [34m[1mCreating and initializing lifecycle service clients[0m[0m
[ERROR] [planner_server-3]: process has died [pid 20026, exit code -6, cmd '/opt/ros/humble/lib/nav2_planner/planner_server --ros-args --log-level info --ros-args -r __node:=planner_server -r __ns:=/ --params-file /tmp/launch_params_pl1tbamm -r /global_costmap/scan:=/scan -r /local_costmap/scan:=/scan -r /tf:=tf -r /tf_static:=tf_static'].
[lifecycle_manager-8] [INFO] [1705629353.550370405] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[ERROR] [controller_server-1]: process has died [pid 20022, exit code -6, cmd '/opt/ros/humble/lib/nav2_controller/controller_server --ros-args --log-level info --ros-args -r __ns:=/ --params-file /tmp/launch_params_pl1tbamm -r /global_costmap/scan:=/scan -r /local_costmap/scan:=/scan -r /tf:=tf -r /tf_static:=tf_static -r cmd_vel:=cmd_vel_nav'].
[ERROR] [bt_navigator-5]: process has died [pid 20030, exit code -6, cmd '/opt/ros/humble/lib/nav2_bt_navigator/bt_navigator --ros-args --log-level info --ros-args -r __node:=bt_navigator -r __ns:=/ --params-file /tmp/launch_params_pl1tbamm -r /global_costmap/scan:=/scan -r /local_costmap/scan:=/scan -r /tf:=tf -r /tf_static:=tf_static'].
[lifecycle_manager-8] [INFO] [1705629355.550851406] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[... cut ...]
[lifecycle_manager-8] Failed to publish log message to rosout: publisher's context is invalid, at ./src/rcl/publisher.c:389
[lifecycle_manager-8] [INFO] [1705629397.272507991] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1705629397.322761664] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1705629397.373107632] [lifecycle_manager_navigation]: Waiting for service controller_server/get_state...

hhhhhhwww, 2024-01-19:
ros2 launch turtlebot4_viz view_robot.launch.py

[INFO] [launch]: All log files can be found below /home/lxx/.ros/log/2024-01-19-09-59-00-400641-lxx-vm-20442
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [20446]
[rviz2-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[rviz2-1] [INFO] [1705629541.278261950] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1705629541.278521689] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[rviz2-1] [INFO] [1705629541.318901730] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1705629542.248952221] [rviz2]: Trying to create a map of size 1006 x 1674 using 1 swatches
[rviz2-1] [ERROR] [1705629543.102761777] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
[rviz2-1] active samplers with a different type refer to the same texture image unit
[rviz2-1] [INFO] [1705629543.786210127] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629540.834 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629543.914395891] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629541.180 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629544.074520259] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629541.251 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629544.202556781] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629541.386 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629544.329830221] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629541.521 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629544.490511608] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629541.668 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629544.617800849] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629541.803 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629544.778009370] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629541.937 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629544.906243009] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629542.085 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629545.065771749] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629542.220 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629545.194636312] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629542.355 for reason 'discarding message because the queue is full'
[rviz2-1] [INFO] [1705629545.322033570] [rviz2]: Message Filter dropping message: frame 'rplidar_link' at time 1705629542.503 for reason 'discarding message because the queue is full'
[rviz2-1] 
[rviz2-1] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[rviz2-1] This error state is being overwritten:
[rviz2-1] 
[rviz2-1]   'rcl node's context is invalid, at ./src/rcl/node.c:428'
[rviz2-1] 
[rviz2-1] with this new error message:
[rviz2-1] 
[rviz2-1]   'the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/wait.c:130'
[rviz2-1] 
[rviz2-1] rcutils_reset_error() should be called after error handling to avoid this.
[rviz2-1] <<<
[rviz2-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[rviz2-1]   what():  failed to initialize wait set: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at ./src/rcl/wait.c:130
[ERROR] [rviz2-1]: process has died [pid 20446, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/lxx/turtlebot4_ws/install/turtlebot4_viz/share/turtlebot4_viz/rviz/robot.rviz --ros-args -r __node:=rviz2 -r __ns:=/ --params-file /tmp/launch_params_cbxhwkxz -r /tf:=tf -r /tf_static:=tf_static'].

hhhhhhwww, 2024-01-19:
looking forward to your reply

hhhhhhwww, 2024-01-19:
[image]
[image]
[image]

smatarCPR, 2024-02-05:
Hello,
Thank you for reaching out on the Turtlebot4 Github issues page. From the logs provided, there appears to be a repeated error throughout:
```
[smoother_server-2] [ERROR] [1705629351.512110213] [rcl]: Failed to parse global arguments
[smoother_server-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[planner_server-3] [ERROR] [1705629351.516292793] [rcl]: Failed to parse global arguments
[planner_server-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[planner_server-3]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[waypoint_follower-6] [ERROR] [1705629351.517754288] [rcl]: Failed to parse global arguments
[waypoint_follower-6] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[waypoint_follower-6]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[smoother_server-2]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[controller_server-1] [ERROR] [1705629351.519623687] [rcl]: Failed to parse global arguments
[controller_server-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[controller_server-1]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[velocity_smoother-7] [ERROR] [1705629351.532404951] [rcl]: Failed to parse global arguments
[velocity_smoother-7] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[velocity_smoother-7]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[bt_navigator-5] [ERROR] [1705629351.535321218] [rcl]: Failed to parse global arguments
[bt_navigator-5] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[bt_navigator-5]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /tmp/launch_params_pl1tbamm'. Error: Cannot have a value before ros__parameters at line 85, at ./src/parse.c:805, at ./src/rcl/arguments.c:406
[behavior_server-4] [ERROR] [1705629351.541072020] [rcl]: Failed to parse global arguments
```

It appears that line 85 on a custom launch params file is invalid and cannot be parsed. Out of curiosity have you made any modifications to the launch files on the Turtlebot 4 or your computer? If you know which file this error is referring to what is being parsed on line 85?

Looking forward to hearing from you soon.
Best Regards,
Saif

hhhhhhwww, 2024-02-26:
I'm not sure if I made any changes to the launch file. But when I run Nav2 Goal, it won't start. Can I download the launch file again？
Looking forward to hearing from you soon.

smatarCPR, 2024-03-04:
Hello,
Apologies for the late reply on this. Out of curiosity, would it be possible to clarify the decision to install the packages from source?

If possible try reinstalling the packages using the debian option and see if the issue persists. Looking forward to hearing from you soon.
Best Regards,
Saif

hhhhhhwww, 2024-03-08:
Yes,I ran the debian version.It sometimes runs smoothly,but sometimes it doesn't work properly with nav2. I suspect it's my computer configuration that doesn't work. Here's what warn appears when I run normally
[image]

hhhhhhwww, 2024-03-08:
[image]

smatarCPR, 2024-03-11:
Hello @hhhhhhwww ,
Would it be possible to resend the screenshot? The current link appears to be broken and links back to the ticket. Looking forward to hearing from you soon.
Best Regards,
Saif

hhhhhhwww, 2024-03-12:
[image]

hhhhhhwww, 2024-03-12:
Why can't I send pictures directly

hhhhhhwww, 2024-03-12:
But now I want to know, if I want to put Wi-Fi CSI information into turtlebot4's AMCL, how do I modify the bot's code? I'm using a debian package

helloiee, 2024-03-17:
Hi, I have also encountered this problem recently. Have you resolved it

smatarCPR, 2024-03-18:
Hello @hhhhhhwww ,
If you wish to make changes to the robot's base code, you will need to install from source so that you have access to the source code of the robot.

The Turtlebot 4's AMCL is based on Nav2, meaning making changes to AMCL will require you to install Nav2 from source. THe Nav2 utility is not run locally on the Turtlebot 4, it is run on your remote computer. You can install Nav2 from source on your remote computer and make the changes necessary there.

Let me know if I can help with anything else.
Best Regards,
Saif

hhhhhhwww, 2024-03-19:
Yes, I uninstalled the binary package ros-humble-nav2-**, installed the source code version of nav2, but now I don't know how to start nav2 to run Turtlebot4. I can't find specific ros2 launch commands related to nav2. Could you provide some instructions or commands for starting Turtlebot4 with nav2 from source code?
Best Regards

hilary-luo, 2024-03-19:
@hhhhhhwww Nav2 documentation can be found on their website: https://navigation.ros.org/

I believe that what may be missing is a foundational understanding of how ROS 2 workspaces work. Once you download it you need to build your workspace and then source it in the terminals that you are using. To build this knowledge, I would suggest that you first start by going through the basic ROS 2 node tutorial (available in [python](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/first_node_python.html) or [c++](https://turtlebot.github.io/turtlebot4-user-manual/tutorials/first_node_cpp.html)). This should help you understand the basics of building your nodes and workspace and then sourcing it and running it.

Turtlebot packages launch nav2 so as long as they can find the nav2 packages (meaning your workspace is sourced properly) and as long as you don't change how nav2 is launched then you should be able to launch nav2 the same way through the turtlebot4 launch files. If you want to understand what nav nodes that the turtlebot packages launch then look at the launch files that you are calling in the source code (e.g. [slam launch here](https://github.com/turtlebot/turtlebot4/blob/humble/turtlebot4_navigation/launch/slam.launch.py))

I hope this helps to kickstart your journey into working from source. We would like to continue supporting you with Turtlebot4 however I want you to know that we have limited time and can only really dedicate time to help you with Turtlebot4 issues, not general ROS 2 development support. Nav2 has similar venues for help and there is the greater ROS community available to help at robotics.stackexchange.com

hhhhhhwww, 2024-03-20:
yes,I know what you mean.
Thank you.

closed 2024-04-03 by smatarCPR