turtlebot/turtlebot4 #347: Weird problem on turtlebot4 simulator

FranciscoGomez90, 2024-01-25:
### Robot Model

Turtlebot4 Standard

### ROS distro

Galactic

### Networking Configuration

Simple Discovery

### OS

Ubuntu 20.04

### Built from source or installed?

Installed

### Package version

From source, commit "500cfb88f11334fb17488797c3d1bd631ae18923".
Current version from the Ubuntu repo shows the same behavior

### Type of issue

Other

### Expected behaviour

I expect the simulation to run correctly, with no weird artifacts in Gazebo Ignition

### Actual behaviour

Gazebo runs but shows weird artifacts in the visualization section. RViz also reports problems with TF and maps and some other topics. 
[image]

### Error messages

```bash
fran@fran-XPS-15-9570:~/ros_ws$ ros2 launch turtlebot4_ignition_bringup ignition.launch.py
[INFO] [launch]: All log files can be found below /home/fran/.ros/log/2024-01-25-08-26-57-063921-fran-XPS-15-9570-5402
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ign gazebo-1]: process started with pid [5406]
[INFO] [parameter_bridge-2]: process started with pid [5408]
[INFO] [parameter_bridge-3]: process started with pid [5411]
[INFO] [parameter_bridge-4]: process started with pid [5413]
[INFO] [parameter_bridge-5]: process started with pid [5415]
[INFO] [parameter_bridge-6]: process started with pid [5417]
[INFO] [parameter_bridge-7]: process started with pid [5419]
[INFO] [parameter_bridge-8]: process started with pid [5421]
[INFO] [parameter_bridge-9]: process started with pid [5423]
[INFO] [parameter_bridge-10]: process started with pid [5425]
[INFO] [parameter_bridge-11]: process started with pid [5427]
[INFO] [parameter_bridge-12]: process started with pid [5429]
[INFO] [parameter_bridge-13]: process started with pid [5446]
[INFO] [parameter_bridge-14]: process started with pid [5452]
[INFO] [robot_state_publisher-15]: process started with pid [5454]
[INFO] [robot_state_publisher-16]: process started with pid [5456]
[INFO] [static_transform_publisher-17]: process started with pid [5458]
[INFO] [create-18]: process started with pid [5470]
[INFO] [create-19]: process started with pid [5474]
[INFO] [spawner-20]: process started with pid [5482]
[INFO] [hazards_vector_publisher-21]: process started with pid [5504]
[INFO] [ir_intensity_vector_publisher-22]: process started with pid [5506]
[INFO] [motion_control-23]: process started with pid [5532]
[INFO] [wheel_status_publisher-24]: process started with pid [5562]
[INFO] [mock_publisher-25]: process started with pid [5590]
[INFO] [robot_state-26]: process started with pid [5632]
[... 507 lines cut ...]
[ign gazebo-1] [Dbg] [GpuLidarSensor.cc:149] Lidar points for [turtlebot4::base_link::cliff_front_left] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_front_left/scan/points]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [turtlebot4::base_link::cliff_front_right] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_front_right/scan]
[ign gazebo-1] [Dbg] [GpuLidarSensor.cc:149] Lidar points for [turtlebot4::base_link::cliff_front_right] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_front_right/scan/points]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [turtlebot4::base_link::cliff_side_left] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_side_left/scan]
[ign gazebo-1] [Dbg] [GpuLidarSensor.cc:149] Lidar points for [turtlebot4::base_link::cliff_side_left] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_side_left/scan/points]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [turtlebot4::base_link::cliff_side_right] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_side_right/scan]
[ign gazebo-1] [Dbg] [GpuLidarSensor.cc:149] Lidar points for [turtlebot4::base_link::cliff_side_right] advertised on [world/depot/model/turtlebot4/link/base_link/sensor/cliff_side_right/scan/points]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [turtlebot4::ir_intensity_front_center_left::ir_intensity_front_center_left] advertised on [world/depot/model/turtlebot4/link/ir_intensity_front_center_left/sensor/ir_intensity_front_center_left/scan]
[ign gazebo-1] [Dbg] [GpuLidarSensor.cc:149] Lidar points for [turtlebot4::ir_intensity_front_center_left::ir_intensity_front_center_left] advertised on [world/depot/model/turtlebot4/link/ir_intensity_front_center_left/sensor/ir_intensity_front_center_left/scan/points]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [turtlebot4::ir_intensity_front_center_right::ir_intensity_front_center_right] advertised on [world/depot/model/turtlebot4/link/ir_intensity_front_center_right/sensor/ir_intensity_front_center_right/scan]
[ign gazebo-1] [Dbg] [GpuLidarSensor.cc:149] Lidar points for [turtlebot4::ir_intensity_front_center_right::ir_intensity_front_center_right] advertised on [world/depot/model/turtlebot4/link/ir_intensity_front_center_right/sensor/ir_intensity_front_center_right/scan/points]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [turtlebot4::ir_intensity_front_left::ir_intensity_front_left] advertised on [world/depot/model/turtlebot4/link/ir_intensity_front_left/sensor/ir_intensity_front_left/scan]
[ign gazebo-1] [Dbg] [GpuLidarSensor.cc:149] Lidar points for [turtlebot4::ir_intensity_front_left::ir_intensity_front_left] advertised on [world/depot/model/turtlebot4/link/ir_intensity_front_left/sensor/ir_intensity_front_left/scan/points]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [turtlebot4::ir_intensity_front_right::ir_intensity_front_right] advertised on [world/depot/model/turtlebot4/link/ir_intensity_front_right/sensor/ir_intensity_front_right/scan]
[ign gazebo-1] [Dbg] [GpuLidarSensor.cc:149] Lidar points for [turtlebot4::ir_intensity_front_right::ir_intensity_front_right] advertised on [world/depot/model/turtlebot4/link/ir_intensity_front_right/sensor/ir_intensity_front_right/scan/points]
[ign gazebo-1] [Dbg] [Lidar.cc:122] Laser scans for [turtlebot4::ir_intensity_left::ir_intensity_left] advertised on [world/depot/model/turtlebot4/link/ir_intensity_left/sensor/ir_intensity_left/scan]
[ign gazebo-1] [Wrn] [OgreGpuRays.cc:253] Horizontal FOV for GPU rays is capped at 180 degrees.
[ign gazebo-1] [GUI] [Wrn] [ColladaLoader.cc:2394] Triangle input semantic: 'COLOR' is currently not supported
[ign gazebo-1] [GUI] [Wrn] [ColladaLoader.cc:2394] Triangle input semantic: 'COLOR' is currently not supported
```

### To Reproduce

1. Clone github or install from apt
2. ros2 launch turtlebot4_ignition_bringup ignition.launch.py

hilary-luo, 2024-02-20:
@FranciscoGomez90 Can you please describe more clearly what your issue is? What are the artifacts and how are they interfering with the operation of the simulation? I cannot tell from that picture what is being shown, all I can see is that you have not pressed play on your simulation yet. I don't see any errors in the output and you haven't specified details about the rviz errors that you refer to.

RustyCPR, 2024-03-04:
Hi @FranciscoGomez90 
Just reviewing open issues - did you have an update to @hilary-luo questions? 
Thanks!

FranciscoGomez90, 2024-03-05:
I had to move on and used ROS1 and turtlebot 2...
The error is as shown in the image. Gazebo rendered weird planes instead of the testing environment, which was an office or a house. I followed the tutorial of turtlebot_simulation for ROS2, so no custom models or code. No errors are shown in the logs as well, so I don't know how to solve the issue.

RustyCPR, 2024-03-11:
Hi Francisco - thanks for letting us know. 
If you want to try it with TB4 in the future - please consider upgrading to Humble.

RustyCPR, 2024-04-02:
Closing due to inactivity.

closed 2024-04-02 by RustyCPR