turtlebot/turtlebot4 #349: Reverse Gear Jerks

ValentinLeglise, 2024-01-25:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Other Linux

### Built from source or installed?

Installed

### Package version

Package: ros-humble-turtlebot4-navigation
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 2470
Maintainer: rkreinin <[rkreinin@clearpathrobotics.com](mailto:rkreinin@clearpathrobotics.com)>
Architecture: amd64
Version: 1.0.3-1jammy.20230920.011534
Depends: ros-humble-nav2-bringup, ros-humble-nav2-simple-commander, ros-humble-slam-toolbox, ros-humble-ros-workspace
Description: Turtlebot4 Navigation

### Type of issue

Other

### Expected behaviour

the robot of my customer moves backwards without jerking

### Actual behaviour

While teleoperating the robot, when it goes backwards, it jerks hardly for no reason. Video:

https://drive.google.com/file/d/1N1n5OCX39Yscp2fIjiPJoPiKel2FhaiK/view?usp=sharing

### To Reproduce

1 wget -qO - https://raw.githubusercontent.com/turtlebot/turtlebot4_setup/humble/turtlebot4_discovery/configure_discovery.sh | bash <(cat) </dev/tty
2 source .bashrc
3 ros2 daemon stop; ros2 daemon start
4 ros2 topic list

And then execute:

ros2 run teleop_twist_keyboard teleop_twist_keyboard

### Other notes

OS: Lubuntu jammy

RustyCPR, 2024-01-26:
Hi @ValentinLeglise
Can you please set the safety override to `backup_only` or `full` in the create3 [webserver](https://iroboteducation.github.io/create3_docs/webserver/application/#:~:text=minutes%20to%20complete.-,Application%20Configuration,-There%20are%20two) to drive backwards? 
The cliff sensors are on the front half of the robot so it is intended to be driven forwards mainly.

RustyCPR, 2024-02-12:
@ValentinLeglise any feedback on this request? If I don't hear from you I will close this ticket. 
Thanks,
Rusty

ValentinLeglise, 2024-02-22:
Hello,

Sorry for this late reply.

My customer solved this problem by desactivating maximum speed security.

Thank you for your support. You can close this ticket.

RustyCPR, 2024-02-22:
Thanks for confirming.