turtlebot/turtlebot4 #350: My turtlebot4 doesn't boot up

robottb4, 2024-01-26:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

.

### Type of issue

Battery

### Expected behaviour

When connected to the docking station. RPLidar starts spinning, OLED display window displays ip address of TB4 and 5 leds are turned on with green light.

### Actual behaviour

My turtlebot4 used to boot up successfully. A few days ago, it started not wake up. When I connect the tb4 to the docking station. Green led on top of the docking station turns on and boot sequence seems to start. However,
- The RPLiDAR does not spin (no motion at all)
- OLED display window is blank and nearby five leds are also off.
- Only, the large button on the create platform slowly turns on and off
I let the tb4 connected to the docking station all day long, but it stays in that state (no motion of RPLidar, no message on the OLED display window, 5 leds are turned off). This situation continued a few days. I checked the internal power cabling (inlcuding Create and RPi board), but they look properly connected.

### Error messages

```bash
.
```

### To Reproduce

.

### Other notes

.

RustyCPR, 2024-02-28:
Hi @robottb4 
Can you remove the TB4 from the Dock. 
Remove the back door from the Create 3 base. 
There is a [create 3 adapter board](https://turtlebot.github.io/turtlebot4-user-manual/electrical/media/create_adapter.png) with 2 cables connected. 
Can you remove it by pulling it out and then reinstall it? 
My guess is this board may have come out.

RustyCPR, 2024-03-04:
Hi @robottb4 Any update on my request? 
Thanks

RustyCPR, 2024-03-18:
Hello,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,

closed 2024-03-18 by RustyCPR