turtlebot/turtlebot4 #351: Turtlebot 4 not spawning in Gazebo

roeroer, 2024-01-28:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

1.0.1-1jammy.20240104.142840

### Type of issue

Other

### Expected behaviour

After following the installation steps on here https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html
and then running _ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py_ I expected Gazebo to open with the default world and a model of the turtlebot 4

### Actual behaviour

Gazebo opens but there seems to be an issue with spawning the turtlebot model, it does not appear. A few weeks ago everything was working fine but now suddenly the turtlebot will not spawn in Gazebo

### Error messages

```bash
I get pretty much the same warnings as here
https://github.com/turtlebot/turtlebot4_simulator/issues/32#issuecomment-1551163970

it states that the fuel.ignitionrobotics.org URL is deprecated and Unable to deserialize sdf::Model
even tho I used ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
```

### To Reproduce

run ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
Gazebo opens, but without the turtlebot model. The Turtlebot4 HMI appears but no LED lights are on

hilary-luo, 2024-01-29:
@roeroer Since you do say that it used to work for you I would suggest first trying to:
1) Reboot your PC (the only time I had this issue it was fixed on a reboot)
2) Verify if your dependencies are all present and correct - try updating your packages / repairing install - or look at what you have installed / removed since it was last working because that may have broken it)
3) Ensure any workspaces are sourced correctly to have access to said dependencies if necessary

roeroer, 2024-02-07:
That didnt help, the turtlebot is still not spawning. It worked a few weeks ago and when I used the PC the next time it didnt work anymore, I dont think I changed anything since then. I tried rebooting and upgrading everything, but still does not work.

Here it was it says exactly in the terminal when I use ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

hilary-luo, 2024-02-12:
@roeroer The steps that I shared were for the issue that you had referenced and the warning that you had originally mentioned which doesn't quite line up with the terminal output. Is that last picture of the terminal output the very end of the output or is there more that you have not included here? I do not see any errors up to this point.

As a note, in the future, do please share your terminal output in your original issue and please do copy and paste it as text so it is easier to review and search through.

roeroer, 2024-02-19:
I don't know what the problem was, but I uninstalled everything and then installed ROS2 and the simulator package again. Now the TurtleBot is spawning but I'm having some problems in rviz.

When I'm using ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true world:=maze everything opens fine and I can drive the robot around and I can see it in rviz, but I dont know to generate a map while driving around?
rviz looks like this:
[image]

Also I get the following output:
[TerminalOutput.txt](https://github.com/turtlebot/turtlebot4/files/14332401/TerminalOutput.txt)

topic list:
/battery_state
/behavior_server/transition_event
/behavior_tree_log
/bond
/bt_navigator/transition_event
/bumper_contact
/clock
/cmd_vel
/cmd_vel_teleop
/controller_server/transition_event
/diagnostics
/diffdrive_controller/cmd_vel_unstamped
/dock_status
/downsampled_costmap
/downsampled_costmap_updates
/dynamic_joint_states
/function_calls
/global_costmap/costmap
/global_costmap/costmap_raw
/global_costmap/costmap_updates
/global_costmap/global_costmap/transition_event
/global_costmap/published_footprint
/global_costmap/voxel_marked_cloud
/goal_pose
/hazard_detection
/hmi/buttons
/hmi/display
/hmi/display/message
/hmi/led
/interface_buttons
/ip
/ir_intensity
/joint_states
/joy
/kidnap_status
/local_costmap/clearing_endpoints
/local_costmap/costmap
/local_costmap/costmap_raw
/local_costmap/costmap_updates
/local_costmap/local_costmap/transition_event
/local_costmap/published_footprint
/local_costmap/voxel_grid
/local_costmap/voxel_marked_cloud
/local_plan
/map
/map_metadata
/map_updates
/mobile_base/sensors/bumper_pointcloud
/oakd/rgb/preview/image_raw
/odom
/parameter_events
/particle_cloud
/plan
/planner_server/transition_event
/pose
/preempt_teleop
/robot_description
/rosout
/scan
/sim_ground_truth_dock_pose
/sim_ground_truth_pose
/slam_toolbox/feedback
/slam_toolbox/graph_visualization
/slam_toolbox/scan_visualization
/slam_toolbox/update
/slip_status
/standard_dock_description
/stop_status
/tf
/tf_static
/velocity_smoother/transition_event
/waypoint_follower/transition_event
/waypoints
/wheel_status
/wheel_ticks
/wheel_vels

hilary-luo, 2024-02-20:
Glad to hear you figured out some of your issues. Generally if reinstalling works then that means that something interfered with the files, or the dependencies of the package (could be some other update or install/uninstall that was run or a manual change). 

The mapping should start as soon as you press play on the simulation.

Did you try also uninstalling and reinstalling the Humble Nav2 packages? If not try that, and make sure that you are running the latest packages for both turtlebot4 and navigation. I am unable to replicate your issue. The terminal output that you shared doesn't make sense to me because the system is complaining that it is outside of the costmap which shouldn't happen if you haven't modified the packages / default configurations.

roeroer, 2024-02-20:
I pressed play in Gazebo and can move the turtlebot there. The model also moves in rviz but it does not create a map or show walls/shelves. 
Since I just uninstalled ros with all its packages and then installed everything again, the Nav2 package should also be freshly installed. But I will try to do it again. I didn't make any changes to the packages or configurations... I only added my own world files to the worlds folder but didn't modify any of the excisting files.

I accidentaly installed the simulator package and ignition fortress through apt before I realised I need the source installation for using my own world files. I'm not sure if i uninstalled the debian package before source installing. Could that cause any problems?

hilary-luo, 2024-02-20:
You can run source versions of packages at the same time as having the installed versions but you do need to make sure that you source the one that you want to use in the right order. If you just source `/opt/ros/humble/setup.bash` then it will only use the installed packages. If you then source your custom workspace afterwards, then it will redirect to the custom workspace for any packages that are in the workspace (even if they were already in opt/ros). If you do this in the wrong order then it will not use your custom workspace. 

That being said, you should only need the simulator package from source for using your own world files, not ignition. If you are running ignition from source I would suggest switching that back to the installed debian and then verify this issue again.

Knowing that you are running partly from source does change some things in where we take the troubleshooting. As it stands have you made any changes yet? (including adding your world files) and have you been sourcing this custom workspace before running it?

roeroer, 2024-02-25:
So I tried uninstalling and installing the nav2 packages which didn't work and I also did the source installation of the Simulator package again but I still get the messages about the costmap. I always source my workspace afterwards (I just looked it up to be sure I source correctly and now I am unsure. I always did it as you described here: https://github.com/turtlebot/turtlebot4_simulator/issues/47#issuecomment-1652357567  (so with source install/setup.bash) but in the ros documentation https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html I understand that I need to source with source install/local_setup.bash. So which one do I need to use?

Yes I only installed the simulator package from source, ignition is running from the debian package.

As for the changes, last week I had only added some world files, but now with the new source installation of the package I didn't make any changes or modifications yet but it is still not working. The terminal output now looks like this (I saw a few errors):
[terminal_25.02.txt](https://github.com/turtlebot/turtlebot4/files/14396517/terminal_25.02.txt)

ajjaya18, 2024-03-25:
@roeroer can you solve?

hilary-luo, 2024-03-25:
@roeroer if you run everything from debian including the simulator package do you still see this same issue? I have been trying to see if I can replicate your situation with no success. Also if you try and launch without nav2 enabled, does that change anything? I cannot come up with any reason that the robot should be spawned outside of the costmap unless changes have been made to the source files.

In regards to sourcing setup.bash vs local_setup.bash they are similar in what they achieve with a slight difference. local_setup.bash will only source that specific workspace where that file resides. If you source setup.bash it will first source any parent workspaces (that that workspace is dependent on) and then the local one. The ROS Documentation you linked actually does a good job at explaining this, saying:

> Sourcing the local_setup of the overlay will only add the packages available in the overlay to your environment. setup sources the overlay as well as the underlay it was created in, allowing you to utilize both workspaces.
>
>So, sourcing your main ROS 2 installation’s setup and then the ros2_ws overlay’s local_setup, like you just did, is the same as just sourcing ros2_ws’s setup, because that includes the environment of its underlay.

It is also explained here: https://colcon.readthedocs.io/en/released/developer/environment.html#workspace-level