turtlebot/turtlebot4 #352: Comm and battery LEDs not working, no connection with controller

roeroer, 2024-01-28:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Discovery Server

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

22.04 ros2 humble

### Type of issue

Other

### Expected behaviour

all LEDs should turn on when powering on the robot and the controller should automatically connect for manual control

### Actual behaviour

the Comm and Batt LEDs do not turn on when powering on the robot and steering the Turtlebot with the Controller is no longer possibble

### To Reproduce

we switched everything to discovery server (following the steps from here https://turtlebot.github.io/turtlebot4-user-manual/setup/discovery_server.html ) because we cant connect the turtlebot to a network
after factory reset of the create the LEDs did not turn on again and the controller didnt connect automatically anymore. We tried connecting the Controller manually and  we tried installing a new image on the Raspberry Pi, nothing helped so far

RustyCPR, 2024-02-29:
Hi @roeroer 
Can you SSH into the pi with a wired connection and verify what the screen is displaying?
You could also reinstall the OS on the RPI - https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#recovering-the-raspberry-pi

If your Bluetooth controller requires to be Bluetooth paired, ensure you charge it using a mobile phone USB charger and then follow this process: https://turtlebot.github.io/turtlebot4-user-manual/setup/basic.html#turtlebot-4-controller-setup 
Regards,
Rusty

RustyCPR, 2024-03-18:
Hello,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,

closed 2024-03-18 by RustyCPR