turtlebot/turtlebot4 #353: Bumper cannot be disabled

halimimaha, 2024-01-30:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

.

### Type of issue

Other

### Expected behaviour

.

### Actual behaviour

I followed this tutorial: https://iroboteducation.github.io/create3_docs/api/reflexes/ but when I send the command to disable the bumber it doesn't work, the bumber is still activate, any suggestions ? 

### To Reproduce

i sent this command : 

ros2 param set /robot1/motion_control reflexes.REFLEX_BUMP false

closed 2024-01-31 by halimimaha

nisarganc, 2024-09-02:
It worked when I added these lines to ROS 2 parameters file in web server of create 3:
motion_control:
  ros__parameters:
    safety_override: "full"
    reflexes_enabled: false

Note that false is a boolean and not a string!!

nisarganc, 2024-09-06:
I had been facing the same issue and following these steps helped me disable all the reflexes. 
1. Update the Create 3 firmware to latest version (H2.6) as well as all packages on Raspberry Pi packages (sudo apt update && sudo apt upgrade)
2. Edit ros config script in web server of Create 3 as follows:
/robot1/motion_control:
    ros__parameters:
        safety_override: "full"
        reflexes_enabled: false

Note that new firmware update asks you to add namespace of the robot for the motion_control node and reflex parameters are boolean and not string.