turtlebot/turtlebot4 #354: Turtlebot4 Simulation turtlebot fails to respond to commands despite /cmv_vel published

JHPHELAN, 2024-02-01:
### Robot Model

Select One

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Package: ros-humble-turtlebot4-simulator
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 41
Maintainer: rkreinin <rkreinin@clearpathrobotics.com>
Architecture: amd64
Version: 1.0.1-1jammy.20240126.090637

### Type of issue

Other

### Expected behaviour

Upon 
ubuntu@LinuxBox:~$ ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
or
ubuntu@LinuxBox:~$ ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py slam:=true nav2:=true rviz:=true
expected Gazebo to launch and robot to respond to on-screen or keyboard movement commands.

### Actual behaviour

The simulated robot failed to respond to movement commands even though 
$ ros2 topic echo /cmd_vel 
demonstrated published commands responding to GUI or keyboard commands.
Sending UNDOCK command via GUI did not solve problem.
Computer reboot and software re-install failed to solve problem.

### Error messages

```bash
NO error messages.
```

### To Reproduce

ubuntu@LinuxBox:~ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
In Gazebo select BUTTONS.  Click buttons.  No response.
Select KEYBOARD.  Press keys as instructed. No response.
Select Turtlebot4 HMI button (4) [undock].  Try commands above again.  No response.

### Other notes

ubuntu@LinuxBox:~$ ros2 topic info /cmd_vel
Type: geometry_msgs/msg/Twist
Publisher count: 1
Subscription count: 2

ubuntu@LinuxBox:~$ ros2 topic echo /cmd_vel
linear:
  x: 1.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
linear:
  x: 0.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: -0.5

kfu02, 2024-02-02:
Note: I have a similar problem, but on hardware. I am publishing messages to /cmd_vel manually from the command line, ros is correctly receiving them, but the turtlebot does not move.

RustyCPR, 2024-03-04:
@JHPHELAN  Can you please confirm you pressed play?  Located bottom left of the Gazebo screen. 

[image]

JHPHELAN, 2024-03-06:
Tried both ways.  Play / Pause.  Keyboard / Buttons / Sliders.  No movement.  ros2 topic echo /cmd_vel shows commands are being published.  Just no movement of the virtual robot.

RustyCPR, 2024-03-11:
HI @JHPHELAN 
Can you please run only: `ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py`
Please share the terminal outputs after you press play.

RustyCPR, 2024-03-18:
Hi @JHPHELAN - are you still experiencing issues? 
If yes, can you please run only: ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
Please share the terminal outputs after you press play and drive.
Regards,
Rusty

RustyCPR, 2024-04-02:
Closing due to inactivity.

closed 2024-04-02 by RustyCPR

JHPHELAN, 2024-04-05:
I thought I had responded twice with this info, but guess it didn't go through.
[Turtlebot4Errors.txt](https://github.com/turtlebot/turtlebot4/files/14889239/Turtlebot4Errors.txt)
Attached is output of 
ubuntu@linuxbox:~$ ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py
When it became clear it wasn't going anywhere, was stopped with ^C.
Additional shutdown output follows.
In this case Gazebo didn't even open but kept requesting world names.
Followed directions from 
[https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html](url)
Into a FRESH INSTALL of Ubuntu 22.04 and ROS2 Humble.
Did not install -Common nor -Robot prior as I don't have a physical Turtlebot.
I tried following down the rabbit hole of the deprecation warning and changed fuel.ignitionrobotics.org to fuel.gazebosim.org where I could find them, but that didn't help, so backed out.
Suggestions?  Link to upgraded, non-depricated code?
Thanks!
JHP