turtlebot/turtlebot4 #355: Problems with Navigation Stack

CycleMark, 2024-02-01:
### Robot Model

Select One

### ROS distro

Select One

### Networking Configuration

Select One

### OS

Select One

### Built from source or installed?

Select One

### Package version

Nav2

### Type of issue

Select One

### Expected behaviour

I've been having  multiple problems getting my TB4 to work reliably. 

I've been able to create map reasonable well but when I run:

ros2 launch turtlebot4_navigation nav_bringup.launch.py slam:=off localization:=true map:=cabin_map.yaml

I get issues like:
Failed to transform initial pose in time (Lookup would require extrapolation into the future.  Requested time 1706750419.591249 but the latest data is at time 1706750419.546862, when looking up transform from frame [base_link] to frame [odom])

[controller_server-6] [INFO] [1706749067.275821642] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

[planner_server-6] [INFO] [1706748859.600925426] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist

Any of which mean I can't get the robot to autonomously navigate.

Initial question(s) are do you know what will be triggering these messages. I've run the following commands:

Ensured all services were stopped in the first place.

ros2 launch turtlebot4_bringup lite.launch.py
ros2 launch turtlebot4_navigation nav_bringup.launch.py slam:=off localization:=true map:=cabin_map.yaml

Then sent the pose from rviz2 on my Linux machine. I know the to machines might be a few milliseconds out re: timing which would give rise to one of those issues. In that instance how can I set the transform time tolerance?

Thanks

Mark

### Actual behaviour

Correct and reliable  navigation

### To Reproduce

ros2 launch turtlebot4_bringup lite.launch.py
ros2 launch turtlebot4_navigation nav_bringup.launch.py slam:=off localization:=true map:=cabin_map.yaml

hilary-luo, 2024-02-01:
@CycleMark Please revise your issue and provide the following fields:

- Robot Model
- ROS distro
- Networking Configuration
- OS
- Built from source or installed?
- Package version
- Type of issue

Once you have done so then we can look into your issue.

CycleMark, 2024-02-01:
Just updates - sorry I thought I'd done that when I logged the call

RustyCPR, 2024-03-04:
Hi @CycleMark 
To address your question:  _how can I set the transform time tolerance?_ = Fix time syncing 
Our team reviewed this issue along with #370 

To address both issues can you please update to ROS2 Humble and  Use the[ discovery server network configuration. ](https://turtlebot.github.io/turtlebot4-user-manual/setup/networking.html#discovery-server)
Setup NTP syncing from create3 to Pi (for NTP instructions see details here: https://github.com/turtlebot/turtlebot4/issues/216#issue-1797043215

Regards,
Rusty

RustyCPR, 2024-03-18:
Hello,
This issue is being closed due to inactivity. If you are still experiencing the issue, feel free to reopen this ticket when you are ready to continue the troubleshooting process.
Best Regards,
Rusty

closed 2024-03-18 by RustyCPR