turtlebot/turtlebot4 #356: TB4 computer doesn't work - Screen OLED and LEDs don't light up.

ValentinLeglise, 2024-02-01:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

I do not know

### OS

Ubuntu 20.04

### Built from source or installed?

Installed

### Package version

*

### Type of issue

Other

### Expected behaviour

My customer's TB4 PC should switch on at the same time as the rest of the robot. The OLED screen and green LEDs should light up when the robot starts up.

### Actual behaviour

The robot base button light up and boot, but the PC of the TB4 doesn't. The OLED screen and the LEDs show nothing. However, the LIDAR turns well. When the robot was last used, everything was working normally. But the next day, the PC wouldn't turn on.
You can see the two videos:

https://drive.google.com/file/d/1Fd2MmTs7aWFUBwIRyOgNr9QdFtCCp6ZE/view?usp=drive_link
https://drive.google.com/file/d/1ARd96A_yeMfr9Y5yMMc3ryj6BBruhd_r/view?usp=drive_link

### To Reproduce

*

ValentinLeglise, 2024-02-08:
Hello ClearPath.

The problem has been solved. It was a problem with the installation of some ROS instances. Sorry for the inconvenience

Regards.

closed 2024-02-08 by ValentinLeglise

Alice-devpe, 2024-03-21:
How did you solved the problem? cause i reinstalled the software in an SD card, but still nothing appears in the screen