turtlebot/turtlebot4 #357: Switch on turtlebot 4 without docking station

CycleMark, 2024-02-02:
### Robot Model

Turtlebot4 Lite

### ROS distro

Galactic

### Networking Configuration

Discovery Server

### OS

Ubuntu 20.04

### Built from source or installed?

Installed

### Package version

Galatic

### Type of issue

Other

### Expected behaviour

For the life of me I can't find the answer to this. Is it possible to power on the TB4 without having to place in on the dock? 

Thanks

Mark

### Actual behaviour

Power on bot without dock

### To Reproduce

N/A

hilary-luo, 2024-02-20:
No, unfortunately you do need to place it on the docking station to turn it on. This restriction comes from the create3 base that the robot is built from.

CycleMark, 2024-03-02:
OKay thanks - an annoying "feature" - will close the issue [if anyone finds a way to hack around this would love to know]

Cheers

Mark

closed 2024-03-02 by CycleMark