turtlebot/turtlebot4 #358: Initial wait period is needed before turtlebot responds to cmd_vel command

xiaoshan-lin, 2024-02-02:
### Robot Model

Turtlebot4 Standard

### ROS distro

Humble

### Networking Configuration

Simple Discovery

### OS

Ubuntu 22.04

### Built from source or installed?

Installed

### Package version

Not sure. The raspberry pi on the turtlebot comes with installed ROS package for turtlebots, we simply did a sudo apt update && sudo apt upgrade. Firmware version of create3 is H2.3.

### Type of issue

Other

### Expected behaviour

After running `ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros_args --remap cmd_vel:=turtlebot1/cmd_vel`, we should be able to control the turtlebot immediately by pressing the keys.

### Actual behaviour

After running `ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros_args --remap cmd_vel:=turtlebot1/cmd_vel`, we usually need to wait for 5~40 seconds (it varies), before we can use the keyboard to teleop the turtlebot. 

This happens for both of our turtlebots.

### Error messages

```bash
No error messages.
```

### To Reproduce

1. connect pc, raspberry pi, and create3 to the same network
2. set up simple discovery by following the official documentation
3. set up robot namespace in raspberry pi and create3 webserver.
4. open an terminal and run ros2 run teleop_twist_keyboard teleop_twist_keyboard --ros_args --remap cmd_vel:=turtlebot1/cmd_vel     (or turtlebot2/cmd_vel )

### Other notes

Not using SLAM as we have a motion capture system. We wrote a simple control script that publishes command to the cmd_vel topic of each turtlebot. We have the same issue as mentioned above for the teleop. After we run the control script, within the first 20~40 seconds, the robot is not respond to the cmd_vel commands. The turtlebots work properly after such an initial wait period. Here are some facts that might be important:
1. The wait period tends to increase if we have more turtlebots.
2. During the initial wait period, (when the turtlebots are not responding to commands), we ssh into the raspberry pi and can see the velocity command by running ros2 topic echo /turtlebot1/cmd_vel (or ros2 topic echo /turtlebot2/cmd_vel)

We want to understand why this happens and seek your suggestions. Thanks in advance.

hilary-luo, 2024-02-20:
My understanding then is that you only see this delay directly after starting to run the teleop_twist keyboard or your script, then after that delay it seems to work fine (not every time that you publish a command). Is this true?

In addition to answering that question, can you share a log from the create3 immediately after you replicate the situation with only one robot on the network (helpful if you also provide the timestamp at which you sent the command and the timestamp at which it started moving as well).

RustyCPR, 2024-03-18:
Hi @xiaoshan-lin - are you still experiencing issues? 
If yes, can you please answer @hilary-luo questions / requests? 
Best regards,
Rusty

RustyCPR, 2024-04-02:
Closing due to inactivity.

closed 2024-04-02 by RustyCPR